emsesp/EMS-ESP #666: Is there a special su password needed for mqtt send commands?

tp1de, 2020-12-17:
I would like to issue special send commands by mqqt to the ems-esp gateway and wait for the respective response.
It doesn't work for me. I do not get any response:
- Is there a special su password needed for send commands or can I use my own one?
- Is this command correct: {"cmd":"send","data":"call system send 0D 90 FF 00 20 01 F5"} / topic: ems-esp/system ?

proddy, 2020-12-17:
there is no authentication on the REST API calls (which makes it very unsecure). I did think about adding it the URL but it'll still show up as plain text. The better solution would be to use an API key generated by EMS-ESP. It's on the wish-list

The syntax for MQTT is almost correct. The payload is {"cmd":"send","data":"0D 90 FF 00 20 01 F5"}

Always try first in the Console, using the same format so you can see what happens and what is returned. So in this case it's `call system ....`.

tp1de, 2020-12-17:
@proddy  Thanks a lot command is working now! 
I tried on console with "call system send ...." . It worked I took it to mqqt 1:1. This was the mistake.

I am using actually mqtt for commands and not web-api for security reasons. Since I have my own mqtt-server I can control security. 

I just ask myself, if there will be any advantage of using mqtt vs. rest api? Or will be rest api the solution for the future?
What would you recommend mid-term?

For my km200 gateway I read and write with rest api commands since 2 1/2 years. Just de- and encryption is a bit complex. (rijndael 128 / ECB) and the gateway is very slow and not as responsive as ems-esp. But on the km200 a lot more data-points are available compared to the actual version of ems-esp - so I use both since some data-points disappeared in the last firmware-version from buderus and are now availabe in ems-esp.

proddy, 2020-12-17:
I built the web api for https://emsesp.github.io/docs/#/loxone but it turned out to be handy for other actions, especially in debugging because you bookmark the URL.

tp1de, 2020-12-17:
@proddy Yes I can confirm that the web api works fine and stable - just access control is missing. But excellent functionality !

I use the web api calls from node-red to read and write values. It works on the ems-esp as well as directly to the km200. For the Buderus gateway de- and encryption is additionally needed and response time is extremly long. The ems-esp is 100 times faster especially with the esp32.

closed 2020-12-22 by tp1de