emsesp/EMS-ESP #667: Status heating_active and tapwater_active toggle between 0 / 1 when read with mqtt

tp1de, 2020-12-17:
I use my mqtt server. Status info is send every 10 seconds by ems-esp gateway. 
The topics heating_active and tapwater_active are send every 10 seconds with value 1 (on). Some milliseconds later the data is send again with value 0 (off).

I recorded the screen with MQTT explorer:
[screen.zip](https://github.com/proddy/EMS-ESP/files/5709322/screen.zip)

MichaelDvP, 2020-12-17:
Can you check with `watch 8` if the boiler publishes messages 0x18 and 0xE4 with different values. These topics are only published on change, not with fixed timing.

proddy, 2020-12-17:
thanks for reporting and making the video. In addition to what Michael asked, are you on the latest dev build? It's strange that I don't see the 'status' topic which shows the version.

tp1de, 2020-12-17:
I am on b7. Status is now shown. 

[image]

tp1de, 2020-12-17:
@MichaelDvP 
> Can you check with `watch 8` if the boiler publishes messages 0x18 and 0xE4 with different values. These topics are only published on change, not with fixed timing.

When heating is off and no ww-needed then the status is constantly on 0. When heating on and ww-pump is on to load the ww-buffer the every second the mqqt-message is send with the toggle between 1 and 0 as shown in the video.

ems-esp:/$ watch 8
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x08
001+01:31:00.947 N 0: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 F0 48 09 CC 3E 01 C3 46 46 07 00 00 00 00 00 01 99 02 3F 10 00 01 C3 80                                                    00
001+01:31:01.097 N 1: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A5 80 00 7F FF 80 00 (offset 27)
001+01:31:02.591 N 2: [emsesp] Boiler(0x08) -> Me(0x0D), UBATotalUptime(0x14), data: 13 CD C6
001+01:31:02.848 N 3: [emsesp] Boiler(0x08) -> Me(0x0D), UBAParametersPlus(0xE6), data: 01 50 1E 5A 46 12 64 00 06 FA 1E 03 05 64 00 00 00 28 00 41 03 00 00 00 00 00 00
001+01:31:03.097 N 4: [emsesp] Boiler(0x08) -> Me(0x0D), UBAParametersPlus(0xE6), data: 00 00 00 00 00 (offset 25)
001+01:31:03.344 N 5: [emsesp] Boiler(0x08) -> Me(0x0D), UBAParameterWWPlus(0xEA), data: 00 00 00 00 00 01 34 FB 00 0A 01 01 46 D8 00 B4 3C 3C 2D 0A 3C 64 64 00 00 01
001+01:31:04.019 N 6: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 F0 48 09 CC 3E 01 C6 46 46 07 00 00 00 00 00 01 9B 02 3B 10 00 01 C6 80                                                    00
001+01:31:04.172 N 7: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A5 80 00 7F FF 80 00 (offset 27)
001+01:31:05.944 N 8: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 F0 48 09 CC 3E 01 C7 46 46 07 00 00 00 00 00 01 9C 02 3A 10 00 01 C7 80                                                    00
001+01:31:06.099 N 9: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A5 80 00 7F FF 80 00 (offset 27)
001+01:31:06.960 N 10: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 01 00 01 00 00 02 00 00 00 00 00 01 C8 46 46 50 00 2A 46 3E
001+01:31:07.142 N 11: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 F0 48 09 CC 3E 01 C9 46 47 07 00 00 00 00 00 01 9E 02 3A 10 00 01 C9 80                                                    00
001+01:31:07.297 N 12: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A5 80 00 7F FF 80 00 (offset 27)
001+01:31:07.476 N 13: [emsesp] Gateway(0x48) <- Boiler(0x08), UBAMonitorFast(0x18), data: 02 (offset 11)
001+01:31:07.487 N 14: [emsesp] Boiler(0x08) -> Gateway(0x48), UBAMonitorFast(0x18), data: <empty> (offset 11)
001+01:31:07.566 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBADHWStatus(0xE9), data: 34 01 9E 01 9E 00 00 00 00 46 3C 00 11 31 00 A5 C7 00 04 DC 64 00 00 34 00 34
001+01:31:07.728 N 16: [emsesp] Boiler(0x08) -> All(0x00), ?(0x8E4), data: 00 00 00 F0 48 00 00 30 46 3E
001+01:31:09.067 N 17: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 F0 48 09 CC 3E 01 CB 46 47 07 00 00 00 00 00 01 A1 02 37 10 00 01 CB 80                                                    00
001+01:31:09.222 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A5 80 00 7F FF 80 00 (offset 27)
001+01:31:09.636 N 19: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 01 02 00
001+01:31:09.877 N 20: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2E0), data: 01 30 64 00 01
001+01:31:09.949 N 21: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0x2EA), data: 00
001+01:31:10.993 N 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 F0 48 09 CC 3E 01 CC 46 47 07 00 00 00 00 00 01 A2 02 36 10 00 01 CC 80                                                    00
001+01:31:11.147 N 23: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 01 A5 80 00 7F FF 80 00 (offset 27)
ems-esp:/$ watch off
Watching telegrams is off

MichaelDvP, 2020-12-17:
Is this a sequence where the heating active toggles? There is only N13/N14, but the empty telegram should not do anything. The E4 always have state 07 and do not change.
Please try a `log all` to see what triggers the publish of `heating_active`.

tp1de, 2020-12-17:
The sequence was done when heating and ww-buffer load was active in parallel. When ww-buffer was loaded boiler was off and no further changes where done.

MichaelDvP, 2020-12-17:
Yes i know, state 7 means burner+heating+ww-loading on. But it there is no change and you complain that mqtt state changes.

MichaelDvP, 2020-12-18:
I think i know what can cause the issue. The KM200 seem to request single data like in N13 above. If it do this also for E4, which is completly broadcasted every10 sec, this influence the heating active. I'll fix it, but it would be better do have a `log all` to verify my suspicion.

tp1de, 2020-12-18:
@MichaelDvP this is the log you asked for  - just heating active.
[ems.log](https://github.com/proddy/EMS-ESP/files/5715304/ems.log)

tp1de, 2020-12-18:
@MichaelDvP I downloaded actual source code from your repository in github and compiled a new bin-file.
With heating on the toggle of heating_active stopped and in mqtt status is constantly on 1 without new updates. Seems to be ok.

I will observe when ww-pump is on if the tapwater_active status will be ok now as well.

MichaelDvP, 2020-12-18:
Yes, i see in the log that is was not the KM200, it was triggert by the second part of the telegram, but should be fixed now.
But i think there are some other telegrams where the offset/length is not proper checked.

tp1de, 2020-12-18:
tapwater_active is now stable as well. We could close this topic.
> But i think there are some other telegrams where the offset/length is not proper checked.

Just tell me where I can support.

proddy, 2020-12-18:
I think Michael just fixed that. He's unstoppable!

closed 2020-12-22 by proddy