emsesp/EMS-ESP #668: more ems-sensors to mqtt

bartplessers, 2020-12-18:
Hi,

is it possible to get more information about the ems-esp system in mqtt?
In hass I only see now:
`sensor.ems_esp_status`

I would love to have some other information of the ems-esp too, like
- Received telegrams	
- Rx line quality
- Sent telegrams
- Tx line quality
- Device (Platform / SDK)
- System Uptime
- CPU Frequency
- Heap (Free / Max Alloc)
- Free System Memory
- Sketch (Size / Free)
- Flash Chip (Size / Speed)
- File System (Used / Total)
etc...

This would be great to do some debugging. Or just to impress my girlfriend with a flashy dashboard were you can see how many telegrams are already received :-) :-) :-)

thanx for taking a look at this!

grtz
B

proddy, 2020-12-22:
we would need to split over the topics.

`heartbeat` is published every minute with dynamic values. There is where the Rx/Tx line quality and count could be sent.

`info` is a one-off topic sent on boot up. Stuff like the device platform, cpu clock, sketch size, flash size etc is better suited here.

proddy, 2021-01-17:
in the ESP32 build.

closed 2021-01-17 by proddy