emsesp/EMS-ESP #669: v2.1.1b7 - release for ESP32 doesn`t work

sq4bja, 2020-12-18:
Hi,
The new firmware from dev tree 2.1.1b7 doesn't work (compiled for esp32). Now, i'm using 2.1.1b6. I'm trying to upload BIN file form github using web interface and i'm getting "Error during upload"
[image]

MichaelDvP, 2020-12-18:
I can not confirm, i've downloaded the bin and flashed via upload, no issues.
Please try download again.

sq4bja, 2020-12-18:
The same again. Firmware file looks OK.
[image]

MichaelDvP, 2020-12-18:
Try reloading the webpage before uploading, and/or restart esp. My working downloaded file is exactly same size.

sq4bja, 2020-12-18:
I just done it. It doesn`t help. Still the same error.

MichaelDvP, 2020-12-18:
Does flashing main or b6 work?

sq4bja, 2020-12-18:
I can`t flash again b6 or any other. It looks like errot in b6-ci firmware. 
Only the use of espota.py helped. Now I have B7. It looks like some previous firmware bug that made it impossible to upgrade via www.
Tom

SpaceTeddy, 2020-12-20:
@sq4bja I had similar issues with downloading an compiling the dev tree. Compiling was OK, but after update OTA or web, I was still on 2.1.0. @MichaelDvP gave me the hint to download the the bin file. after downloading this bin file via web it worked fine.

proddy, 2020-12-22:
duplicate of #653

closed 2020-12-22 by proddy