emsesp/EMS-ESP #670: MQTT - first setup - publishing after reboot of device

GrexAut, 2020-12-20:
**Bug description**
As the title says, MQTT activated, MQTT is connected, but nothing publishing (checked with MQTT Explorer):
mosquitto log:
New client connected from 192.168.2.12 as ems-esp (c1, k60, u'emsesp').

**Steps to reproduce**
1.) Flash esp8266
2.) Connect to Wifi
3.) Activate MQTT, setup hostname, user, password
4.) check with MQTT explorer for published topics, will not appear.
5.) Change format from nested to "Home Assistant", there will be topics published, set it back, there are no topics with hostname / ems-esp
6.) reboot the device
7.) ems-esp will show up in MQTT Explorer

**Expected behavior**
MQTT should be beginning of publishing after it is activated, and not to reboot

**Screenshots**
I think it is not needed

**Device information**
/api?device=system&cmd=report.* ==> invalid error
/api?device=system&cmd=report is working:
{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:09:05.235",
    "freemem": 44,
    "fragmem": 20
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 1385,
    "#read requests sent": 82,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Junkers Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.06)",
      "handlers": "0xA3 0x06 0xA2 0x2A5 0x2B9 0x2AF 0x2A6 0x2BA 0x2B0 0x2A7 0x2BB 0x2B1 0x2A8 0x2BC 0x2B2 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x20, ProductID:160, Version:24.05)",
      "handlers": "0x2D7"
    },
    {
      "type": "Mixer",
      "name": "MM200 (DeviceID:0x21, ProductID:161, Version:29.04)",
      "handlers": "0x2D8"
    },
    {
      "type": "Mixer",
      "name": null,
      "handlers": "0x2D9"
    },
    {
      "type": "Gateway",
      "n

proddy, 2020-12-20:
oh, that's not normal behaviour. Could you try connecting to the Console via Telnet, with `log all` and see if there are any MQTT connection errors? Also worth upgrading to the latest dev release (2.1.1) which fixes many issues, although 2.1.0 should just work out of the box.

GrexAut, 2020-12-20:
MQTT is working fine after frist reboot, so it is working for me, maybe this information helps someone, who not did "try it to turn it off and on again" ;)

Output of log all:
`
┌──────────────────────────────────────────┐
│ EMS-ESP version 2.1.0                    │
│ https://github.com/proddy/EMS-ESP        │
│                                          │
│ type help to show available commands     │
└──────────────────────────────────────────┘

ems-esp:/$ log all
Log level = all
000+00:24:51.126 D 0: [mqtt] Publishing topic ems-esp/thermostat_data_hc2 (#1006                                                                                                                           , attempt #1, size 151, pid 1)
000+00:24:51.327 D 1: [mqtt] Publishing topic ems-esp/thermostat_data_hc3 (#1007                                                                                                                           , attempt #1, size 155, pid 1)
000+00:24:51.528 D 2: [mqtt] Publishing topic ems-esp/mixer_data_hc1 (#1008, att                                                                                                                           empt #1, size 82, pid 1)
000+00:24:51.609 D 3: [emsesp] No telegram type handler found for ID 0xCBD (src                                                                                                                            0x48)
000+00:24:51.729 D 4: [mqtt] Publishing topic ems-esp/mixer_data_hc2 (#1009, att                                                                                                                           empt #1, size 81, pid 1)
000+00:24:51.930 D 5: [mqtt] Publishing topic ems-esp/mixer_data_hc3 (#1010, att                                                                                                                           empt #1, size 83, pid 1)
000+00:24:53.388 D 6: [emsesp] Received RC300Monitor
000+00:24:53.576 D 7: [emsesp] Received RC300Monitor
000+00:24:54.217 D 8: [emsesp] Received RC300Monitor
000+00:24:56.662 D 9: [emsesp] Received MC10Status
000+00:24:56.984 D 10: [emsesp] Received UBAMonitorWW
000+00:24:57.293 D 11: [emsesp] Received UBAMonitorFast
000+00:24:57.526 D 12: [emsesp] Received UBAMonitorFast
000+00:25:00.025 D 17: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:25:00.025 D 18: [emsesp] Fetching values for device ID 0x09
000+00:25:00.025 D 19: [emsesp] Fetching values for device ID 0x10
000+00:25:00.025 D 20: [telegram] Tx read request to device 0x10 for type ID 0x2A5
000+00:25:00.025 D 21: [telegram] Tx read request to device 0x10 for type ID 0x2B9
000+00:25:00.025 D 22: [telegram] Tx read request to device 0x10 for type ID 0x2A6
000+00:25:00.025 D 23: [telegram] Tx read request to device 0x10 for type ID 0x2BA
000+00:25:00.025 D 24: [telegram] Tx read request to device 0x10 for type ID 0x2A7
000+00:25:00.025 D 25: [telegram] Tx read request to device 0x10 for type ID 0x2BB
000+00:25:00.025 D 26: [telegram] Tx read request to device 0x10 for type ID 0x2F5
000+00:25:00.025 D 27: [telegram] Tx read request to device 0x10 for type ID 0x31B
000+00:25:00.025 D 28: [telegram] Tx read request to device 0x10 for type ID 0x23A
000+00:25:00.025 D 29: [telegram] Tx read request to device 0x10 for type ID 0x240
000+00:25:00.025 D 30: [emsesp] Fetching values for device ID 0x20
000+00:25:00.025 D 31: [telegram] Tx read request to device 0x20 for type ID 0x2D7
000+00:25:00.025 D 32: [emsesp] Fetching values for device ID 0x21
000+00:25:00.025 D 33: [telegram] Tx read request to device 0x21 for type ID 0x2D8
000+00:25:00.025 D 34: [emsesp] Fetching values for device ID 0x22
000+00:25:00.025 D 35: [telegram] Tx read request to device 0x22 for type ID 0x2D9
000+00:25:00.025 D 36: [emsesp] Fetching values for device ID 0x48
000+00:25:00.177 D 37: [mqtt] Publishing topic ems-esp/boiler_data (#1011, attempt #1, size 559, pid 1)
000+00:25:00.359 D 38: [telegram] Sending read Tx [#83], telegram: 8B 88 14 00 20 2C
000+00:25:00.378 D 39: [mqtt] Publishing topic ems-esp/boiler_data_ww (#1012, attempt #1, size 427, pid 1)
000+00:25:00.391 D 40: [emsesp] Last Tx read successful
000+00:25:00.395 D 41: [emsesp] Received UBATotalUptime
000+00:25:00.579 D 42: [mqtt] Publishing topic ems-esp/thermostat_data (#1013, attempt #1, size 160, pid 1)
000+00:25:00.780 D 43: [mqtt] Publishing topic ems-esp/thermostat_data_hc1 (#1014, attempt #1, size 151, pid 1)
000+00:25:00.958 D 44: [telegram] Sending read Tx [#84], telegram: 8B 88 16 00 20 24
000+00:25:00.981 D 45: [mqtt] Publishing topic ems-esp/thermostat_data_hc2 (#1015, attempt #1, size 151, pid 1)
000+00:25:01.016 D 46: [emsesp] Last Tx read successful
000+00:25:01.021 D 47: [emsesp] Received UBAParameters
000+00:25:01.182 D 48: [mqtt] Publishing topic ems-esp/thermostat_data_hc3 (#1016, attempt #1, size 155, pid 1)
000+00:25:01.383 D 49: [mqtt] Publishing topic ems-esp/mixer_data_hc1 (#1017, attempt #1, size 82, pid 1)
000+00:25:01.434 D 50: [telegram] Sending read Tx [#100], telegram: 8B 88 16 19 20 16
000+00:25:01.467 D 51: [emsesp] Last Tx read successful
000+00:25:01.470 D 52: [emsesp] Received UBAParameters
000+00:25:01.584 D 53: [mqtt] Publishing topic ems-esp/mixer_data_hc2 (#1018, attempt #1, size 81, pid 1)
000+00:25:01.785 D 54: [mqtt] Publishing topic ems-esp/mixer_data_hc3 (#1019, attempt #1, size 83, pid 1)
000+00:25:01.986 D 55: [mqtt] Publishing topic ems-esp/heartbeat (#1020, attempt #1, size 142, pid 1)
[... cut ...]
000+00:25:21.257 D 149: [emsesp] No telegram type handler found for ID 0xC4 (src 0x10)
000+00:25:21.296 D 150: [mqtt] Publishing topic ems-esp/thermostat_data_hc3 (#1035, attempt #1, size 155, pid 1)
000+00:25:21.497 D 151: [mqtt] Publishing topic ems-esp/mixer_data_hc1 (#1036, attempt #1, size 82, pid 1)
000+00:25:21.698 D 152: [mqtt] Publishing topic ems-esp/mixer_data_hc2 (#1037, attempt #1, size 81, pid 1)
000+00:25:21.899 D 153: [mqtt] Publishing topic ems-esp/mixer_data_hc3 (#1038, attempt #1, size 83, pid 1)
ems-esp:/$
`

proddy, 2020-12-20:
ok, worth mentioning indeed.

closed 2020-12-20 by proddy