emsesp/EMS-ESP #671: Setting holiday mode

hamster65, 2020-12-24:
Hi,

I am trying to set holiday mode. The mqtt command looks like this:
{"cmd":"holiday","data":"24.12.2020-02.01.2021","id":"hc1"}
The result is that RC35 switches to holiday-absent mode (low heat until 2th January).
Sine we are at home... how do I tell it that we are are not away? There is no additional parameter in the documentation.

Thanks!

MichaelDvP, 2020-12-26:
For now it's only 'away from home'. I thought about how to implement and call a command (`vacation` for 'away'  and `holiday` for 'at home'?), but i think it's better not to spend ram to register a rarly used command. 
What do you think about the separator between the dates `-` for 'away'  and `+` for 'at home'? e.g. 
`call thermostat holiday 02.01.2021-09.01.2021` for away and `call thermostat holiday 06.02.2021+13.02.2021` for at home?

hamster65, 2020-12-26:
Hi Michael, 
the + separator would work fine and I think its meaning would be clear.
The holiday function is really useful to us because of our work schedule. I am building a dedicated mqtt hardware button so set a holiday for the next day. And for actual holidays, It is very convenient to select the second day with on click through the ui.

MichaelDvP, 2020-12-26:
Ok, you can try the dev. The RC35 supports one ' away from home' and one ' at home'  duration in addition for each heatingcircuit.  
If you want to check use `read 10 3F` for hc1, 49 for hc2.  'Away' is offset 87-92, 'at home' are the last 6 bytes (93-98) in order dmydmy.

hamster65, 2020-12-26:
Thanks! It works fine, I tested with + and - . To end the holiday function (and before setting a new date), you have to dates in the past.

closed 2020-12-26 by hamster65