emsesp/EMS-ESP #672: EMS status: disconnected

Chuckame, 2020-12-27:
**Bug description**
Unable to connect.

**Steps to reproduce**
Using `EMS level shifter v0.9` + `NodeMCU lolin v3`, unable to connect to the EMS bus:
[image]

Tried with tx mode 0 to 3, nothing working

**Expected behavior**
Bus connected

**Device information**
```json
{
  "System": {
    "version": "2.1.0",
    "uptime": "000+00:20:03.812",
    "freemem": 60,
    "fragmem": 1
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 0,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 3,
    "tx_gpio": 1,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 0,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "disconnected"
  },
  "Devices": []
}
```

Pins wiring board -> nodemcu:
- vc -> 3v3
- tx -> tx (gpio 1)
- rx -> rx (gpio 3)

**Additional context**
Migrated from 1.9.5 unstable (dev) to 2.1.0 (stable)

# Question : 
How to check if the board is broken ? maybe some logs to read/ cmd to send, voltage to get ?

bbqkees, 2020-12-28:
EMS-ESP on the ESP8266 uses the pinswap function to reassign the serial port to the alternative pins:
RX GPIO 13
TX GPIO 15
This is because the ESP8266 only has one serial port and on boot it spews out debug data on the default TX pin, this confuses the bus.

Chuckame, 2020-12-28:
Will try 
D1 GPIO 5
D2 GPIO 4

bbqkees, 2020-12-28:
RX is on GPIO 13 which is D7 on the NodeMCU.
TX is on GPIO 15 which is D8 on the NodeMCU.

Chuckame, 2020-12-28:
I just discovered that have 3v pin that is not working... 
Now it is working, thanks

closed 2020-12-28 by Chuckame