emsesp/EMS-ESP #675: v2.2.2 not stable,

aHVzY2g, 2020-12-30:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*
ESP8266 BBW Gateway throws exception on telnet commands and maybe on WebSite interactions. 

**Steps to reproduce**
*Steps to reproduce the behavior.*
1. Clear Flash with, ESP Download tool.
2. Flash the new version.
3. Configure Wifi, NTP and MQTT.
4. Issue command "show values", over telnet. Sometimes it takes more than one command.

**Expected behavior**
*A clear and concise description of what you expected to happen.*
After the command "show values" is issued it should give the known output.

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=info*
This API call also leads to a crash.
```
ems-esp:/system$ show
Uptime:        000+00:00:43.097
Chip ID:       0x00dd6d85
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  31
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x001640C8 (4194304 bytes)
Reset reason:  Exception
Reset info:    Fatal exception:29 flag:2 (Exception) epc1:0x4000e0de epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000008 depc:0x00000000
Sketch size:   974544 bytes (2170880 bytes free)
Free heap:                6192 bytes
Free mem:                 24  %
Heap fragmentation:       5 %
Maximum free block size:  5848 bytes
Free continuations stack: 1216 bytes
```

**Additional context**
*Add any other context about the problem here.*
More Information about my setup is in issue #359 .

I will provide some serial output later in the day.

proddy, 2020-12-30:
The heap is very low (6192). Try restarting and using http://ems-esp.local/api?device=system&cmd=info without the Telnet/Console open. It would be interesting to see how many active devices you have. Did you have any problems with 2.1.0 or any of the earlier dev builds? 

If we still can't get it stable you may need to upgrade to an ESP32

aHVzY2g, 2020-12-30:
The API call still leads to a crash. I didn't had any issues with the version I used before, but I can't remember which version it was. I will downgrade to 2.1 to make sure. 
Would it be possible to swap the wemos d1 mini inside the bbq gateway with a pin compatible esp32. Something like this https://www.bastelgarage.ch/esp32minikit-wemos ?

proddy, 2020-12-30:
yes, I think that ESP32 should work. @bbqkees has done a lot of testing recently so knows better which is compatible.

Still, I'd like to know why 2.1 works and 2.2 doesn't. I'm guessing nothing has changed with your EMS setup (#359) so it could well be a bug. What I can do is emulate your device using the test code and see if I get the same results. It was like 1 boiler, 3 mixers, 2 solar units, controller and a thermostat?

bbqkees, 2020-12-30:
@aHVzY2g Yes that ESP32 will do. But you need to solder the header on it, and then remove some plastic inside the housing. And the side won't fit anymore, and you have to power the ESP32 via USB or service jack.

Maybe it's just easier to stay on 2.1 if that worked without issues.

aHVzY2g, 2020-12-30:
@bbqkees that's fine, I have a 3D-Printer available. @proddy I can confirm that 2.1 works without any problems. 
Here is the api output from 2.1
```
{
  "WIFI": {
    "ssid": "Zeile4",
    "hostname": "ems-esp",
    "static_ip_config": "off"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": "on",
    "host": "mqtt.zeile4.lc",
    "port": 1883,
    "username": "",
    "client_id": "ems-esp",
    "keep_alive": 60,
    "clean_session": "on",
    "max_topic_length": 128,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off"
  },
  "NTP": {
    "enabled": "on",
    "server": "192.168.1.1",
    "tz_label": "Europe/Berlin",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": "on",
    "port": 8266
  },
  "Settings": {
    "tx_mode": 1,
    "ems_bus_id": 11,
    "syslog_level": -1,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "master_thermostat": 0,
    "shower_timer": "off",
    "shower_alert": "off",
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  }
}
```

Also the device show output
```
ems-esp:/system$ show
Uptime:        000+00:03:58.007
Chip ID:       0x00dd6d85
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  31
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x001640C8 (4194304 bytes)
Reset reason:  External System
Reset info:    External System
Sketch size:   956640 bytes (2187264 bytes free)
Free heap:                9296 bytes
Free mem:                 34  %
Heap fragmentation:       2 %
Maximum free block size:  8960 bytes
Free continuations stack: 936 bytes
```
 I will post an Serial Output from the crash of v2.2 asap.

proddy, 2020-12-30:
there was 1 bug in the 2.2.0 posted which was fixed this morning. If you're taking the firmware build from GitHub make sure it's the one from today (dec 30th)

aHVzY2g, 2020-12-30:
Sry probably I didn't do enough research, but how can I connect to serial on the esp8266. According to this documentation https://emsesp.github.io/docs/#/Console it is only possible for esp32. Any hint how I can grap the exception stack?

aHVzY2g, 2020-12-30:
I should clarify, in general I know how to do it :D

MichaelDvP, 2020-12-31:
@aHVzY2g: Is mqtt working?  Can you check the output of `boiler_data`, `boiler_data_ww`, etc. In v2.2.0 there are many new ems+ parameters, together with your large number of devices this can consume to much memory on esp8266. Can you also check if web-api works if terminal is closed? eg. `/api?device=boiler&cmd=info`. On which commands does the esp crash?

aHVzY2g, 2020-12-31:
Mqtt seems to be working. I can post the output if you would like to have it. You are lucky, the api call worked after 3 tries with terminal closed:

```
{
  "heatingActive": "on",
  "tapwaterActive": "off",
  "selFlowTemp": 47,
  "selBurnPow": 100,
  "curBurnPow": 23,
  "pumpMod": 64,
  "outdoorTemp": 4.2,
  "curFlowTemp": 46.9,
  "retTemp": 42,
  "sysPress": 1.8,
  "burnGas": "on",
  "flameCurr": 13.5,
  "heatPump": "on",
  "fanWork": "on",
  "ignWork": "off",
  "heatingActivated": "on",
  "heatingTemp": 80,
  "pumpModMax": 62,
  "pumpModMin": 10,
  "pumpDelay": 1,
  "burnMinPeriod": 10,
  "burnMinPower": 0,
  "burnMaxPower": 100,
  "boilHystOn": -6,
  "boilHystOff": 6,
  "burnStarts": 8800,
  "burnWorkMin": 248162,
  "heatWorkMin": 171783,
  "UBAuptime": 622612,
  "serviceCode": "",
  "serviceCodeNumber": 200,
  "lastCode": "EA(227) 31.10.2019 16:48",
  "maintenanceMessage": " ",
  "maintenance": "off",
  "wWComfort": "Hot",
  "wWSelTemp": 60,
  "wWDisinfectionTemp": 70,
  "wWType": "off",
  "wWChargeType": "3-way valve",
  "wWCircPump": "off",
  "wWCircPumpMode": "2x3min",
  "wWCirc": "off",
  "wWActivated": "on",
  "wWHeat": "off"
}
```

The esp crashes on "show values"

MichaelDvP, 2020-12-31:
So it's only the `show devices` and `call system info` commands that crashes? Then it's probably not a memory overflow, we have to search in `EMSESP::show_devices` for a nullptr in devices or telegram-functions. Please check and post the output to `show` command.

MichaelDvP, 2020-12-31:
Is it only `show values` and `call system info` that crashes? Please check `call thermostat info`, `call boiler info`, `call mixer info`, `call solar info` to identify the device. Does `show devices` give an output?

aHVzY2g, 2020-12-31:
@MichaelDvP 
Sow values = crash (sometimes you have to issue the command twice)
Call system info = crash (sometimes you have to issue the command twice)
Call thermostat info = works
Call boiler info = works
Call mixer info = works
Call solar info = works
show devices = works (yes it gives an output)

@proddy 
You are correct the setup is the same as in #359

proddy, 2020-12-31:
@aHVzY2g apart from the `show values` causing the ESP to crash, does everything else work? i.e. MQTT is publishing the correct data and you can see the values in the web? Unfortunately, you can't use the Serial and EMS Bus together. To use Serial you'll need to compile with the build flag `EMSESP_FORCE_SERIAL` but then it won't detect any EMS devices since the UART is disabled. The easiest way is to wait until your ESP32 arrives and try it. I could write some test code to simulate your exact setup and run it in standalone mode (i.e. with Serial on) but I don't have the time.

@MichaelDvP I still suspect a memory issue. `show values` uses 4kb (EMSESP_MAX_JSON_SIZE_MAX_DYN) of heap memory, which if there is more than 25% fragmentation and could cause a problem.

MichaelDvP, 2020-12-31:
Yes, `show values` is really large, but `calls system info` uses only 2kb buffer as all `call <devcie> info` commands. Maybe `system info` blows the buffer because of the number of devices? 
@aHVzY2g can you check the attached software, i skipped the devices from `system info` and spilt the terminal output of `show values` to 2kb buffer. 
[EMS-ESP-2_2_1b0-esp8266.zip](https://github.com/proddy/EMS-ESP/files/5757690/EMS-ESP-2_2_1b0-esp8266.zip)

proddy, 2020-12-31:
> Maybe system info blows the buffer because of the number of devices?
That's what I'm thinking. Marlon is a super user. I think it's time we fork and focus on the ESP32 port for the next gen users. And build in some fail safe into the ESP8266 code that checks for free heap before assigning it.

MichaelDvP, 2020-12-31:
@proddy: I see that `call system info` have a settings section which is completly redundant to `call system settings` command. Skip this section to make room for more devices in the 2k buffer?

MichaelDvP, 2020-12-31:
Also seen that Marlons SM100 for warm water buffer has no values yet, there are a lot of unknown temperatures in the telegrams. If we add all these temperatures the 8266 is complete out of memory. I agree that switching to esp32 is right for this number of devices.

aHVzY2g, 2020-12-31:
> @aHVzY2g apart from the `show values` causing the ESP to crash, does everything else work? i.e. MQTT is publishing the correct data and you can see the values in the web? Unfortunately, you can't [...]

Yeah I just noticed that, as I looked into the source code. MQTT works fine. The web interface shows values but only sometimes (it's like a 10 % chance it does), also it leads the ESP to a crash occasionally. In fact right now I had a really hard time to reach it overall, signing in took 13 sec.

@MichaelDvP 
The bin you just send is more stable. I had only one crash calling show values and no crash calling system show.

MichaelDvP, 2020-12-31:
Does "only one crash" for `show values` mean that there are also some trys that work?

aHVzY2g, 2020-12-31:
Sry, that is what I meant. It works now. I had one crash so far. I just spammed it over(probably 50 times) telnet with "show values" and it didn't crash. I will monitor how stable its now.

TimoFriedri, 2021-01-06:
I am facing similar issues:

Running 2.2.0 (Device (Platform / SDK) esp8266 / 2.2.2-dev(38a443e)
ESP8266

Webpage does not load, just circles and circles ...
If I connect via telnet and type show values, I just get the 3 devices names but no values.
I had a beta installed before which frooze when I typed show values ( don't remember which beta it was, I think from december) 

----
After a restart via telnet:
- show values works again for some time
- webui works for some time

aHVzY2g, 2021-01-15:
All right from what I can tell it's far more stable now, with the Version from @MichaelDvP . The only issue is in the web interface, with really slow loading times. But I couldn't nail down if it's a network Issue on my site or the ESP.

proddy, 2021-01-15:
the web server could be slow if the wifi strength is <50%. We'll merge Michael's changes into the main soon.

closed 2021-01-15 by proddy

aHVzY2g, 2021-02-08:
with the new version "show Values" leads again to a crash? Did you include the change @MichaelDvP did? 
```
Uptime:        000+00:01:53.774
Chip ID:       0x00dd6d85
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  31
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x001640C8 (4194304 bytes)
Reset reason:  Exception
Reset info:    Fatal exception:29 flag:2 (Exception) epc1:0x4000e0be epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000008 depc:0x00000000
Sketch size:   974544 bytes (2170880 bytes free)
Free heap:                6832 bytes
Free mem:                 26  %
Heap fragmentation:       6 %
Maximum free block size:  6392 bytes
Free continuations stack: 1216 bytes
```
@TimoFriedri do you use Firefox I noticed that the UI is more responsive if I use Chrome.