emsesp/EMS-ESP #677: V2.2.0 unstable after 15 hours

lmdecroos, 2020-12-30:
I had version 2.1.0 running stable for more than a month.
After version V2.2.0 running for 15 hours the EMS gateway caused issues on my RMF20 / RFC20RF circuit and my boiler ended in a A22 error. After a power cycle of by boiler the communication with my RC20RF was restored, but the EMS gateway didn't recover.

see syslog ( INFO level) :
Dec 29 21:40:43 ems-esp - <U+FEFF>000+00:00:00.000 I 0: [syslog] Log level set to INFO
Dec 29 21:40:43 ems-esp - <U+FEFF>000+00:00:00.000 I 1: [emsesp] Syslog started
Dec 29 21:40:43 ems-esp - <U+FEFF>000+00:00:00.000 I 2: [emsesp] EMS Device library loaded with 74 records
Dec 29 21:40:43 ems-esp - <U+FEFF>000+00:00:04.657 I 3: [emsesp] WiFi Connected with IP=192.168.0.121, hostname=ems-esp
Dec 29 21:40:44 ems-esp - <U+FEFF>000+00:00:04.689 I 4: [mqtt] MQTT connected
Dec 29 21:41:01 ems-esp - <U+FEFF>000+00:00:23.009 I 5: [telnet] New connection from [192.168.0.20]:57526 accepted
Dec 29 21:41:01 ems-esp - <U+FEFF>000+00:00:23.009 I 6: [shell] Allocated console pty0 for connection from [192.168.0.20]:57526
Dec 29 21:41:59 ems-esp - <U+FEFF>000+00:01:21.169 I 7: [telnet] Connection from [192.168.0.20]:57526 closed
Dec 29 21:41:59 ems-esp - <U+FEFF>000+00:01:21.169 I 8: [shell] Shutdown console pty0 for connection from [192.168.0.20]:57526
Dec 30 12:19:01 ems-esp - <U+FEFF>000+00:00:00.000 I 0: [syslog] Log level set to INFO
Dec 30 12:19:01 ems-esp - <U+FEFF>000+00:00:00.000 I 1: [emsesp] Syslog started
Dec 30 12:19:01 ems-esp - <U+FEFF>000+00:00:00.000 I 2: [emsesp] EMS Device library loaded with 74 records
Dec 30 12:19:01 ems-esp - <U+FEFF>000+00:00:04.653 I 3: [emsesp] WiFi Connected with IP=192.168.0.121, hostname=ems-esp
Dec 30 12:19:01 ems-esp - <U+FEFF>000+00:00:04.692 I 4: [mqtt] MQTT connected
Dec 30 12:19:19 ems-esp - <U+FEFF>000+00:00:23.556 E 5: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
Dec 30 12:19:24 ems-esp - <U+FEFF>000+00:00:28.755 E 6: [telegram] Last Tx Read op

I tried a software restart and power cycle of the gateway and even another power cycle of my GB162 boiler. But the EMS didn't recover from the crash and was only able to find my RM20 controller.

see syslog ( DEBUG level) :
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:00.000 I 0: [syslog] Log level set to DEBUG
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:00.000 I 1: [emsesp] Syslog started
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:00.000 D 2: [telegram] Tx read request to device 0x08 for type ID 0x07
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:00.000 I 3: [emsesp] EMS Device library loaded with 74 records
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:02.685 D 4: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:02.685 D 5: [telegram] Tx read request to device 0x08 for type ID 0x02
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:02.883 D 6: [emsesp] No telegram type handler found for ID 0x16 (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:03.584 D 7: [emsesp] No telegram type handler found for ID 0x29 (src 0x09)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:03.724 D 8: [emsesp] No telegram type handler found for ID 0x1C (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:04.653 I 9: [emsesp] WiFi Connected with IP=192.168.0.121, hostname=ems-esp
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:04.686 I 10: [mqtt] MQTT connected
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:04.686 D 11: [mqtt] Subscribing to topic: ems-esp/system
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:04.739 D 12: [emsesp] No telegram type handler found for ID 0x24 (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:04.887 D 13: [mqtt] Publishing topic ems-esp/info (#01, retain=0, try#1, size 56, pid 1)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:04.975 D 14: [emsesp] No telegram type handler found for ID 0x33 (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.088 D 15: [mqtt] Publishing topic ems-esp/status (#02, retain=1, try#1, size 6, pid 1)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.182 D 16: [emsesp] No telegram type handler found for ID 0x34 (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.424 D 17: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.424 D 18: [telegram] Tx read request to device 0x08 for type ID 0x02
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.424 D 19: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.424 D 20: [telegram] Tx read request to device 0x09 for type ID 0x02
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.699 D 21: [emsesp] No telegram type handler found for ID 0x16 (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:05.911 D 22: [emsesp] No telegram type handler found for ID 0x18 (src 0x08)
Dec 30 18:04:19 ems-esp - <U+FEFF>000+00:00:06.188 D 23: [emsesp] No telegram type handler found for ID 0x19 (src 0x08)
Dec 30 18:04:20 ems-esp - <U+FEFF>000+00:00:06.932 D 24: [emsesp] No telegram type handler found for ID 0x1C (src 0x08)
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.691 D 25: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.691 D 26: [telegram] Tx read request to device 0x08 for type ID 0x02
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.691 D 27: [emsesp] New EMS device detected with ID 0x09. Requesting version information.
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.691 D 28: [telegram] Tx read request to device 0x09 for type ID 0x02
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.691 D 29: [emsesp] New EMS device detected with ID 0x10. Requesting version information.
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.691 D 30: [telegram] Tx read request to device 0x10 for type ID 0x02
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:07.904 D 31: [emsesp] No telegram type handler found for ID 0x18 (src 0x08)
Dec 30 18:04:21 ems-esp - <U+FEFF>000+00:00:08.170 D 32: [emsesp] No telegram type handler found for ID 0x9B (src 0x11)
Dec 30 18:04:22 ems-esp - <U+FEFF>000+00:00:08.476 D 33: [emsesp] No telegram type handler found for ID 0x9C (src 0x11)
Dec 30 18:04:23 ems-esp - <U+FEFF>000+00:00:09.310 D 34: [emsesp] New EMS device detected with ID 0x08. Requesting version information.
Dec 30 18:04:23 ems-esp - <U+FEFF>000+00:00:09.310 D 35: [telegram] Tx read request to device 0x08 for type ID 0x02
Dec 30 18:04:23 ems-esp - <U+FEFF>000+00:00:09.310 D 36: [emsesp] New EMS device detected with ID 0x09. Requesting version information.

Only after downgrading to version 2.1 my EMS gateway recovered and started to work as expected.

Any idea why version 2.2.0 crashed and was not able to recover?

Thanks in advance

proddy, 2020-12-30:
Damn, you're the second to report issues with the latest 2.2.0. There have been many changes since 2.1.0 and this version has been used by many under the dev branch (and called 2.1.1) so I'm surprised it's unstable. 

I would say stick with 2.1.0 until we can figure out the issue. 

Also
- did you ever try one of the earlier dev builds (like 2.1.1b12)
- can you list which devices you have

lmdecroos, 2020-12-30:
Hi,

No I have never tried one of V2.2 beta’s as V2.1.0 was running so stable and delivered all functionality I need.

ems-esp:/$ show devices
These EMS devices are currently active:

Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Thermostat: RFM20 Remote (DeviceID:0x18, ProductID:94, Version:02.01)
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 

Thermostat: RC20RF (DeviceID:0x19, ProductID:93, Version:02.00)
 This Thermostat will respond to telegram type IDs: 0xAF 

Thermostat: RFM20 Remote (DeviceID:0x1A, ProductID:94, Version:02.01)

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:21.08) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x3E 0x3D 0x48 0x47 0x52 0x51 0x5C 0x5B 0xA5 0x37 

Mixer: MM10 (DeviceID:0x21, ProductID:69, Version:02.01)
 This Mixer will respond to telegram type IDs: 0xAA 0xAB 0xAC 

Switch: WM10 (DeviceID:0x11, ProductID:71, Version:02.00)

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)

MichaelDvP, 2020-12-30:
@lmdecroos please download again the version from today. There was a bug with thermostats at id 0x18 that is fixed now, resulting in "no handler found"  messages.

lmdecroos, 2020-12-30:
I have uploaded today's 2.2.0 build 
974544 Dec 30 10:59 EMS-ESP-2_2_0-esp8266.bin

I will monitor and keep you informed

lmdecroos, 2021-01-01:
Looks much better :)

EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 182133
  #read requests sent: 64501
  #write requests sent: 0
  #incomplete telegrams: 1
  #tx fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%

Jan  1 17:06:22 ems-esp - 001+19:02:05.660 E 11: [telegram] Rx: 10 00 3D FE 91 (CRC 91 != 04)

proddy, 2021-01-01:
great!

closed 2021-01-03 by proddy