emsesp/EMS-ESP #679: password re-enter prompt is not shown

proddy, 2021-01-03:
The 2nd console prompt when changing password (`passwd` or `set wifi password`) is not shown.

[image]

I suspect a side affect of some changes that were made in `shell.cpp` when multi-line editing was introduced.

@MichaelDvP do you mind taking a look?

proddy, 2021-01-03:
still not working on ESP32 in serial. the prompt is not always shown. I'll have a closer look next week.

MichaelDvP, 2021-01-04:
I could reproduce before, but can not reproduce a missing prompt after the change. Working for me on 8266, esp32 console and serial.
I had commented out some prompts in  `shell.cpp` (in loop blocking and loop delay) because the input routine now always displays a new promp for cursor handling. 

But while testing i see that parallel connections do not work proper. If i exit system context in on terminal the commands are also killed for the other and i can not leave the context. Also if i close on connection the second is useless.

proddy, 2021-01-04:
I will test again, could be a mistake on my side. I'm using the 2.3 code on ESP32.

The parallel connections is a known side effect of the dynamic context loading I built to save on precious ESP8266 memory early on in v2. I created an issue https://github.com/proddy/EMS-ESP/issues/409 to fix the problem if losing contexts when exiting from a shell but it was too complex, so I stopped. When we move over to ESP32 I can remove all this dynamic context mess and this problem will go. Also all the 'system' commands can come back to the root.

proddy, 2021-01-04:
confirmed it is fixed, thanks.

closed 2021-01-04 by proddy

MichaelDvP, 2021-01-04:
I've fixed it in 2.3 too.

I think parallel connections are very low priority, it was the first time i used it, and only for testing. I can not see any use case.

commit 95fa4b43 (2021-03-18): fix #679