emsesp/EMS-ESP #680: Unable to go under 20 degrees on heat mode on HA

Chuckame, 2021-01-05:
**Bug description**
I'm unable to select a temperature from HA under 20 degrees. 

**Steps to reproduce**
Current temp: 20+
Select temp on HA of 19.5 or less
Wait 3 or 4 seconds
The wanted temp back to 20

**Expected behavior**
Should go under 20 degrees. 

**Screenshots**
When selecting 19 degrees : 
[image]
After 3 or 4 seconds, back to 20 : 
[image]

**Device information**
```json
{
  "System": {
    "version": "2.1.0",
    "uptime": "008+13:44:22.544",
    "freemem": 45,
    "fragmem": 3
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 3,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "on",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 543221,
    "#read requests sent": 74112,
    "#write requests sent": 31,
    "#incomplete telegrams": 16,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 99,
    "#MQTT publish fails": 44,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:20.05)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "FR100 (DeviceID:0x10, ProductID:107, Version:15.07)",
      "handlers": "0xA3 0x06 0xA2 0x16F 0x170 0x171 0x172 0x179 0x17A 0x17B 0x17C"
    },
    {
      "type": "Solar",
      "name": "ISM1 (DeviceID:0x30, ProductID:101, Version:23.03)",
      "handlers": "0x103 0x101"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:20.05)",
      "handlers": ""
    }
  ]
}
```

**Additional context**
Unable to have half of Temps, e.g. 20.5 set the temp to 21, 21.5 set to 22, etc... Maybe related?

proddy, 2021-01-06:
Can you upgrade to 2.2.0 and try the command in the Console? `su` and then `call thermostat temp 19.5`. If that works then the problem is in your HA code.

Chuckame, 2021-01-13:
Hello... Discovered that in heating mode, we can't go under 19 degrees on the thermostat... Sorry

closed 2021-01-13 by Chuckame