emsesp/EMS-ESP #681: Unable to boot my esp32.

crocchietta, 2021-01-07:
Hello. I am new to the world of ESPs.
I have flashed an ESP32-WROOM-32 with the firmware EMS-ESP-2_2_0-esp32.bin.
I used with success the command
`esptool.py -p / dev / ttyUSB0 -b 921600 write_flash 0x00000 EMS-ESP-2_2_0-esp32.bin`
Then the red led of the ESP32 flashes quickly.

With minicom, I get looped :
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)                     
flash read err, 1000                                                            
ets_main.c 371                                                                  
ets Jun  8 2016 00:22:57
```
is there something else to do before flashing this firmware?

DeepCoreSystem, 2021-01-07:
Hi, I had the same problem yesterday!

Unpacked a brand new D1 mini ESP32 with WROOM-32 and tried to flash the 2.2.0 firmware with the esptool. I thought I bricked it, because I got also the message in a loop you posted above!

My solution was to use the [ESP Home Flasher tool ](https://github.com/esphome/esphome-flasher/releases) ...
After this I had a working ESM-ESP 2.2.0 !!

proddy, 2021-01-07:
@crocchietta using `esptool.py` is tricky with the ESP32. I just copied what PlatformIO does. This is my script:

`esptool.py --chip esp32 --port "COM3" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000  EMS-ESP-2_2_0-esp32.bin`

The files `bootloader_dio_40m.bin`, `partitions.bin` and  `boot_app0.bin` can be found in the `scripts` folder.

It's great that you're using the ESP32. This is definitely the way forward. I'm working on 2.3 now which will be exclusive to the ESP32 and have a lot more features, like the ability to control the thermostat and boiler settings from the WebUI.

crocchietta, 2021-01-07:
@DeepCoreSystem : Thanks, I didn't know about this esphome project.

@proddy : thanks, it is working fine now.
It is therefore not possible to use what the documentation indicates in the "Via USB" part: https://emsesp.github.io/docs/#/Uploading-firmware
For a beginner with ESP, it is very misleading and confusing !

I have the idea later to modify the firmware to no longer use a wifi connection but rather wired. Do you think this is achievable with an ESP32? And turn off the wifi?
Or should we instead use something else (arduino, etc.)?

proddy, 2021-01-07:
Agree, I will update the documentation on how to upgrade. BBQKees has some pretty good notes on this too.

Adding Ethernet is on the to-do list of ESP32. I've been doing some testing using an Olimex board.

DeepCoreSystem, 2021-01-07:
@crocchietta : I learned about ESP Home Flasher Tool because of the [WLED project](http://wled.me) and the problems I had there with esptool.py also.

crocchietta, 2021-01-07:
@ DeepCoreSystem : I haven't fully understood the usefulness of ESP Home yet. I'll have to read the documentation.
@ proddy  :  That's great. Is this card type: ESP32-POE?
I have an ELM Leblanc Mégalis (Heatronics3) with FR100 condensing boiler. I haven't tried ESP-EMs with it yet, hope it will work fine.

bbqkees, 2021-01-07:
@crocchietta  Do NOT use any non-isolated POE board with the EMS bus!
48V POE is not so safe to use if you don't take precautions.
Also lot's of cheap Ethernet boards do not have sufficient filtering which might interfere with the bus as well.

crocchietta, 2021-01-10:
Ok, i take note of that.

I am trying to connect the ESP32 with the EMS interface boards v1. For TX and RX connections (interface card side), I wonder if I should use the ESP32 pins identified Txd and Rxd or other GPIOpins ?

bbqkees, 2021-01-10:
EMS-ESP for the ESP32 is pre set to:
TX: GPIO 5
RX: GPIO 23
Dallas: GPIO 18.

Depending on your board these might have different pin numbers.
You can also change the GPIO pins in the EMS-ESP setting menu.

proddy, 2021-01-17:
please re-open if you're still having issues.

closed 2021-01-17 by proddy