emsesp/EMS-ESP #683: MQTT

WeltenbummlerKA, 2021-01-08:
**Is your feature request related to a problem? Please describe.**
The MQTT Topic is only the Hostname of the network setting or the Topic for Homeassistent.

**Describe the solution you'd like**
I would prefer to create an own Full Topic for the MQTT. So I can fully integrate to my existing topics. 
At the moment is called Junkers and I wanted it like show at the picture 1 below with Smarthome.

**Describe alternatives you've considered**
Maybe it's not difficult to integrate it like Tasmota (picture 2)?

**Additional context**
Picture 1:
[image]

Picture 2:
[image]

proddy, 2021-01-08:
We had this in version 1.9 and removed it. It was called MQTT Base. Would it work if we bring it back and use it as a prefix to each topic, so instead of `ems-esp` it would be `<base>/<hostname>`?

WeltenbummlerKA, 2021-01-09:
Maybe it will work like that with `<base>`. If it's possible to integrate it to modify the mqtt topic it would be helpful for my opinion. If not maybe there is a chance to change it direct at the code and make a binary by myself?

MichaelDvP, 2021-01-09:
@proddy Is there a reason that the topic have to contain the hostname? Why not only a `<base>`  instead of `<hostname>` , that can be the same as the hostname, but also something completely different like a long path `home/heating/buderus`.

proddy, 2021-01-09:
Depends on what problem you're trying to solve. Could you look at the current topic names in https://emsesp.github.io/docs/#/MQTT and let us know what you had in mind?

proddy, 2021-01-09:
> @proddy Is there a reason that the topic have to contain the hostname? Why not only a `<base>` instead of `<hostname>` , that can be the same as the hostname, but also something completely different like a long path `home/heating/buderus`.

not really. I did this when I was running multiple versions of EMS-ESP for testing (e.g. ESP8266 and ESP32) and the hostname is the only unique identifier for a board.

I'm ok to bring back `<base>/%topic%` where the default <base> would be the Wifi hostname. I'll do a first implementation in the 2.3 branch.

WeltenbummlerKA, 2021-01-10:
@proddy I use Tasmota and would like to have my mqtt topics equal. When I use only the hostname than I'm to restricted with the chars and also I get another topic in my FritzBox. It would be nice when you will implement that in the next branch.
Do you have also in mind maybe to implement it into ioBroker like you done with Home Assistent?

MichaelDvP, 2021-01-11:
I've made a test build with base setting in the mqtt-settings. @WeltenbummlerKA Is this the way you think? [https://github.com/MichaelDvP/EMS-ESP/releases/tag/dev](https://github.com/MichaelDvP/EMS-ESP/releases/tag/dev)

Tasmota is another structure with %prefix% `tele`, `stat` and `cmnd`. I dont think it's the right for ems-esp. Do you have the tasmotas on mqtt-adapter in ioBroker or do you use the sonoff-adapter? 
To implement ems-esp to ioBroker it is not a function of ems-esp (like the HA-discovery), it needs a own adapter in ioBroker, or use the mqtt and a little script.

proddy, 2021-01-11:
nice @MichaelDvP, that's how I would have implemented it too. I like the addition of the #sensor fails too. When accepted we'll need to merge these manually into the 2.3.

MichaelDvP, 2021-01-11:
The `dallas_fails` are only published in `call system info` for now, but can be added t heartbeat too.  I use dallas with publish every minute and want to know if i miss some errors, but my 7 sensors do not have any error in 4 days i recorded. I think the spikes from another issue are caused by a faulty sensor.

MichaelDvP, 2021-01-15:
@WeltenbummlerKA Have you tested the build? Please give a feedback. I'd like to get in sync with the official dev again, either by merging or deleting the `Base` changes.

WeltenbummlerKA, 2021-01-16:
> I've made a test build with base setting in the mqtt-settings. @WeltenbummlerKA Is this the way you think? https://github.com/MichaelDvP/EMS-ESP/releases/tag/dev
>
> Tasmota is another structure with [...]

Sorry I didn't checked for new messages. I use mosquito as my mqtt broker. I will flash your code and tell you if it's better for me right now. I think that it will take time till next weekend because I'm temporary on a business trip.

MichaelDvP, 2021-01-25:
Since there is no further feedback i've pushed the `Base` settings as mentioned. It's now in dev v2.2.1b1 and refactor-branch.

closed 2021-01-25 by MichaelDvP