emsesp/EMS-ESP #684: Undiscovered devices (Junkers ZSBH 26-4 A23)

Oggerschummer, 2021-01-09:
**Bug description**
I'm facing unknown devices discovered on the bus after connecting to a Junkers Cerapur Aero ZSBH 26-4 A23 (A hybrid device)
**Steps to reproduce**
Simply run a scan

**Expected behavior**
Devices should be recognized and the corresponding data be presented through MQTT
Missing temperatures, DHW1, Holiday Modes and more that is available through a KM200.

**Device information**
`{
"System": {
"version": "2.1.0",
"uptime": "000+04:30:29.951",
"freemem": 39,
"fragmem": 4
},
"Settings": {
"enabled": "on",
"publish_time_boiler": 10,
"publish_time_thermostat": 10,
"publish_time_solar": 10,
"publish_time_mixer": 10,
"publish_time_other": 10,
"publish_time_sensor": 10,
"mqtt_format": 1,
"mqtt_qos": 0,
"mqtt_retain": "off",
"tx_mode": 3,
"ems_bus_id": 11,
"master_thermostat": 0,
"rx_gpio": 13,
"tx_gpio": 15,
"dallas_gpio": 14,
"dallas_parasite": "off",
"led_gpio": 2,
"hide_led": "off",
"api_enabled": "on",
"bool_format": 1,
"analog_enabled": "off"
},
"Status": {
"bus": "connected",
"bus protocol": "HT3",
"#telegrams received": 20262,
"#read requests sent": 1900,
"#write requests sent": 0,
"#incomplete telegrams": 95,
"#tx fails": 3,
"rx line quality": 100,
"tx line quality": 99,
"#MQTT publish fails": 0,
"#dallas sensors": 0
},
"Devices": [
{
"type": "Thermostat",
"name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:33.02)",
"handlers": "0xA3 0x06 0xA2 0x2A5 0x2B9 0x2AF 0x2A6 0x2BA 0x2B0 0x2A7 0x2BB 0x2B1 0x2A8 0x2BC 0x2B2 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
},
{
"type": "Gateway",
"name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)",
"handlers": ""
},
{
"type": "Controller",
"name": "Controller (DeviceID:0x09, ProductID:152, Version:02.02)",
"handlers": ""
},
{
"type": "Unknown",
"name": "Junkers unknown (DeviceID:0x08 ProductID:167, Version:05.06)",
"handlers": ""
}
]
}`

**Additional context**
Gateway II, connected through service port.

glitter-ball, 2021-01-09:
Wow! Hybrid gas boiler and air-source heat pump in one box? Interesting...

proddy, 2021-01-09:
try the firmware binary from https://github.com/proddy/EMS-ESP/releases/tag/dev

Oggerschummer, 2021-01-09:
Thumbs up! This looks good, thanks for the fast response.
Will have to see into the details tomorrow, but first impression is very positive.

Oggerschummer, 2021-01-10:
Working fine, thanks for the very fast support! We can close this request.

closed 2021-01-10 by Oggerschummer