emsesp/EMS-ESP #685: Question - Info on ADC input

glitter-ball, 2021-01-09:
[Apologies for formatting - no template when I opened the issue and no labels either *8( ]

Somewhere on Github, Gitter or @bbqkees site, there was a previous discussion about the ADC input on the ESP8266. I think maybe @MichaelDvP was using it?

I didn't have a use for it... until today when I found my faithful electricity monitor has died. I thought about keeping the clip-on current transformer and trying to do something with ESPhome or similar to get it into HA. This is a fair bit of work and leaves me with yet another device to power. Then... I realised that, since a major electrical upgrade a few years back, my electrical intake is right next to the boiler and, therefore... EMS-ESP. Things could be looking up *8)

So, I can get the output from the current sensor and convert this to a DC voltage and feed into the ADC input of my EMS-ESP and send it on to HA. Little bit of hardware tweaking needed to get the ADC input out of the gateway box. All being well, I can then get regular current measurements into HA. Calibrating the current sensor and taking a stab at the average voltage, I can convert this to power and then have a basic energy monitor. Hurrah.

But I could do with a bit more info on the ESP8266 ADC input, specifically:

- Are there pads on the gateway board where I can pick it up easily?
- What input voltage range does it expect? Bipolar or unipolar?
- How does the software support the ADC input as standard? I see the Enable ADC option and an ADC value (11 in my case) in the heartbeat MQTT topic, updated every minute. Bit of a long gap between readings to pickup transient changes but accuracy isn't going to be perfect, anyway.

Any tips appreciated - thanks.

MichaelDvP, 2021-01-10:
Esp8266 has a single channel adc on A0 with 10bit and 1V full scale (unipolar). Linearity and stability is not very good. On the D1-mini there are resistors to scale to 3.2V (220k-100k) on the module A0-pin. The pin is on a free pad on some gateways, afair only V1.6 does not have it. 
Due to the bad adc specs the adc is measured every 0.5 sec with a 512 weighted moving average (~5min) and published in heartbeat ervery minute. The output is 10bit 0-1024 raw scale. It's mainly for a slow changing sensor signal. I use it for a pressure sensor.

ESP32 have 2 adcs each with 10 channels and better specs. The D1-mini32 module does not have resistors and scale is 1V. Software uses gpio36-ADC1_0 which is in same position as A0 on 8266-D1-mini.

MichaelDvP, 2021-01-17:
Sorry i was wrong, esp32 adcs are 12 bit and reference voltage is 3.3V. (Tested with nodeMCU32, gpio36).

MichaelDvP, 2021-01-22:
After testing, measuring and comparing the different modules i have to correct me again.  The internal ref voltage of 8266 is 1.1V, with the resistors of D1-mini that gives a voltage range of 3.5V for 1024 digits. The esp32 has some internal scaling parameters and the arduino `analogRead()` scales the input to mV resolution, i.e 4,096V for 4096 digits full scale. I think i'll add a scaling to mV to esp8266 D1-mini  too. That makes it easier to change between the modules.

MichaelDvP, 2021-01-25:
In new dev esp32 and esp8266 scales to mV.

proddy, 2021-03-08:
closing this for now.

closed 2021-10-04 by proddy