emsesp/EMS-ESP #686: Non ASCII chars in serviceCode during warm water heating

rherget, 2021-01-10:
**Bug description**
Probably the main bug is in my boiler, but ems-esp doesn't double check:
My boiler sends non ASCII characters for "serviceCode" while doing warm water heating:
[telegram] Rx: 08 00 18 00 53 03 4F 64 64 0B 11 35 01 02 3F 83 00 80 00 02 10 FF **F0 48** 00 00 FF 01 10 83 00 B7

As far as I understand `Boiler::process_UBAMonitorFast` these bold values are used to create a string which is later on used for creating JSON and sending this to MQTT:

(ngrepped communication between EMS-ESP & mqtt server)
T 2021/01/10 15:16:57.961323 xxx.xxx.xxx.120:49812 -> xxx.xxx.xxx.2:1883 [AP] #14688
  0....ems-esp/boiler_data{"heatingActive":"on","tapwaterActive":"on","selFlowTemp":83,"selBurnPow":100,"curBurnPow":100,"pumpMod":100,"outdoorTemp":2.5,"curFlowTemp":84.6,"boilTemp":83.6,"burnGas":"on","flameCurr":52.8,"heatPump":"on","fanWork":"on","ignWork":"off","heatingActivated":"on","heatingTemp":90,"pumpModMax":100,"pumpModMin":100,"pumpDelay":5,"burnMinPeriod":10,"burnMinPower":0,"burnMaxPower":100,"boilHystOn":-6,"boilHystOff":6,"burnStarts":108156,"burnWorkMin":551329,"heatWorkMin":379213,"UBAuptime":7398241,"serviceCode"
T 2021/01/10 15:16:57.966551 xxx.xxx.xxx.120:49812 -> xxx.xxx.xxx.2:1883 [AP] #14690
  :"**.H**","serviceCodeNumber":0,"lastCode":"6L(517) 03.12.2018 08:21","maintenanceMessage":" ","maintenance":"off"}
(see the bold "." which marks the not printable character)

Later on my home assistant fails when parsing this JSON from MQTT:
[homeassistant.components.mqtt] Can't decode payload b'{"heatingActive":"on","tapwaterActive":"on","selFlowTemp":83,"selBurnPow":100,"curBurnPow":100,"pumpMod":100,"outdoorTemp":2.5,"curFlowTemp":84.6,"boilTemp":83.6,"burnGas":"on","flameCurr":52.8,"heatPump":"on","fanWork":"on","ignWork":"off","heatingActivated":"on","heatingTemp":90,"pumpModMax":100,"pumpModMin":100,"pumpDelay":5,"burnMinPeriod":10,"burnMinPower":0,"burnMaxPower":100,"boilHystOn":-6,"boilHystOff":6,"burnStarts":108156,"burnWorkMin":551329,"heatWorkMin":379213,"UBAuptime":7398241,"serviceCode":"\xf0H","serviceCodeNumber":0,"lastCode":"6L(517) 03.12.2018 08:21","maintenanceMessage":" ","maintenance":"off"}' on ems-esp/boiler_data with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._subscribe_topics.<locals>.message_received at 0x7f3634537940>>)

I'm running v2.2.0 (happens with v2.1.0 too).

**Steps to reproduce**
probably hard without my boiler...

**Expected behavior**
Create valid JSON. The JSON above gets https://jsonlint.com/ to complain about it:
Error: Parse error on line 30:
...41,	"serviceCode": "\xf0H",	"serviceCo
How to achieve this - not sure. Remove non ASCII chars? maybe similar to ngrep?

**Device information**
**ems-esp:/system$ show**
Uptime:        000+08:41:13.951
Chip ID:       0x0095a7cb
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  31
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x001640EF (4194304 bytes)
Reset reason:  Hardware Watchdog
Reset info:    Fatal exception:4 flag:1 (Hardware Watchdog) epc1:0x400043e6 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
Sketch size:   974544 bytes (2170880 bytes free)
Free heap:                6712 bytes
Free mem:                 26  %
Heap fragmentation:       29 %
Maximum free block size:  4432 bytes
Free continuations stack: 344 bytes

**ems-esp:/$ show ems**
EMS Bus is connected.

EMS Bus info:
  Tx mode: 1
  Bus protocol: Buderus
  #telegrams received: 32170
  #read requests sent: 6861
  #write requests sent: 0
  #incomplete telegrams: 0
  #tx fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%

Rx Queue is empty

Tx Queue is empty

**ems-esp:/$ show devices**
These EMS devices are currently active:

(1) Boiler: GB125/MC10 (DeviceID:0x08, ProductID:72, Version:02.04)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 

(3) Thermostat: RC30 (DeviceID:0x10, ProductID:67, Version:02.05) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x3E 0x3D 0x3F 0x48 0x47 0x49 0x52 0x51 0x53 0x5C 0x5B 0x5D 0xA5 0x37 

(2) Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.00)

proddy, 2021-01-10:
can you do check the following
- look at the boiler what the service code is (if it has a display) and check if its the same as EMS-ESP
- do a `watch 18` so we can see what the code is

rherget, 2021-01-10:
During "idle" time, it show "0Y" and this is from the console:
000+09:25:35.936 N 291: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 2C 02 48 64 00 01 01 20 00 02 2F 83 00 80 00 00 00 FF 30 59 00 00 FF 00 00 83 00
000+09:25:36.317 N 292: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 96 74 32 00 (offset 27)
(seems correct to me)

I will try to catch it during warm water heating.

rherget, 2021-01-10:
According to a pdf I found, this is not an error, it is even documented this way:

[image]

The German descriptions match exactly my situations.

MichaelDvP, 2021-01-11:
This code with 3 dashes is also described in ems-handbook, i think it's common for all boilers. I also see it, but ioBroker shows a extended char and do not give a n error. If HA needs a valid ascii we should replace that character by a valid ascii that is not used in servicecodes . Suggestions `#` or `~` or `*` or `>`?

proddy, 2021-01-11:
~

rherget, 2021-01-11:
> This code with 3 dashes is also described in ems-handbook, i think it's common for all boilers. I also see it, but ioBroker shows a extended char and do not give a n error. If HA needs a valid ascii [...]

Probably not the non-ascii char is  aproblem for HA, but how the char is encoded in the JSON. But on the other hand i do not think that we would get good results by just mapping this to proper utf-8.
So: ~

proddy, 2021-02-05:
Michael implemented this in dev a while back. Closing.

closed 2021-02-05 by proddy