emsesp/EMS-ESP #690: (ESP32) pre-configured pin layouts

proddy, 2021-01-17:
From the Web, select a pre-configured layout to support the new range of BBQKees boards. So ESP8266 and ESP32 variants.

proddy, 2021-01-19:
Was discussing this with BBQKees today. We decided to call it a Board Profile, which is basically a drop-down list in the Web settings that sets the default values for GPIOs (Rx, Tx, Dallas, Ethernet, LED invert) that match his gateway boards and auto-fill in the settings on the page.

Also a build option -DBOARD_PROFILE=<x> that sets the default values in the firmware so Kees can upload onto the board and it'll work out of the box.

SpaceTeddy, 2021-01-21:
sounds good. Please support also the Wemos MINI D1 ESP32, which works out of the box right now.

bbqkees, 2021-01-21:
Yes board profiles are not exclusively tied to BBQKees boards. It's for all common configurations.