emsesp/EMS-ESP #691: (ESP32) Watchdog 1-WIRE DS18b20

sq4bja, 2021-01-26:
Hi,
I use several ds18b20 thermometers to read boiler temperatures. Sometimes there are problems with the operation of the 1-wire bus. To restore proper operation it is necessary to reset the esp32 module. Is it possible for ems-esp to expose any pin to high state to power dc 3.3v 1-wire sensors? In the absence of a temperature update (watchdog), the 3.3V power supply would be disconnected for 10 seconds to reset the 1-wire bus. This would greatly improve the stability of the system.

Tom

proddy, 2021-01-26:
I guess so, Michael is the expert in this area so let's see what he comes back with.

MichaelDvP, 2021-01-27:
@sq4bja What problems are you facing? I have also 7 ds18b20 attached, but they are working without issue (~1 fail per day). 
I need to understand what is the issue.
Resetting the esp32 does not cut the 3.3V from sensors, why should this help? Is it a issue with the dallas or the esp32? Every read-sequence is started with a 1-wire-reset, why should a power-reset help?

To switch an io-port is easy (can also be done by pin-command), but i don't understand what should be the trigger for this. 

The boards from bbqkees have 3.3V on the JST hardwired, you need to solder somthing to use a io-port. Or do you have parasite-mode, than a powercut could be done by the io. But be aware that esp32 can only source 12 mA on a port and dallas needs 1.5mA each, more than 8 sensors will cause problems.

sq4bja, 2021-01-27:
@MichaelDvP  Thanks for your interest in my enhancement.
The problem with the operation of the DS18b20 mainly concerns fakes. On the market, however, it is difficult to distinguish what is a fake and what is the original. I also found fakes of bad quality. Of course, they usually work properly, but can sometimes freeze. The problem is, for example, the increased intensity of the high-frequency field due to a nearby transmitter. Sometimes the interference is caused by the WiFi network and sometimes something I don't have a clue yet. Correction of pull-up resistors does not help. Sensor failure causes no update of temperature values. Only disconnecting ESP32 from power solves the problem. As you wrote, the software reset itself does not reset the sensors' power supply (I use the 3-cable version).
So I propose to power the 1-WIRE main from any pin. After starting EMS-ESP, the program would turn it high. In a situation where the temperature had not been updated for 1 minute or the sensors stopped responding, the EMS-ESP would go into a low state for 5 seconds. You can also reset the bus when the temperature exceeds the accepted norm (e.g. -127C)
As for the current efficiency of the ESP32 outputs or other ESP boards, I agree. With the number of thermometers greater than 8, it would be necessary to use a keying transistor, or skip the watchdog and turn on the bus power directly under the 3.3V pin.
This would help a lot to keep the heating system running smoothly.

Tom

MichaelDvP, 2021-01-27:
What software version are you using? Can you use the latest dev 2.2.1b1 or v3.0.0b1 and check the dallas fails in heartbeat or `call system info`. Sensors out of valid range (-55 to +125)  and sensors not responding "freeze" for 30 sec (5 cycles) and than the value is empty/missing. 

Are all sensors freeze the same time or only one? Do you get any error or warning in the log? We need to know if the sensor are not responding or report a constant value. If they report the same value every time, how to decide if the sensor needs a reset or the temperature is not changing? (My sensors on solar flows haven't changes for some days now.)

Simply switching a port and connect the sensor does not work, if Vdd goes low, the sensor will enter parasite mode. So we need to switch also the dataline to low, which needs a new/modified 1wire-lib.

sq4bja, 2021-01-27:
I am using version 2.2.1b1. I was able to reproduce a 1-wire bus bug. I analyzed the logs and I can see that the EMS-ESP is picking up the temperature, but the values of all sensors are not changing (they are frozen). 

`000+00:10:50.706 T 972: [mqtt] {"sensor1":{"id":"28-B07E-AE13-1901","temp":34.9},"sensor2":{"id":"28-F043-2800-0080","temp":47.9},"sensor3":{"id":"28-28DA-B413-1901","temp":29.3},"sensor4":{"id":"28-0E2C-2800-0080","temp":45},"sensor5":{"id":"28-6E76-BA13-1901","temp":42.1}}
`
```
ems-esp:/# call dallassensor info
{
  "sensor1": {
    "id": "28-B07E-AE13-1901",
    "temp": 34.9
  },
  "sensor2": {
    "id": "28-F043-2800-0080",
    "temp": 47.9
  },
  "sensor3": {
    "id": "28-28DA-B413-1901",
    "temp": 29.3
  },
  "sensor4": {
    "id": "28-0E2C-2800-0080",
    "temp": 45
  },
  "sensor5": {
    "id": "28-6E76-BA13-1901",
    "temp": 42.1
  }
}
```
This applies to all sensors connected to the bus. 
And it will remain so until the power is disconnected.
It looks like the sensors themselves are freezing. The data appears to be received but not updated. There may be errors, but the "log all" command does not display them. 
I believe that at the moment the anomaly that will identify the problem is the lack of updating all sensors. This can be verified, for example, within 5 minutes. Each sensor will change the indication even by 0.1C within 5 minutes.

Tom

MichaelDvP, 2021-01-28:
It looks like the sensor pesence pulse is missing, indeed we need some error handling for this. Try to switch parasite mode on in ems-esp and leave sensors supplied with normal power. In parasite mode the presence puls is ignored and you'll get an error message in the log. 
But i suppose this is not a bad sensor, it's a loose contact or bad soldering on the pullup of dataline. Than it's not only the power-cycle of esp32, its more the mechanical movement that resets the dallas function.

sq4bja, 2021-01-28:
On your advice, I turned on the Dallas Parasite Mode option. While waiting for the error to appear, I tried to suspend the 1-wire bus. The problem did not occur. I suppose enabling the Dallas Parasite Mode made the 1-wire bus work more stable. Regardless of my case, I think that it is worth considering adding a watchdog option depending on the maintenance of constant temperature indications for a period of 5 minutes or the lack of response of sensors connected to the 1-wire bus.

proddy, 2021-01-28:
is it worth looking into another 1-wire library for the ESP32, like https://github.com/DavidAntliff/esp32-owb ?

MichaelDvP, 2021-01-28:
@proddy i don't think ist esp32 or library specific, i suppose `bus_reset()` in line 70 in `dallassensor.cpp`(dev) returns false because bus is not idle. This is the only cause without error message and leaving sensor values untouched. A defective or to high resistance pullup can cause this, or one of the sensors drains the dataline (a lower value of pullup will help than too). In parasite mode the line is driven high by the processor and pullup is used only while receiving, 

@sq4bja I've pushed a version to my repo that switches gpio27 ~5 sec to low if the line is not idle. Also sensor values are cleared on this error. [see here](https://github.com/MichaelDvP/EMS-ESP/releases). I've checked my sensors and found sensors staying constant over 6 hours and then changing 0.1 degree. I don't like a logic based on unchanged values, i'm sure the issue can be detected better. 

The error checking i'll push also to dev repo.