emsesp/EMS-ESP #693: [Q] WiFi-lost behaviour! Exposing the AP mode again?

curlyel, 2021-01-28:
Just ordered the gateway from Kees and preparing myself for hooking it up and configuring... ;-)

I'm curious to know, how the EMS-ESP acts if it can't connect to the configred WiFi - either because wrongly entered credentials or a simple outage of the WiFi.

I did not find an explanation on it neither in the wiki nor in the issues...

Therefore two questions:

- Will it revert to it's AP mode again with the default WPA credentials and therefore exposing my heating system to the neighborhood?
- In case I've entered a wrong WPA key, is there an option to recover without reflashing with ESPtool etc.?

Thanks again ;-)

MichaelDvP, 2021-01-28:
If you look in the AP-settings you'll find the first menu item `Provide Access Point`  with options `Always`, `When WiFi disconnected` and `Never`. I think this is selfexplaining. 

Also you can check WiFi connection after setting up before leaving the AP. 
And last if you set AP to `Never` and have no wifi connection you can create a Wifi-AP (Hotspot in your smartphone) with the "wrong" settings.

curlyel, 2021-01-28:
Thanks for explaining. Sounds reasonable ;-)

So the best would be to set it to `Always` or `When Wifi disconnected` during initial setup to check if the ESP can really connect to the Wifi with the provided credentials and then (after reboot) to `never `if concerned about unintentional activation of AP mode - right?

MichaelDvP, 2021-01-28:
The default setting is `When Wifi disconnected` and since AP is also protected it is not much risk to have the AP on during boot or if wifi is disturbed.

On esp32 there is also the serial console (via usb), which can be used to reconfigure wifi-settings. But 8266 has only one uart which is used for the ems-bus.

curlyel, 2021-01-28:
Understood, many thanks for clarifying!

closed 2021-01-28 by curlyel