emsesp/EMS-ESP #694: Possible MQTT cmd broken in 2.2.1b1

gurkc006, 2021-01-29:
**Bug description**
Im using the newest EMS-ESP 2.2.1b1 on a Gateway Premium II (P2). After updating to this version, no cmd send to thermostat using MQTT is working.
I already tested setting temp using the console and that works perfectly. So it seems, that either I used a wrong cmd syntax or there is something broken in the MQTT reception.
(sorry, this is my first bug report, so please advise me, if you need more/other information)

**Steps to reproduce**
For example to set the selTemp:
MQTT publish ems-esp/thermostat = {"cmd":"daytemp","data":11,"id":1}

**Expected behavior**
With the previos version 2.2.0 the response from the thermostat was seen in reception of the changed value in /thermostat-data-hc1.

**Device information**
{
  "System": {
    "version": "2.2.1b1",
    "uptime": "000+03:15:29.016",
    "freemem": 39,
    "fragmem": 5
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 7846,
    "#read requests sent": 2566,
    "#write requests sent": 5,
    "#incomplete telegrams": 0,
    "#tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:02.05)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC30 (DeviceID:0x10, ProductID:67, Version:02.05)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x3E 0x3D 0x3F 0x48 0x47 0x49 0x52 0x51 0x53 0x5C 0x5B 0x5D 0xA5 0x37"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.00)",
      "handlers": ""
    }
  ]
}

proddy, 2021-01-29:
can you check the value of MQTT base from the Settings? Should be "ems-esp" to be compatible with 2.2.0

gurkc006, 2021-01-29:
Checked this. it's set to "ems-esp". I check MQTT data with MQTT Explorer. See attached screenshot of trying to set temperature with 
ems-esp/thermostat = {"cmd":"temp","data":11,"id":1} 

[image]

No reaction in the reported set values...
Maybe I should activate a mosquitto on my pc and keep it running in verbose mode to check subscriptions of EMS-ESP to the MQTT broker...

gurkc006, 2021-01-29:
Ok, sorry. I tested connection to a mosquitto broker an my PC. First I just changed the MQTT host setting in the webinterface of the EMS-ESP and clicked "Save". After this, all new MQTT messages were published to my mosquitto broker. But no subscriptions were opend. So I restarted the EMS-ESP via the webinterface and then after restart all subscriptions were made and I could send commands again.
I then changed the host setting back to my productive system, Saved, Reboot and then everything worked again.
So it seems to be important to perfom a reboot after changing settings in the web frontend.
Good to know.
Sorry for the troubles.

gurkc006, 2021-01-29:
Issue was solved by rebooting EMS-ESP.

closed 2021-01-29 by gurkc006