emsesp/EMS-ESP #695: Unexpected behavior of thermostat_data after setting temperature over MQTT

bennybubble, 2021-01-29:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)* **YES**
* *searched the [documentation help section](https://emsesp.github.io/docs)* **YES**

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
Hard to describe but let me try:
*When setting the temperature over MQTT, the next respond with **thermostat_data** contains sometimes the old (or current) temperature and sometimes the new (set) temperature. 
If it contained the 'old ' temperature, we have to wait for the next **thermostat_data** transmisson which contains the new temperature and this can take up to a minute.*

**Steps to reproduce**
*See above.*

**Expected behavior**
*As soon as the temperature is set, the next thermostat_data should contain the new set temperature and not skip 1*

**Screenshots**
*Hard to do since it it is time realted.*

**Device information**
*         ‘Ð©Å "Nefit Proline (DeviceID:0x08 ProductID:122, Version:03.09)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC20/Moduline 300 (DeviceID:0x17, ProductID:77, Version:03.03)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x91 0xA8"
    },
    {
      "type": "Contr protoco—ßq         {
  "System": {
    "version": "2.2.0",
    "uptime": "000+00:33:11.332",
    "freemem": 36,
    "fragmem": 14
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 60,
    "publish_time_thermostat": 60,
    "publish_time_solar": 60,
    "publish_time_mixer": 10,
    "publish_time_other": 60,
    "publish_time_sensor": 60,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
y": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 2
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit Proline (DeviceID:0x08 ProductID:122, Version:03.09)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC20/Moduline 300 (DeviceID:0x17, ProductID:77, Version:03.03)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x91 0xA8"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:114, Version:01.06)",
      "handlers": ""
    }
  ]
}*

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2021-01-30:
I think this is fixed in v2.2.1b1. ([this commit](https://github.com/proddy/EMS-ESP/commit/cdf6aaefd914eab179e5a8fc9bbb6df5455ed94f))
Normally when setting a value the telegram is read back to update the value. But if the  the value is set in one (set-) telegram but readback in another (monitor-) telegram it takes some time for the thermostat to process. I've tested with different delays before readback and get for `temp`-command 50% success for 1 sec, 90% for 1.5sec and 100% for 2 sec. That's why it takes these 2 sec now to publish back the new value.

bennybubble, 2021-01-30:
Seems to work now, the small delay is not a problem.
Thanks for helping me out!

closed 2021-01-30 by bennybubble