emsesp/EMS-ESP #696: RC20 thermostat not detected

fizista, 2021-01-31:
**Bug description**
EMS-ESP does not recognize RC20 controller.

[image]
[image]

```
ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Logamax Plus GB022 (DeviceID:0x08, ProductID:84, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Unknown: unknown (DeviceID:0x09, ProductID:84, Version:01.09)

000+00:38:26.824 D 77: [emsesp] Received UBAMonitorFast
000+00:38:27.004 D 78: [emsesp] Received UBAMonitorWW
000+00:38:36.797 D 79: [emsesp] Received UBAMonitorFast
000+00:38:36.977 D 80: [emsesp] Received UBAMonitorWW
ems-esp:/# scan devices
000+00:38:43.644 D 81: [telegram] Tx read request to device 0x08 for type ID 0x07
000+00:38:43.765 D 82: [telegram] Sending read Tx [#23], telegram: 0B 88 07 00 20 A8
000+00:38:43.815 D 83: [emsesp] Last Tx read successful
000+00:38:46.771 D 84: [emsesp] Received UBAMonitorFast

```

**Device information**
```
{
  "WIFI": {
    "ssid": "szuszu24i",
    "hostname": "ems-esp",
    "static_ip_config": "off"
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": "off",
    "host": "192.168.1.40",
    "port": 1883,
    "username": "",
    "client_id": "ems-esp",
    "keep_alive": 60,
    "clean_session": "on",
    "max_topic_length": 128,
    "publish_time_boiler": 5,
    "publish_time_thermostat": 5,
    "publish_time_solar": 5,
    "publish_time_mixer": 5,
    "publish_time_other": 5,
    "publish_time_sensor": 5,
    "mqtt_format": 3,
    "mqtt_qos": 1,
    "mqtt_retain": "off"
  },
  "NTP": {
    "enabled": "on",
    "server": "time.google.com",
    "tz_label": "Europe/Warsaw",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": "off",
    "port": 8266
  },
  "Settings": {
    "tx_mode": 1,
    "ems_bus_id": 11,
    "syslog_level": 7,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "master_thermostat": 0,
    "shower_timer": "on",
    "shower_alert": "on",
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "on",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "on",
    "bool_format": 1,
    "analog_enabled": "off"
  }
}
```

proddy, 2021-01-31:
try `scan devices` again but after `log all` so we can see the data coming in.

MichaelDvP, 2021-01-31:
The internal GB022 controller 
> Unknown: unknown (DeviceID:0x09, ProductID:84, Version:01.09)

is missing in database, but the thermostat picture shows no display on the RC20, i think it's not connected to the ems-bus.

proddy, 2021-01-31:
I think he may have disconnected it to take photo of the back

fizista, 2021-02-01:
This is what the inside of the thermostat looks like
[image]

After unplugging the thermostat this is the information I get when scanning:

[image]

```
ems-esp:/# log all
Log level = all
ems-esp:/# scan devices
000+00:05:43.922 D 3: [telegram] Tx read request to device 0x08 for type ID 0x07
000+00:05:44.022 D 4: [telegram] Sending read Tx [#12], telegram: 0B 88 07 00 20 A8
000+00:05:44.072 D 5: [emsesp] Last Tx read successful
000+00:05:45.807 D 6: [emsesp] Received UBAMonitorFast
000+00:05:45.986 D 7: [emsesp] Received UBAMonitorWW
000+00:05:55.805 D 8: [emsesp] Received UBAMonitorFast
000+00:05:55.985 D 9: [emsesp] Received UBAMonitorWW
ems-esp:/# scan devices
000+00:05:58.362 D 10: [telegram] Tx read request to device 0x08 for type ID 0x07
000+00:05:58.434 D 11: [telegram] Sending read Tx [#13], telegram: 0B 88 07 00 20 A8
000+00:05:58.484 D 12: [emsesp] Last Tx read successful
000+00:06:00.010 D 13: [emsesp] Fetching values for device ID 0x08
000+00:06:00.010 D 14: [telegram] Tx read request to device 0x08 for type ID 0x14
000+00:06:00.010 D 15: [telegram] Tx read request to device 0x08 for type ID 0x16
000+00:06:00.010 D 16: [telegram] Tx read request to device 0x08 for type ID 0x19
000+00:06:00.010 D 17: [telegram] Tx read request to device 0x08 for type ID 0x33
000+00:06:00.010 D 18: [emsesp] Fetching values for device ID 0x09
000+00:06:00.154 D 19: [telegram] Sending read Tx [#14], telegram: 0B 88 14 00 20 E4
000+00:06:00.578 D 20: [telegram] Sending read Tx [#18], telegram: 0B 88 14 00 20 E4
000+00:06:01.002 D 21: [telegram] Sending read Tx [#19], telegram: 0B 88 14 00 20 E4
000+00:06:01.426 D 22: [telegram] Sending read Tx [#20], telegram: 0B 88 14 00 20 E4
000+00:06:01.725 E 23: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:06:01.850 D 24: [telegram] Sending read Tx [#15], telegram: 0B 88 16 00 20 EC
000+00:06:01.892 D 25: [emsesp] Last Tx read successful
000+00:06:01.896 D 26: [emsesp] Received UBAParameters
000+00:06:02.049 D 27: [telegram] Sending read Tx [#16], telegram: 0B 88 19 00 20 D0
000+00:06:02.092 D 28: [emsesp] Last Tx read successful
000+00:06:02.096 D 29: [emsesp] Received UBAMonitorSlow
000+00:06:02.249 D 30: [telegram] Sending read Tx [#17], telegram: 0B 88 33 00 20 78
000+00:06:02.293 D 31: [emsesp] Last Tx read successful
000+00:06:02.298 D 32: [emsesp] Received UBAParameterWW
000+00:06:05.778 D 33: [emsesp] Received UBAMonitorFast
000+00:06:05.957 D 34: [emsesp] Received UBAMonitorSlow
000+00:06:06.157 D 35: [emsesp] Received UBAMonitorWW
ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Logamax Plus GB022 (DeviceID:0x08, ProductID:84, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Unknown: unknown (DeviceID:0x09, ProductID:84, Version:01.09)

000+00:06:15.726 D 36: [emsesp] Received UBAMonitorFast
000+00:06:15.906 D 37: [emsesp] Received UBAMonitorWW
000+00:06:25.725 D 38: [emsesp] Received UBAMonitorFast
000+00:06:25.905 D 39: [emsesp] Received UBAMonitorWW

```

After plugging in the thermostat, this is the information I get when scanning:

[image]

```
ems-esp:/$ su
Password: 
000+00:04:16.111 N 0: [shell] su session opened on console
ems-esp:/# log all
Log level = all
[... 49 lines cut ...]
000+00:15:01.712 E 274: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request.
000+00:15:01.837 D 275: [telegram] Sending read Tx [#116], telegram: 0B 88 16 00 20 EC
000+00:15:01.879 D 276: [emsesp] Last Tx read successful
000+00:15:01.883 D 277: [emsesp] Received UBAParameters
000+00:15:02.036 D 278: [telegram] Sending read Tx [#117], telegram: 0B 88 19 00 20 D0
000+00:15:02.079 D 279: [emsesp] Last Tx read successful
000+00:15:02.083 D 280: [emsesp] Received UBAMonitorSlow
000+00:15:02.236 D 281: [telegram] Sending read Tx [#118], telegram: 0B 88 33 00 20 78
000+00:15:02.280 D 282: [emsesp] Last Tx read successful
000+00:15:02.285 D 283: [emsesp] Received UBAParameterWW
ems-esp:/# show devices
These EMS devices are currently active:

Boiler: Logamax Plus GB022 (DeviceID:0x08, ProductID:84, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

Unknown: unknown (DeviceID:0x09, ProductID:84, Version:01.09)

```

Sometimes it displays such information in the console:

```
000+00:16:47.033 D 331: [emsesp] No telegram type handler found for ID 0xAE (src 0x17)
```

proddy, 2021-02-01:
Michael's just added the RC20 to the database...

MichaelDvP, 2021-02-01:
Yes, thanks to this information:
>Unknown: unknown (DeviceID:0x17, ProductID:66, Version:01.04)
> 000+00:16:47.033 D 331: [emsesp] No telegram type handler found for ID 0xAE (src 0x17)

we know the device-id and the command set.
@fizista please update to the latest dev and check if all is working. I'm a bit confused that the thermostat seems to disappear after a while.

proddy, 2021-02-01:
@MichaelDvP I was also expecting to see a message like `[emsesp] Unrecognized EMS device (device ID 0x17, product ID 66). Please report on GitHub.` after a `scan devices` command but we never got around to fixing the `EMSESP::clear_all_devices()` function.

Also I think @fizista is using quite an old version of the firmware as we removed those `[emsesp] Received UBAMonitorSlow` type messages a while ago?

fizista, 2021-02-01:
It seems to be working.

I was also able to add the thermostat to Home Assistant.

```
ems-esp:/# show devices
These EMS devices are currently active:

(1) Boiler: GB022 (DeviceID:0x08, ProductID:84, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 

(3) Thermostat: ES72 (DeviceID:0x17, ProductID:66, Version:01.04) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0xAE 0xAD 

(2) Controller: GB022 (DeviceID:0x09, ProductID:84, Version:01.09)
```

```
{
  "System": {
    "version": "2.2.1b1",
    "uptime": "000+00:25:33.993",
    "freemem": 30,
    "fragmem": 19
  },
  "Status": {
    "bus": "connected, instable tx",
    "bus protocol": "Buderus",
    "#telegrams received": 656,
    "#read requests sent": 154,
    "#write requests sent": 64,
    "#incomplete telegrams": 7,
    "#tx fails": 28,
    "rx line quality": 100,
    "tx line quality": 81,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GB022 (DeviceID:0x08, ProductID:84, Version:01.09)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "ES72 (DeviceID:0x17, ProductID:66, Version:01.04)",
      "handlers": "0xA3 0x06 0xA2 0x12 0xAE 0xAD"
    },
    {
      "type": "Controller",
      "name": "GB022 (DeviceID:0x09, ProductID:84, Version:01.09)",
      "handlers": ""
    }
  ]
}
```

closed 2021-02-01 by fizista

proddy, 2021-02-01:
you're welcome

thanks @MichaelDvP !

MichaelDvP, 2021-02-01:
Ups, now we have a bug, i have not seen the exisiting ES72, device 66,  with wrong command-set. Fixed it. 
@fizista please update again, the thermostat should be `ES72/RC20` in the new update

fizista, 2021-02-01:
Uploaded. It's ok.