emsesp/EMS-ESP #697: Thermostat RC20 is not detected after restarting EMS-ESP

fizista, 2021-02-01:
**Bug description**
The EMS-ESP is powered by a USB port.

Thermostat RC20 is not detected after restarting EMS-ESP

**Steps to reproduce**

* EMS-ESP is turn on
* GB022 is turn off
* Switch on GB022
* RC20 and GB022 thermostat are detected, so everything is ok up to this point.
* go to http://[IP address]/system/status
* click restart
* After restarting the EMS-ESP, the RC20 thermostat is not detected
* "scan device" does not find the thermostat
* "scan device deep" finds thermostat!!

**Expected behavior**

EMS-ESP should remember the Thermostat and somehow query if it still exists after startup.

**Device information**

Development Build v2.2.1b1 c872f19

```
{
  "System": {
    "version": "2.2.1b1",
    "uptime": "000+00:05:32.233",
    "freemem": 46,
    "fragmem": 1
  },
  "Status": {
    "bus": "connected, instable tx",
    "bus protocol": "Buderus",
    "#telegrams received": 124,
    "#read requests sent": 30,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 8,
    "rx line quality": 100,
    "tx line quality": 73,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GB022 (DeviceID:0x08, ProductID:84, Version:01.09)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Controller",
      "name": "GB022 (DeviceID:0x09, ProductID:84, Version:01.09)",
      "handlers": ""
    }
  ]
}
```

**Additional context**

```
#############################################
######## GB022 is turn on
#############################################

ems-esp:/# log all
[... 216 lines cut ...]
000+00:02:17.864 D 166: [emsesp] Last Tx read successful
000+00:02:17.864 T 167: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0xAE), data: 04 02 2D 00 E3 00 00 00
000+00:02:17.864 D 168: [emsesp] Toggling fetch for device ID 0x17, telegram ID 0xAE to 1
000+00:02:17.864 D 169: [emsesp] Toggling fetch for device ID 0x17, telegram ID 0xAD to 1
000+00:02:18.029 D 170: [telegram] Sending read Tx [#88], telegram: 0B 97 12 00 20 04
000+00:02:18.095 D 171: [emsesp] Last Tx read successful
000+00:02:18.095 T 172: [emsesp] Thermostat(0x17) -> Me(0x0B), RCErrorMessage(0x12), data: <empty>
000+00:02:18.095 D 173: [emsesp] This telegram (RCErrorMessage) is not recognized by the EMS bus
000+00:02:18.095 D 174: [emsesp] Toggling fetch for device ID 0x17, telegram ID 0x12 to 0
000+00:02:18.095 D 175: [emsesp] No telegram type handler found for ID 0x12 (src 0x17)
000+00:02:18.392 T 176: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 7C 00 00 00 10 00 00 02 31 02 31 80 00 FF FF FF 30 48 00 00 00 00 00
000+00:02:18.571 T 177: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 37 02 31 02 31 21 00 00 01 00
ems-esp:/# 
ems-esp:/# show devices
These EMS devices are currently active:

(1) Boiler: GB022 (DeviceID:0x08, ProductID:84, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 

(3) Thermostat: RC20 (DeviceID:0x17, ProductID:66, Version:01.04) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0xAE 0xAD 

(2) Controller: GB022 (DeviceID:0x09, ProductID:84, Version:01.09)

```

MichaelDvP, 2021-02-01:
@fizista ems-esp ask the boiler (busmaster) for the connected devices. But is seems that the thermostat sometimes is not recognized by the boiler. Here the boiler reports only boiler (0x08) and controller (0x09), no thermostat, other devices. 
> 000+00:01:31.055 T 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBADevices(0x07), data: 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00

@proddy I think there was a similar case some times ago, but can't find it. Do you remember? Maybe this special thermostat does not acknowledge the master-poll, then the boiler reports it only for short time after a thermostat sending.

proddy, 2021-02-01:
@MichaelDvP yes, I remember implementing `devices save` which would remember the detected master Thermostat after a restart. The guy had an RC30 and it wouldn't show up on the bus in a UBADevices telegram. It was quite tricky to program and when I started v2 I decided to skip it. See #323 

This is the 2nd time it's popped up. Maybe we need to build in a fail-safe just for old Thermostats like this product_id 66 RC20 and the product_id 67 RC30 as in the old issue.

MichaelDvP, 2021-02-01:
@proddy i'm thinking about to check unknown devices that sends on the bus. 
If we add to  `EMSESP::process_telegram(` a scan for unknown devices like (marked lines):
```
    bool found       = false;
->  bool knowndevice = false;
    for (const auto & emsdevice : emsdevices) {
        if (emsdevice) {
            if (emsdevice->is_device_id(telegram->src)) {
->              knowndevice = true;
                found       = emsdevice->handle_telegram(telegram);
                // if we correctly processes the telegram follow up with sending it via MQTT if needed
                if (found && Mqtt::connected()) {
                    if ((mqtt_.get_publish_onchange(emsdevice->device_type()) && emsdevice->updated_values()) || telegram->type_id == publish_id_) {
                        if (telegram->type_id == publish_id_) {
                            publish_id_ = 0;
                        }
                        publish_device_values(emsdevice->device_type()); // publish to MQTT if we explicitly have too
                    }
                }
                break;
            }
        }
    }

    if (!found) {
        LOG_DEBUG(F("No telegram type handler found for ID 0x%02X (src 0x%02X)"), telegram->type_id, telegram->src);
        if (watch() == WATCH_UNKNOWN) {
            LOG_NOTICE(pretty_telegram(telegram).c_str());
        }
->      if (!knowndevice && (telegram->src != EMSbus::ems_bus_id()) && (telegram->src != 0x0B) && (telegram->src != 0x0C) && (telegram->src != 0x0D)) {
->         send_read_request(EMSdevice::EMS_TYPE_VERSION, telegram->src);   
->      }
    }
``` 
This should detect the 0x17 on the first sending of the thermostat.

proddy, 2021-02-01:
yes, that would work too! I like it.

MichaelDvP, 2021-02-01:
Tested it, but it needs some finetuning on startup. With tx-delay (for a possible KM200) the 07-telegram is only queued and every incoming boiler telegram floods the tx-queue until we can send it out. But after the startup it works as expected.

proddy, 2021-02-01:
would it help if the read request to 0x07 jumps to the top of the queue?

MichaelDvP, 2021-02-01:
> would it help if the read request to 0x07 jumps to the top of the queue?

No, it's on top of the queue and kicked off when queue is filled. Better to wait until we get the answer to the 0x07 (for us), 
Since the 0x07 is also send regularly it does not matter much, but the way i pushed now is clean.

@fizista please check the new dev build.

fizista, 2021-02-01:
Well done! It works.

```
ems-esp:/$ su
Password: 
000+00:00:26.235 N 0: [shell] su session opened on console
ems-esp:/# log all
Log level = all
000+00:00:31.426 T 1: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 31 64 64 00
000+00:00:31.878 D 2: [mqtt] Publishing topic ems-esp/boiler_data (#14, retain=0, try#1, size 389, pid 15)
000+00:00:31.878 T 3: [mqtt] {"heatingActive":"on","tapwaterActive":"off","selFlowTemp":49,"selBurnPow":100,"curBurnPow":25,"pumpMod":100,"curFlowTemp":48.8,"boilTemp":49.5,"burnGas":"on","heatPump":"on","fanWork":"on","ignWork":"off","heatingActivated":"on","heatingTemp":50,"pumpDel
000+00:00:32.079 D 4: [mqtt] Publishing topic ems-esp/boiler_data_ww (#15, retain=0, try#1, size 418, pid 16)
000+00:00:32.079 T 5: [mqtt] {"wWComfort":"Hot","wWSelTemp":55,"wWSetTemp":55,"wWDisinfectionTemp":0,"wWType":"flow","wWChargeType":"3-way valve","wWCircPump":"off","wWCircPumpMode":"0x3min","wWCirc":"off","wWCurTemp":52.1,"wWCurTemp2":52.1,"wWCurFlow":0,"wwStorageTemp1":52.1,"wwStor
000+00:00:32.280 D 6: [mqtt] Publishing topic ems-esp/thermostat_data (#16, retain=0, try#1, size 85, pid 17)
000+00:00:32.280 T 7: [mqtt] {"hc1":{"seltemp":22.5,"currtemp":22.6,"hatemp":22.6,"mode":"auto","modetype":"day"}}
ems-esp:/# show devices
These EMS devices are currently active:

(1) Boiler: Logamax Plus GB022 (DeviceID:0x08, ProductID:84, Version:01.09)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 

(3) Thermostat: ES72/RC20 (DeviceID:0x17, ProductID:66, Version:01.04) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0xAE 0xAD 

(2) Controller: GB022 (DeviceID:0x09, ProductID:84, Version:01.09)

```

closed 2021-02-01 by fizista