emsesp/EMS-ESP #698: API request for specific HC

PieterBDP, 2021-02-06:
**Is your feature request related to a problem? Please describe.**
I'm using Loxone to query the API and show values in Loxone.
I'm able to filter values using search strings for specific words (e.g. "pumpStatus":) using [ip-adres]/api?device=mixer&cmd=info
But when you have multiple devices (let's say 2 mixers or 2 thermostats) I cannot search for HC2

**Describe the solution you'd like**
[ip-adres]/api?device=mixer&cmd=info&id=1 or hc=1 to query for the first HC
same way it works for setting values

proddy, 2021-02-06:
we'll see about adding id as an optional parameter to the REST API

MichaelDvP, 2021-02-07:
@PieterBDP Some questions: 
Is it only the multiple names? Can we stay with nested json (e.g. `{"hc1":{"pumpStatus":1}}`, or do you also need plain structure (e.g. `{"pumpStatus":1}`? 
Is it ok if the data structure for id=0 gives the whole structure including hcs and id 1.. gives only specific hc? (The thermostat has some values like `datetime` that are independent of hc. 
Do you use esp32 or esp8266? (in esp32 v3.x.x the change is only a few lines of code to set the tag_filter)

PieterBDP, 2021-02-07:
Staying nested is perfect.
I would prefer api?device=mixer&cmd=info gives the whole structure and id=1 gives the specific hc (this way nothing has to change for other users)
I'm using esp8266

proddy, 2021-02-07:
good point Michael. We need to start thinking of putting v2 into maintenance mode (with only fixing critical bugs) and all new features to ESP32's v3. Unless the changes are quite easy to implement in v2.

MichaelDvP, 2021-02-07:
I think this change is quite easy and (most important for 8266) not consuming memory. Testing now and i'll push it today.
Since most people are on 8266 we should decide case by case what is worth and possible for 8266.

@PieterBDP some mixers have also DHW-modes in switch positions 9 and 10, called `wwc1` and `wwc2` in ems-esp, i'll put them to id=9 and id=10 matching the swiitch position.

MichaelDvP, 2021-02-07:
Added to esp8266 and esp32 dev branches.

closed 2021-02-18 by proddy