emsesp/EMS-ESP #699: EMS-ESP "heating temperature" problem to send to the Boiler

MarkRoks, 2021-02-11:
Hello,

For several weeks I have been a user of an EMS-ESP in combination with my Nefit Trendline II GW-5 30kW. I read and control the boiler through this gateway. All this from a link in Node-Red, which in turn communicates with the gateway via MQTT.

Now I have made my own heating curve in Node-Red and I send the desired flow temperature to the boiler. For this I use the command "heatingtemp". This is going well; I see in the gateway the "Heating temperature setting" and then the "Selected flow temperature" both change to this controlled value.

After a while, both values ​​"hang" on a fictitious (last) sent value. Sending the setpoint again results in no change in the displayed value in the gateway and thus to the boiler. The boiler therefore remains at this set flow temperature as long as no drastic measures (restart of the gateway) are taken.

This currently means that in severe night frosts, the boiler will be kept at a low value of somewhere during the day / early evening. But it no longer heats the house well into the night, within a reasonable period of time. This struck me after some extremely cold nights in recent days.

I first looked at Node-Red, but it sends the correct value to MQTT. Then looked at the MQTT in the gateway. Here were a number of MQTT errors, but why? Then looked at the boiler and it correctly anticipates the gateway (it remains at the low value).

The conclusion was that something goes wrong in the gateway. This confirms that after a restart of the gateway it is all working properly again ......

Am I doing something wrong (wrong command to send a desired value), or is the gateway unable to process the number of received telegrams to the boiler and does it get stuck on this (on this one command)? The rest will all arrive neatly and I can also read correctly, namely .......

MichaelDvP, 2021-02-12:
Just a guess (it's to little information about system and issue): A time ago i also had some mqtt disconnects, rebooting ems-esp helps for a few hours than it disconnects again and reconnection failed. At last: After restarting the mqtt-broker the connection was stable.

proddy, 2021-02-12:
@MarkRoks first thing to check is for MQTT fails and Tx fails from the system stats. I would suggest setting up a syslog and tracing the behaviour to see if there are messages being sent that don't get through (or are rejected) by the Boiler. Try and isolate the problem by using something like MQTT Explorer to inject the commands or bypassing MQTT entirely and using the Web API.

MarkRoks, 2021-02-12:
Hello,

Thank you for the advice received.

Indeed, I also think that data transfer from / to MQTT is going wrong. I have just checked the EMS status and the MQTT errors. But both are perfect. EMS data transfer TX = 100% and RX = 100% and no MQTT errors at the moment (as of this morning).

I also (temporarily) replaced the MQTT within Node-Red for the API solution. This may cause a more stable data transfer. I will keep an eye on it in the coming period.

Another question: can anyone explain to me the difference between "selBurnPow", "curBurnPow" and "maxBurnPow" from the received data from the EMS-ESP?

MarkRoks, 2021-02-17:
The EMS-ESP gateway still going wrong, after a while. While I dont't get errors on MQTT and data transfer......

proddy, 2021-02-18:
can you be more specific to what issues you're having?

MarkRoks, 2021-02-18:
Hey,

I let the EMS-ESP gateway communicate with Node-Red through MQTT, both ways. However, after a while the communication stops. At first I assumed that Node-Red was going wrong. However, everything worked again after the EMS-ESP gateway was restarted. In addition, I encounter no errors in the EMS-ESP gateway with the MQTT status and EMS status. This surprised me, as it does seem to go wrong somewhere in the EMS-ESP gateway.

In the meantime, the MQTT has been replaced as much (as possible) by Restful API communication. This is a lot more stable. But because another part is about MQTT, it crashes once a day and only helps a restart.

proddy, 2021-02-18:
the EMS-ESP crashes are restarts?

MarkRoks, 2021-02-18:
Hey,

I have made my own heating curve within Node-Red. For this I use a number of values ​​from the EMS-ESP gateway (outside temperature / room temperature / etc.). Based on this heating curve, data is also sent back from Node-Red to the EMS-ESP gateway (target flow temperature). The latter value therefore changes quite often in a day.

All values ​​were read / controlled by MQTT. However, if receiving / sending via MQTT is no longer possible, for whatever reason, the climate control is confused.

If the EMS-ESP gateway is restarted, everything immediately works correctly again (receiving / sending the necessary data) and the climate control is also correct again.

The communication of the EMS-ESP gateway via MQTT is currently stuck approx. 1x per day. I have to keep a close eye on this, so that the required and controlled values ​​do continue to match.

In the meantime I managed to read a part through the Restful API, instead of MQTT. The control via the Restful API has meanwhile also worked towards the EMS-ESP gateway.

So it seems to go wrong with the MQTT communication from the EMS-ESP gateway.

proddy, 2021-02-23:
I don't think anyone has experience MQTT issues of data getting 'stuck'. I'm not sure what to do with this, without a reproducable case.

MarkRoks, 2021-02-23:
Hello,

I am now using the Rest API and have no problems sending / receiving data through the EMS-ESP gateway. Despite this, I still see MQTT errors in the gateway every day. I still have the MQTT node in Node-Red and it is connected. But no longer sends / receives anything that I use. I could of course remove it completely. However, I need to make sure that the Rest API has taken over all tasks and continues to work stably.

I look at it for a moment.

Regards,

MarkRoks, 2021-02-23:
Hey,

I saw one thing in the EMS-ESP dashboard; When the burner is used to the maximum (100%), the dashboard indicates online 123% of the EMS-ESP gateway.

Since this is usually only the case with hot water, I did not notice it that quickly. Now that I've started paying attention, this happens every time.

Regards,

proddy, 2021-02-24:
ok, for the burner power % we need some raw data to see what is going on.

proddy, 2021-04-17:
@MarkRoks is this still an issue?

MarkRoks, 2021-04-19:
Hello Proddy,

It has been some time since I have observed this. I expect that what now works correctly.

Greetings,

Mark Roks

proddy, 2021-04-19:
great!

closed 2021-04-19 by proddy