emsesp/EMS-ESP #701: Junkers FR100 thermostat no data

davidw3591, 2021-02-14:
**Question**
Today i connected EMS-ESP to my Cerapur boiler with Junkers FR100 thermostat. I'm receiving all information from the boiler but only the time from the thermostat.

Also seeing some "TX read operation failures".

I'm using this level shifter:
[image]

**Screenshots**
[image]

[image]

**Device information**
```
{
  "System": {
    "version": "2.2.0",
    "uptime": "000+00:01:34.369",
    "freemem": 41,
    "fragmem": 14
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 3,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected, instable tx",
    "bus protocol": "HT3",
    "#telegrams received": 47,
    "#read requests sent": 17,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 52,
    "#MQTT publish fails": 0,
    "#dallas sensors": 1
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.13)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FR100 (DeviceID:0x10 ProductID:107, Version:15.04)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x179 0x17A 0x17B 0x17C"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.13)",
      "handlers": ""
    }
  ]
}
```

How to resolve this?

MichaelDvP, 2021-02-14:
Try to find out what causes the tx-errors: `log all`  

If these errors are empty telegrams for 0x179, ff. try to check `read 10 165`, `read 10 166` etc  for hc1, hc2, ..
Maybe this thermostat is mapped to wrong Junkers telegram set.

If these tx-errors caused by something else, try tx_mode 1.

davidw3591, 2021-02-14:
What would I be looking for during "log all"? Here is a short piece:

[image]

The command _read_ is not found, should it be _watch_?
If I run watch 10 165 or watch 10 166 no incoming telegrams are displayed. I monitored for about 5 mins.

MichaelDvP, 2021-02-15:
I see in #335 that FR100, ProductID 107 is correctly supported and telegrams are ok. Also tx_mode 3 is ok. 

@davidw3591 for commands please read [https://emsesp.github.io/docs/#/Console](https://emsesp.github.io/docs/#/Console)

The log is much to short to see something going wrong, it should be min. 2 minutes of normal operation to fetch the the querys repeated. Please log to text-file, that makes it easier to read and search.

This small part seems to be not in normal operation of the heating system, maybe initialisation phase. The thermostat does not broadcast anything and queries version information from boiler. The boiler broadcasts types 16 and 19, but does not reply for this telegrams, that's not normal.

Also in first post it is only 1,5 minutes uptime and 47 telegrams received (14 requests send without error), that's to short to read all informations from ems. Maybe the thermostat information comes up a few minutes later.

bbqkees, 2021-02-15:
@davidw3591 Nice to see that one of my very first and hand soldered EMS interface boards is still working well.

davidw3591, 2021-02-15:
Hi,

Here is a few minutes of _log all_ output: https://pastebin.com/1KsRvZ3f

Here is the output of the read 10 165/166 commands: https://pastebin.com/W14jSbjP

MichaelDvP, 2021-02-15:
Thanks, the boiler does not reply to telegrams 14, 16, 19, but telegram 33 always gives value back, there is nothing wrong with tx. 
I think this will not hurt, 19 is broadcasted by the boiler, we have this information. 14 is only boiler-uptime and 16 give some controller-settings, maybe the buildin controller do not have these settings.

The thermostat sends ww and boiler values to the boiler, but there is no information for a heating circuit sended. Are you sure that a heating circuit is configured? 
It seems there is no other information on the bus than displayed by ems-esp. With no heating circuit configured this thermostat will only show the time.

davidw3591, 2021-02-15:
What should I look at in the manual?
[FR100 manual](https://www.manualslib.com/products/Bosch-Fr-100-4100369.html)

 My thermostat and boiler are working without issues with each other...

MichaelDvP, 2021-02-16:
If thermostat is working, you should look on the thermostat display and check what informations are on the thermostat, but missing in ems-esp. In the log there is no outdoortemperature, no roomtemperature and no (roomtemp)setpoint, only warm water and boiler settings.

proddy, 2021-03-11:
@davidw3591 did you make any progress on this? Anything we can help with?

davidw3591, 2021-03-15:
> @davidw3591 did you make any progress on this? Anything we can help with?

Hi, I'm currently in the process of a heavy renovation. Will pick up on this is a few weeks.

closed 2021-10-04 by proddy