emsesp/EMS-ESP #702: No pre-build binaries for ESP32 and when compiling from source dev version is  v3.0.0b1

tp1de, 2021-02-15:
I am a bit confused. Since I want to rewrite my node-red flows reading mqtt-messages a was looking for the actual dev version.
I could not find a pre-build binary for the esp32 like before. Any reason for this?

Therefore I dowloaded the respective source and compiled by myself.
I ended with version v3.0.0b1. Is this ok?

This binary seems to work -  I just lost my WLAN-connection. WLAN has to be setup new. All other settings where kept.
Just the mqtt messages changed in structure and data-field names. This is not very convienient. (e.g. mixer valveStatus into status).

proddy, 2021-02-15:
I removed the esp32 binaries as a few people were grabbing the wrong binary for their ESP8266 gateways and bricking the devices.

v3 is the ESP32 version, so all good there. I added Ethernet support so had to shuffle things around, like renaming WiFi to Network.

See https://github.com/proddy/EMS-ESP/discussions/661

tp1de, 2021-02-15:
@proddy thanks for the reply. I haven't seen #661
How would Ethernet work on the esp?

bbqkees, 2021-02-15:
For Ethernet you have to use an additonal LAN board or ESP32 development board with a LAN8720 or TLK110 Ethernet PHY chip.
There is f.i. this very cheap Waveshare 8720 clone board. You need to modify it to work though.
One that works well out of the box is the Olimex ESP32 Gateway dev board.

(A BBQKees Ethernet Gateway is in development and expected somewhere end of Q2.)

tp1de, 2021-02-16:
@bbqkees Thanks Kees. We can close the issue.

closed 2021-02-16 by tp1de