emsesp/EMS-ESP #703: Warm water circulation pump frequency - can not be changed (neither API nor MQTT)

tp1de, 2021-02-16:
I tried to change the ww circulation pump frequency - but without success. 

On the thermostat I can change wwcircmode to  of,on,auto,own by web api or mqqt - both are working fine

On the boiler the changing of wwcirmode to values between 1..7 does not work. (API returns invalid cmd)
Maybe the changes should be send to the thermostat? Any idea how to bring it to work?

tp1de, 2021-02-16:
@MichaelDvP I did the code change but it is still not working ....

MichaelDvP, 2021-02-16:
can you check with commands and `read 8 33` the values, for me it's working:
```
ems-esp:/# read 8 33
000+00:02:55.209 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 46 00 00
ems-esp:/# call boiler wwcircmode 5
000+00:03:11.391 I 2: [boiler] Setting warm water circulation mode 5x3min
ems-esp:/# read 8 33
000+00:03:20.269 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 05 46 00 00
ems-esp:/# call boiler wwcircmode 7
000+00:03:40.364 I 4: [boiler] Setting warm water circulation mode continuos
ems-esp:/# read 8 33
000+00:03:46.427 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 46 00 00
```

tp1de, 2021-02-16:
No data:

[image]

tp1de, 2021-02-16:
I changed bus idto service key an tried again:  

[image]

MichaelDvP, 2021-02-17:
Ok, i think i've found it, please try again.

tp1de, 2021-02-17:
@MichaelDvP 
It is working now - Super.

By the way one question: 
What is the difference between the datafields wWCirc and wWCircPump in boiler_data_ww?

proddy, 2021-02-17:
@tp1de great! There's a brief description of the datafields in https://emsesp.github.io/docs/#/MQTT?id=thermostat

and thanks @MichaelDvP for the quick fix

tp1de, 2021-02-17:
The definition:
wWCircPump = Warm water circulation pump available
wWCirc = Warm water circulation active

But in my case both values are the same and indicating when circulation pump is running. 
[image]

MichaelDvP, 2021-02-17:
I think telegram 0xE9, boiler.cpp line 497 should be the status,`wWCirc` not the configuration. @tp1de Can you check that?
(wWCirc is also in E5, maybe no need to read it in E9).

tp1de, 2021-02-17:
@MichaelDvP 

> I think telegram 0xE9, boiler.cpp line 497 should be the status,`wWCirc` not the configuration. @tp1de Can you check that?

Yes I can confirm - changing to wWCirc keeps wWCircPump constant on 1

> (wWCirc is also in E5, maybe no need to read it in E9).

Yes I can confirm too - for me no need to read wWCirc in E9 - for you @MichaelDvP  too?

MichaelDvP, 2021-02-17:
E5 is the slow monitor sended periodically, maybe E9 status is triggered on change and more accurate. Can you check this with logging while wwCirc is triggered?

I have a ems-boiler and circulation is in telegrams 33 and 18. The boiler do not send 0xE telegrams and i can not test.

tp1de, 2021-02-17:
I can't see any difference in MQTT Explorer:
1. first change to wWcirc
2. take message out in E9
[image]

MichaelDvP, 2021-02-18:
I've checked older logs: E9 is broadcasted completely every 10 sec, E5 is only sended out in parts whenever something changed. I think we should read both to have the current state (E9) and the change (E5). 
Fixed in dev8266 and dev32.

commit f17aa09b (2021-02-16): fix #703
commit dc4e5fd1 (2021-02-16): fix #703
commit 9046a323 (2021-02-17): wwcircmode for ems+, 2.fix #703
closed 2021-02-18 by MichaelDvP