emsesp/EMS-ESP #704: not possible to change Thermostat Temp and Mode via mqtt because of wrong detected HC

SpaceTeddy, 2021-02-17:
**Bug description**
not able to set thermostat temperature and Mode via mqtt. Seems that ems-esp has detected an additional heating circuit (hc2), which is not existing. Wrong detected HC2 has fixed temp data of 40°C. HC2 and HC1 mqtt messages are sent to the mqtt serer. Rebooting the EMS-ESP fixes that problem and only HC1 is detected. I guess, that this wrong detection happens if my RC20RF thermostat has restarted or an other error occours with the room controller.  The RC20RF still "sents" temp and mode data via HC1. 

**Steps to reproduce**
- sent mqtt command to HC1 -> ems-esp shows error, that the command is not accepted
- check show comand in telnet console -> aditonal HC2 detected
- HC2 Temp has fixed temp value of 40°C -> this happens if my RC20RF room controller has error A21 806. 
- I guess that ems-esp gateway detects a new HC if RC20Rf is recovered
- EMS-ESP Vers 2.2.0 

**Expected behavior**
- ems-esp shall not detect a new HC if additional thermostat devices has errors or are rebooted.

**Screenshots**

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=info*

**Additional context**
if error occours again, I will do an show devices command

proddy, 2021-02-17:
@MichaelDvP easiest option I think is to add some debug statements to `Thermostat::heating_circuit(std::shared_ptr<const Telegram> telegram)` to see which incoming telegram is causing the heating circuit to be created?

MichaelDvP, 2021-02-17:
@proddy I think the telegrams are clear: 0x48 (hc2-monitor) gives setpoint, roomtemp, modetype and targetflow, 0x49 the timer program. 
But why is mode for hc1 not accepted? It seems hc1 is registered and the error-message indicates that that hc1-setpoint is not set (hc1 not active). I would like to see  what is in telegram 0x3E (hc1-monitor). I think if a hc is registered it can not become inactive while ems-esp is running. 

@SpaceTeddy Is the mode command always not working or only if RC20RF have errors?

BTW: A21-806 indicates a broken roomtemperature sensor in hc1.

SpaceTeddy, 2021-02-17:
@MichaelDvP 
the A21-806 appears at every power cycle from the boiler. Means, if I switch the Power off/on, the A2-806 is mentioned in the servicecode and servicecodenumber. A2(1)-806 means heating circuit 1 A22-806 heating Circuit 2 ....
Usually the RC20RF provides all Temp/mode informations via HC1 mqtt topic, even if the HC2 is detected. It seems, that the ems-esp can not receive/parse the incomming mqtt command correctly. 
I try to force the failure at home today and hopefully I can provide more informations to you.

MichaelDvP, 2021-02-17:
Afaik the RFM20 have two channels, maybe on powerup it sends data for both channels and the bus receives both telegrams. 
The RFM normaly sends only the remote messages 0xAF, and the master thermostat repeats the values in the hc specific messages. Ems-esp only registers a hc on the master-messages, not the remote messages. 

Can power ems-esp external and `log all` while doing a power cycle on the boiler.  
For the mode command, if hc1 is registered by ems-esp with a valid setpoint (values are displayed), the message " heating circuit 1 not found or activated" should not come.

proddy, 2021-03-11:
@SpaceTeddy did the try the suggestion from Michael?

SpaceTeddy, 2021-03-11:
Not really, because my idea was to check the long term behavior first. Right now, my system is running for 20 days without the shutdown the boiler itself. I guess that the wrong detected HC was init by hot plugging the service key before switching off the boiler. 
You can close it and I it occurred again, I may reopen it.

Greets,
Chris

closed 2021-03-11 by proddy