emsesp/EMS-ESP #705: Reconnection to mqtt server when connection lost

tp1de, 2021-02-17:
When mqqt server connection is lost (e.g. while mqqt server is restarted after sw-update) then ems-esp does not automatically reconnect. This has to be done manually.

I suggest that when connection is lost, every minute a reconnection try should be done.

proddy, 2021-02-17:
I'm pretty sure it does reconnect, but I'll recheck...

tp1de, 2021-02-18:
Hi @proddy,
I tested again and I can comfirm that the mqtt reconnection works if I shutdown the mqtt broker/server for some minutes.
Yesterday I updated the server and mqtt broker was down for approx. 40 minutes. Other mqtt clients reconnected afterwards but ems-esp does not. So I opened this issue.

I will further observe but in the meantime I will close this issue.

closed 2021-02-18 by tp1de