emsesp/EMS-ESP #706: Using REST api for retrieving boiler information with PHP script

flohse123, 2021-02-18:
Hi,
I am trying to retrieve boiler information via REST api with a PHP script from a remote server. http://<myip>/api?device=boiler&cmd=info. It is working perfectly the typing this to my phone browser. But the PHP script is ending with error 500. Do I need to send login or password? How can I do so with PHP?

Here is the code I am using:

`<?php
    $_url = "http://<myIP>/api?device=thermostat&cmd=info";
    $ThermoInfo = file_get_contents($_url);
    echo $ThermoInfo;
?>`

Would love to get a hint because I am stuck here.

proddy, 2021-02-18:
does it work via the browser directly? I'm guessing so.

You don't need any credentials, as long as the setting 'Allow WEB API to write commands' is enabled in the web's Settings page it should work.

flohse123, 2021-02-18:
Ok, thanks. Then it must be my FRITZ!box blocking the script.

flohse123, 2021-02-19:
For all looking for a solution to the same problem: Use cURL:
This code did not work: 
`<?php    $_url = "http://<myIP>/api?device=thermostat&cmd=info";    
$_buffer = implode('', file($_url));    
echo $_buffer;
?>`
This worked like a charm. And it is an example for using cURL with less than 120 lines of code.
`<?php
$ch = curl_init("http://80.137.185.84/api?device=thermostat&cmd=info");
curl_exec($ch);
curl_close($ch);
?>`

proddy, 2021-02-19:
nice, I'll add to the wiki

proddy, 2021-02-21:
https://emsesp.github.io/docs/#/Howto?id=_2-retrieving-data-via-rest-using-php-by-flohse123

closed 2021-02-21 by proddy