emsesp/EMS-ESP #708: reset function to revert to factory settings

proddy, 2021-02-20:
New feature for both v2 (ESP8266) and v3 (ESP32), holding a specific button doing for 5-10 seconds will erase the filesystem and force EMS-ESP to load the factory settings. This can be used in a panic situation when you can't get to the Serial or Web interface.

I'm looking at using the Reset button that comes standard on the ESP8266 and ESP32 dev boards and using either EEPROM/RTC of FS to store the state.

MichaelDvP, 2021-02-21:
Some first thoughts to this:

- For 8266 very usefull if wifi does not connect and AP is set to never. But the reset button is wired to RST-line and i dont think time measurement is possible. Also i dont see a benefit in erasing all data, activating the AP is imho sufficient. 
Something like (untested):
```
void APSettingsService::begin() {
    _fsPersistence.readFromFS();
    struct rst_info *rtc_info = system_get_rst_info();
    if (rtc_info->reason == REASON_EXT_SYS_RST && _state.provisionMode == AP_MODE_NEVER) {
        _state.provisionMode = AP_MODE_DISCONNECTED;
    }
    reconfigureAP();
}
``` 
Or have you something in mind that may crash the esp on startup (maybe io-settings)?

- For esp32 the serial console should always give a fallback to reconfigure, i see no need for the panic button.
Also esp32 does not have a reset-line, the button will trigger always a power-on-reset.

proddy, 2021-02-21:
It seems possible to detect the times between hard resets. I've been looking at this library https://github.com/khoih-prog/ESP_DoubleResetDetector. Also on the ESP32 been playing with the NVM (https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences) as an alternative to persisting in the FS.

Good point about only re-enabling the AP. There have been some people that change the admin password and can't get back in so would also be useful to reset the admin and su passwords.

bbqkees, 2021-02-21:
I get questions about lost passwords etc. quite regularly. Now the only way to restore it is to reupload the firmware.
Not an easy task if you never did something like that before and if you are not that good with computers.
Just enabling the AP does not solve the problem if you can't remember the password, so a full factory reset is more useful.
After a factory reset the Gateway is setup again pretty quickly anyway.
A reset button connected to a GPIO would be a great feature that everybody understands. Press it on boot or f.i. for 10 seconds and then restore to factory mode so people can start over without having to dig into esptool etc.
For boards that do not yet have a button you could use a piece of wire to connect the GPIO to ground. 
Detecting how long the boot button is pressed is also a good option if that is possible at all.

For the ESP32 serial fallback may indeed be there but a reset button is still the most friendly option.
Also I'm working on some new board ideas and those may not all have an on-board USB connector anymore.

proddy, 2021-02-28:
started working on this today. I created a new button library which is a bit over-engineered. You can assign any pin and it will detect button pushes from short, long, very long and a sequence. The idea is that a long press will reset the EMS-ESP to factory settings. We can use sequence and double-tap to fire off special commands. Probably not. But it was fun to program. I'll check it in shortly.

proddy, 2021-03-01:
Committed first version of the button implementation. After consulting BBQKees it's a pull-up button (with a 10k resistor to 3.3v) so GPIO is HIGH when the switch is open and LOW (GND) when pushed. The pin is configurable, called External Button in the WebUI. On boot EMS-ESP will detect for an attached button. 

Button does some tricks
 - single press - does nothing yet. Any ideas?
 - double click - will reconnect the wifi
 - long press  between up to 3 seconds - does nothing yet
 - very long press, up to about 9 seconds will do a factory reset, essentially erasing all the config files and restarting

MichaelDvP, 2021-03-02:
Some remarks/questions to this function:
Why not use the internal weak-pullups if the gpio have one also on esp32?
Gpio 0 is not selectable (0 = off). On the NodeMCU-32s (that bbqkees uses in new design) there is the `boot` button on gpio 0 with internal pullup, [see here](https://robu.in/wp-content/uploads/2019/12/Ai-Thinker-NodeMCU-32S-ESP32-Development-Board.pdf) Should be nice to use it. But then you should trigger the vLongButtonPress on button release to make sure the button is not pressed on reboot. 
The syslog should be started before button function to get the starting message (system.cpp, line 195..)

proddy, 2021-03-02:
thanks Michael for the quick feedback
- I'm using external pull-ups because this is how BBQKees designed it on his newer boards with a 10k resistor (@bbqkees   ?)
- gpio 0 to off not selectable, thanks - I'll fix that. Probably the same with the dallas gpio since I copied the code?
- actually the GPIO list should be pull-down to exclude non input GPIOs. So an ESP32 excluding 0, 1, 6-11, 12, 14 & 15 (taken from [here](https://diyprojects.io/esp32-how-to-use-gpio-digital-io-arduino-code/#.YD4Qv2hKhjF). What do you think?
- I recommended to bbqkees to remove the boot/reset EN button. If we ever move away from the ESP32 dev kits to use native ESP-WROOM-32 chips we'll need a button.
- syslog started before button - good catch, I'll make a fix

bbqkees, 2021-03-02:
Just to clarify things:

There is a difference between a reboot and an reset button.
The reboot button is connected to the EN/RST pin of the ESP chip.
Pressing it will reboot the ESP.

The Reset button is linked to a GPIO pin, and it can have the function we give it.

- Using the ESP32 software pull ups is fine. On the Ethernet prototype version I'm working on there is a button tied to GPIO33 that is pulled high with an external resistor on the board because basically a footprint for a resistor is 'free' if you have the room. Can always leave it unpopulated.
The button is connected to ground so when you press it, the GPIO goes low.
Having an external resistor makes sure that when you forget or misconfigure the software pullup, that the GPIO is never floating and remains high at all times, thus never triggering the reset by accident.

- For existing boards without the button we need the software pullups anyway. You have to use a wire to connect the pin to ground.

- The NodeMCU 32S board indeed has a button tied to GPIO0. Good idea to check its usefulness.
You can use it for EMS-ESP 'long-press to reset' but not if you press it at boot (bootloader/firmware mode) or when you connect an Ethernet module because GPIO0 is also used for resetting the Ethernet PHY chip or a clock signal or something I believe. (Can't remember exactly).

- On the Ethernet prototype version I have both a REBOOT button linked to EN and a RESET button tied to GPIO33.
The REBOOT button is just for testing/development purposes, when you are playing around with the firmware. 
The RESET button is intended for the end users to trigger the reset to factory mode.

- On the ESP8266 Gateways like the Premium II, I introduced the external REBOOT button (tied to EN) at some point because if you were using the internal screw connector and bus powered mode, there was no way to reboot the Gateway without unscrewing the whole case. 
With the pluggable screw connector on the latest revisions you can just unplug the Gateway to restart it so the REBOOT button is not that useful anymore on those boards.
That's also why there is no external reboot button on the Gateway S32. If you are testing/developing, you can use the button on the NodeMCU board itself (and I ran out of board space for the external button as well).

proddy, 2021-03-02:
right, I'll add the s/w pullup. 

Also - should I port this to v2.x (ESP8266) too? The only advantage I see is for people using older gateways boards that are bricked and who are brave enough to unscrew the box and use a fiddly piece of bare wire on a pin for 10 seconds without breathing ;-)

bbqkees, 2021-03-02:
As currently 95% of users will still be on the ESP8266 backporting would be nice.

MichaelDvP, 2021-03-02:
Suggestion for the esp8266: 
D3 (gpio0) have an external pullup (10k), D2 (gpio4) is beside and both are also on J18 on the Gateway. 
Use gpio0 fixed for the PButton and set gpio4 to output-low, Then the user can shorten these pins with a small screwdriver or pincers and don't have to fiddle with bare wire.
Also on8266 gpio0 is used for bootmode, it has to be high on esp-start. The PButton has to trigger the erase/reboot on button release: Remove PButton.cpp line 162 and insert in 146: 
```
    if (state_ == pullMode_ && vLongPressHappened_) {
        resultEvent         = 4;
        vLongPressHappened_ = false;
        longPressHappened_  = false;
    }
```

proddy, 2021-03-02:
I couldn't find J18 on my ESP8266 Gateway v1.6 but I love the idea! And happy if you want to go ahead and implement this if you have time. The tasks related to this were:
- syslog service starting earlier before button
- change line 60 in PButton::init() to `pinMode(pin_, pullMode ? INPUT_PULLUP : INPUT_PULLDOWN);` so it doesn't matter if there is an external pull-up resistor
- check if it works on the ESP8266. I never tested it!
- check if it still works on the ESP32 with your change above. If not use #ifdef ESP8266 ....

I'll work on the Web UI because I'm there anyway with #690 adding Board Profiles

MichaelDvP, 2021-03-02:
J18 is only on Gateways <1.6, [here](https://bbqkees-electronics.nl/wiki/gateway/hardware-details.html), but the side-by-side pins can be accessed on the soldering on the wemos with a pen. I've tried this:
[image]
Result (i can confirm that your PButton works on 8266).
[image]

I'll merge in esp8266_dev later.

proddy, 2021-03-02:
awesome! love quick hacks like this.

MichaelDvP, 2021-03-02:
I see that there is PR#719. Can you please check/merge this first.

proddy, 2021-03-02:
> I see that there is PR#719. Can you please check/merge this first.

done.

bbqkees, 2021-03-03:
Nice work. thanks.

proddy, 2021-03-04:
Michael cleaned up my code and ported to ESP8266 (thanks!) so I think we can close this now. If anyone has any good ideas of what functions to put in the button let me know. Right now, double-click will just reset the wifi

closed 2021-03-04 by proddy