emsesp/EMS-ESP #709: spikes on HA records

florin-laurentiu, 2021-02-21:
hello,

in HA there are every day some spikes on graphic recording of temperature, see attached picture.
i remember that it was in the past something similar , in the temperature recording of negative values if i good remember....
[image]

laurentiu

proddy, 2021-02-21:
this is from the dallas sensors? Which version are you using?

florin-laurentiu, 2021-02-21:
hi,

i use the last development release; it is coming from boiler
sensor: sensor.boiler_burner_current_power

proddy, 2021-02-22:
that's a %, not a temperature. Can you find what the actual value is? It comes from the 0x18 telegram (or 0xE4 for EMS+)

florin-laurentiu, 2021-02-22:
Hello,

sure is not a temperature.....as is boiler current power...
I m not sure from where is coming...please advice what to check.

laurentiu

proddy, 2021-02-23:
like I said, try and get some real values from the actual telegrams. Just showing a screenshot is not very helpful. You can do this using `watch 0x18`  or using syslog to record say 1hr of data.  Maybe there is a reason its spiking?

I'll also measure here to see if I can match it.

closed 2021-10-04 by proddy