emsesp/EMS-ESP #710: MQTT heartbeat not updated.

rkuijer, 2021-02-21:
**Bug description**
Currently using version 2.2.1b3 to verify the solution of issue #369, which is OK. Thanks for that!
During the first tests, I saw that the heartbeat info to MQTT isn't updated. The first indication was that there was a mismatch between the amount of MQTT publish fails, 2 on the AMS-ESP and 1 on the MQTT broker. When I looked at the time stamp came to the conclusion that it's only updated during bootup.

**Steps to reproduce**
No special steps are needed just boot the EMS-ESP and connect to an MQTT broker with Home Assistant Format.
Clean session and retrain flag are checked.
When you watch the MQTT broker you will see that the heartbeat info won't be updated.

**Expected behavior**
Interval update of the heartbeat information.

**Screenshots**
[image]

**Device information**
Using a Gateway Premium II (P2)
```json
{
  "System": {
    "version": "2.2.1b3",
    "uptime": "000+01:21:23.524",
    "freemem": 41,
    "fragmem": 12
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 2513,
    "#read requests sent": 664,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 2,
    "#dallas sensors": 1,
    "#dallas fails": 1
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Topline/GB162 (DeviceID:0x08, ProductID:115, Version:03.06)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC20/Moduline 300 (DeviceID:0x17, ProductID:77, Version:03.01)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x91 0xA8"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:114, Version:01.03)",
      "handlers": ""
    }
  ]
}
```

proddy, 2021-02-21:
Thanks for reporting this. We'll look into it.

closed 2021-02-22 by proddy

rkuijer, 2021-02-22:
@proddy, yep it works again.
Thanks!