emsesp/EMS-ESP #711: Missing temperature readouts for 1st mixer circuit (out of 2 circuits) of IPM (Junkers)

TomoKRK, 2021-02-22:
**Bug description**
For 1st mixer circuit IPM (out of 2 installed) there are missing values for the current and set temperature. The only information
provided is the current pump status. I think this issue is/might be related to #595.
 
   I have two heating circuits:
hc1 - it's non-mixing for heaters/radiators circuit
hc2 - it's mixing (with valve) for floor heating

The missing values are available/displayed on FW200, however are missing in EMS output.

Output from "show values":
Mixer: Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.07)
  (hc 1) Current pump status: on

Mixer: Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.07)
  (hc 2) Current flow temperature: 24.8 °C
  (hc 2) Setpoint flow temperature: 26 °C
  (hc 2) Current pump status: on
  (hc 2) Valve status: 0 %

JSON/MQTT message:
{
  "hc1": {
    "pumpStatus": "on"
  },
  "hc2": {
    **"flowTemp": 24.8,
    "flowSetTemp": 26,**
    "pumpStatus": "on",
    "valveStatus": 0
  }
}

**Expected behavior**
The expected is to provide readouts of **flowTemp** and **flowSetTemp** for hc1 too as it's provided for hc2.

**Device information**
{
  "System": {
    "version": "2.2.0",
    "uptime": "000+00:00:06.593",
    "freemem": 57,
    "fragmem": 1
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 2,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 3,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 2,
    "#read requests sent": 0,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": []
}

**List of devices installed:**
(1) Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:18.08)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 

(3) Thermostat: Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.12) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168 

(4) Mixer: Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.07)
 This Mixer will respond to telegram type IDs: 0x10C 

(5) Mixer: Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.07)
 This Mixer will respond to telegram type IDs: 0x10C 

(2) Controller: HT3 (DeviceID:0x09, ProductID:95, Version:18.08)

TomoKRK, 2021-02-22:
Hello, first, I wanted to thank everyone for the great work on this EMS ESP interface and code. I am enjoying it for several days :)
I know it's not great to comment my own post, however after the quick look at the source code - specifically
mixer.cpp and at line 284  - there is the method for reading out the IPM status void Mixer::process_IPMStatusMessage,
looks like there is a need to change the below if statement:

// do we have a mixed circuit
    if (ismixed == 2) {
        changed_ |= telegram->read_value(flowTemp_, 3); // is * 10
        changed_ |= telegram->read_value(flowSetTemp_, 5);
        changed_ |= telegram->read_value(status_, 2); // valve status
    }
    changed_ |= telegram->read_bitvalue(pumpStatus_, 1, 0); // pump is also in unmixed circuits
to:

    if (ismixed == 2) {
        changed_ |= telegram->read_value(status_, 2); // valve status
    }

    changed_ |= telegram->read_value(flowTemp_, 3); // is * 10
    changed_ |= telegram->read_value(flowSetTemp_, 5);
    changed_ |= telegram->read_bitvalue(pumpStatus_, 1, 0); // pump is also in unmixed circuits

so that the temp readings were read too for unmixed circuits as it was done for #595 with pump status.

Hope that helps in narrowing it down, and I would appreciate the code fix for it. I would/could have done it myslef, but I am not experienced at all with ESP like environments.

proddy, 2021-02-22:
@TomoKRK thanks and nice find. Yes your solution makes sense and is line with how @MichaelDvP fixed the pump status. If your change works then go ahead and put in a PR if you'd prefer me or Michael can make the fix.

TomoKRK, 2021-02-22:
@proddy - thank you for confirmation. I am not sure if that works for 100% as I am not able to test it. I have come to the above conclusion based on the quick code analysis. I think through the analogy to #595 that it will work, however I have no IDE setup and I am not familiar with development environment for ESP aa all. So if you or @MichaelDvP could put the fix in I would appreciate and then I am happy to test it out on my ESP EMS module once it's avail in some beta build.

MichaelDvP, 2021-02-23:
@TomoKRK Can you `watch` or `read` some of the telegrams `10C` from unmixed circuit?
We have only the example in mixer.cpp, line 299 where all temperatures are zero. That seems logical to me: flowtemp and settemp are set by boiler and the IPM can not change the temperature, it can only pump the given boiler-water to the circuit. There is no need to publish temperature values, the boiler do the publish. But these telegrams were fetched in summer and maybe only in summermode the temps are zero and in winter the mixer republishes the boiler values. We have to check.

TomoKRK, 2021-02-23:
@MichaelDvP - please find quick log and I hope it's what you expected let me know if that helps

ems-esp:/# read 20 10C
000+06:27:19.738 N 139: [telegram] Rx: A0 00 FF 00 00 0C 01 01 00 00 00 27 63
000+06:27:20.621 N 140: [telegram] Rx: A0 08 1A 00 27 F5
000+06:27:20.652 N 141: [telegram] Rx: A0 08 1A 02 00 D6
000+06:27:20.939 N 142: [telegram] Rx: A0 0B FF 00 00 0C 01 01 00 00 00 27 C4
000+06:27:20.939 N 143: [emsesp] Mixer(0x20) -> Me(0x0B), IPMSetMessage(0x10C), data: 01 01 00 00 00 27
000+06:27:22.339 N 144: [telegram] Rx: A0 08 1E 00 01 87 18

MichaelDvP, 2021-02-23:
Thanks, there is a flowSetTemp of 39°C, but flowTemp and valve is zero. I'll add the flowSetTemp to to unmixed circuit.

TomoKRK, 2021-02-23:
@MichaelDvP - thank you, and you were correct the flow temp for this circuit comes from boiler. The confusion on my end was coming from the fact that for circuit 1 on the display of FW200 I have both values set and flow for this. 
   However after further comparison of the current readouts from EMS interface, and comparing it to the values displayed on the thermostat FW200 - it turned out that the circuit 1 current flow temperature is in fact "switchTemp" data from JSON message. 
  Additionally to the display of this value in the section of Info->Heating Circuit 1 on FW200, the same value is repeated in the Info section for boiler as "External Flow Temperature".
  So I think that concludes this topic. So thank you for the fix and providing the read out of the flowSetTemp and I would like to apologize for any confusion from my end. As an excuse I will just say that I just started learning :)

MichaelDvP, 2021-02-23:
No need to apologize, that's the way reverse engineering works, we have now one more known value, that's good. Feel free to ask again if you think there is something wrong or missing.

Should we close this?

TomoKRK, 2021-02-23:
yes, we can close it out.

closed 2021-02-23 by MichaelDvP