emsesp/EMS-ESP #712: boiler temperatures not rendered correctly

proddy, 2021-02-22:
got the following values which were discussed above:
```
ems-esp:/# show values
Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:34.12)
  Warm water mix temperature: 430 °C
  Warm water buffer temperature: 523 °C
```
with a factor of 10 off. The CW400 shows them correctly as Mischertemperatur und Pufferspeichertemperatur (see picture of CW400 Pufferspeichertemperature). So these are not warm water buffer temperatures, but an externally connected general buffer storage. The HT3 board layout and connected sensors also show it as non-warm water (sensor  26&32).
Now since @suysh seems to have them encoded properly w/o factor 10, I'm wondering how to get this working correctly?

[image]

[image]

_Originally posted by @joanwa in https://github.com/proddy/EMS-ESP/issues/397#issuecomment-783463776_

suysh, 2021-02-22:
Hmm, strange. On my site it's correct. What version do you use?
[image]
[image]

suysh, 2021-02-22:
Aha, now I see. It's not the solar values, but the Warm water boiler temperature. See also in my values it's wrong.
I recently changed from OpenHAB2.5 to OpenHAB3 and updated also EMS-ESP and are reconfiguring everything. Didn't find this error yet.
```
ems-esp:/$ show values
Boiler: Junkers Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208,                               Version:01.04)
  Heating active: 0
  Warm water/DHW active: 0
  Service code:
  Service code number: 204
  Warm water selected temperature: 50 °C
  Warm water set temperature: 53 °C
  Warm water disinfection temperature: 70 °C
  Selected flow temperature: 60 °C
  Burner selected max power: 11 %
  Burner current power: 0 %
  Pump modulation: 53 %
  Warm water type: layered buffer
  Warm water charging type: 3-way valve
  Warm water circulation pump available: 0
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: 0
  Outside temperature: 16.0 °C
  Warm water current temperature (intern): 38.0 °C
  Warm water current temperature (extern): 51.0 °C
  Warm water current tap water flow: 0.0 l/min
  Current flow temperature: 55.0 °C
  Return temperature: 52.9 °C
  System pressure: 1.7
  Warm water storage temperature (intern): 38.0 °C
  Warm water storage temperature (extern): 51.0 °C
  Warm water activated: 1
  Warm water one time charging: 0
  Warm water disinfecting: 0
  Warm water charging: 0
  Warm water recharging: 0
  Warm water temperature ok: 1
  Warm water activated: 0
  Gas: 0
  Flame current: 0.0 uA
  Boiler pump: 1
  Fan: 0
  Ignition: 0
  Warm water heating: 0
  Heating activated: 1
  Heating temperature setting on the boiler: 82 °C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler circuit pump delay time: 3 min
  Boiler burner min period: 10 min
  Boiler burner min power: 0 %
  Boiler burner max power: 100 %
  Boiler temperature hysteresis on: -6 °C
  Boiler temperature hysteresis off: 6 °C
  **Warm water mix temperature: 582 °C
  Warm water buffer boiler temperature: 562 °C**
  Warm water # starts: 968
  Warm water active time: 9868
  Burner # starts: 4252
  Warm Water active time: 6 days 20 hours 28 minutes
  Total burner operating time: 61 days 19 hours 48 minutes
  Total heat operating time: 54 days 23 hours 20 minutes
  Total UBA working time: 657 days 19 hours 12 minutes

Thermostat: RC300/RC310/Moduline 3000/CW400/Sense II (DeviceID:0x10, ProductID:1                              58, Version:18.06)
  Time: 17:04:46 22/02/2021
  Error code: (0)
  Warm water mode: 2
  Warm water high temperature: 70
  Warm water low temperature: 30
  Heating circuit 1:
    Setpoint room temperature: 24.0 °C
    Current room temperature: 22.7 °C
    Comfort temperature: 14.0 °C
    Eco temperature: 10.0 °C
    Manual temperature: 24.0 °C
    Target flow temperature: 60 °C
    Summer temperature: 25 °C
    Summer mode: 1 °C
    Mode: manual
    Mode type: comfort
  Heating circuit 2:
    Setpoint room temperature: 20.0 °C
    Current room temperature: 22.7 °C
    Comfort temperature: 24.0 °C
    Eco temperature: 15.0 °C
    Manual temperature: 20.0 °C
    Target flow temperature: 0 °C
    Summer temperature: 17 °C
    Summer mode: 1 °C
    Mode: manual
    Mode type: comfort

Mixing Module: MM100 (DeviceID:0x20, ProductID:160, Version:24.05)
  Heating circuit 1:
    Setpoint flow temperature: 60 °C
    Current pump status: 0
    Valve status: 0 %

Mixing Module: MM100 (DeviceID:0x21, ProductID:160, Version:24.05)
  Heating circuit 2:
    Current flow temperature: 28.7 °C
    Setpoint flow temperature: 0 °C
    Current pump status: 0
    Valve status: 0 %

Solar Module: SM100/MS100 (DeviceID:0x30, ProductID:163, Version:27.05)
  Collector temperature (TS1): 36.5 °C
  Bottom temperature (TS2): 51.5 °C
  Solar pump modulation (PS1): 0 %
  Cylinder pump modulation (PS5): 0 %
  Valve status: 0
  Solar pump (PS1) active: 0
  Tank heated: 0
  Collector shutdown: 0
  Energy last hour: 0.0 Wh
  Energy today: 778 Wh
  Energy total: 443.2 kWh

```

joanwa, 2021-02-22:
Great, at least consistent. The solar modules MS100/200 also allow to connect the upper buffer sensor to the module, than it'd be shown in the solar menu. My setup is that solar module records the lower buffer temperature, and the upper temperature via the HT3 boiler and your appears to have the same wiring. 

So I'd suggest the renaming according the board's documentation:

```
Warm water mix temperature -> Mixer temperature
Warm water buffer boiler temperature -> Storage temperature
```

Just had a quick look, but couldn't find the message description below in Norberts EMS docu, will check later again.

```
// 0x2A - MC10Status
// e.g. 88 00 2A 00 00 00 00 00 00 00 00 00 D2 00 00 80 00 00 01 08 80 00 02 47 00
// see https://github.com/proddy/EMS-ESP/issues/397
void Boiler::process_MC10Status(std::shared_ptr<const Telegram> telegram) {
    changed_ |= telegram->read_value(wwMixTemperature_, 14);
    changed_ |= telegram->read_value(wwBufferTemperature_, 18);
}
```

suysh, 2021-02-22:
Are you sure this is the temperature from the solar storage tank?

I think it's somewhere a temperature from the Warm water boiler and not from the Solar Module.
To be honest It's not completely clear where all my sensors are physically located. I'm trying to discover by plotting charts and comparing to reveal the heating principle.

joanwa, 2021-02-22:
I traced the sensor wires from the board, see diagram above. 32 goes to the upper buffer. sensor 25 goes to the warm water buffer inside the boiler. But anyways, we can already agree that they are off by a factor of 10 in your as well as my setup

proddy, 2021-02-22:
@joanwa a PR would be easiest. Do you want to go ahead and modify the names too?

suysh, 2021-02-22:
Indeed the factor of 10 can be changed.

@joanwa Is you setup the same as mine? Solar + Floorheating + Radiator.
[image]
[image]

joanwa, 2021-02-22:
mine is a bit different, but the solar system looks the same. Your setup also shows the lower buffer temp (TS2) connected to your MS100, and the middle/upper (TS3) is connected to your UI300. And this is indeed identical to mine. 
If you note my board diagram, sensor output 32 actually says TS3. Since this seem to be the middle buffer temperature, I'd also name it like this.

joanwa, 2021-02-22:
quick poll. in solar.cpp we name it tankBottomTemp, shall we than name this 

- tankMiddleTemp
- bufferMiddleTemp
- storageMiddleTemp
- ...?
I'd go for consistency and name it `tankMiddleTemp`

The MS200 documentation https://junkers-it.resource.bosch.com/media/website_it/documenti/im/gamma_completa_libretti/termoregolazione_1/ems_2_0/modulo_solare_ms200_ju.pdf
also calls TS3 the middle tank temperature, the upper tank temperature would be TS9

suysh, 2021-02-22:
tankMiddleTemp is fine for me.
I like the (TSx) in the description visible in the Dashboard, compatible with the manuals.

joanwa, 2021-02-22:
#714

joanwa, 2021-02-22:
@suysh the locale for MS200 is already set to show TSx, and now the tankMiddleTemp too:

[image]

proddy, 2021-02-23:
thanks guys. PR approved and merged. I made the changes manually for v3 (ESP32 version)

closed 2021-02-24 by proddy