emsesp/EMS-ESP #713: Boolean Format 1/0 is in text format

suysh, 2021-02-22:
I'm using the latest version v2.2.1b3
Maybe I'm just missing the context of the Boolean Format selection. Then I need extra conversion from text to numbers.

When selecting the Boolean Format to "1/0" the mqtt output for these boolean fields is still in string format instead of numbers.
"on" or "off"
"True" or "False"
"1" or "0"   ==>   is better in integer values 1 or 0 (can be plotted easily)

proddy, 2021-02-22:
yes, I was thinking about that when implementing the boolean format. And hoping no one would notice! I agree, it should be changed.

suysh, 2021-02-22:
I think it was OK in a previous version. I also just noticed because I upgraded and now I get string conversion errors.

proddy, 2021-02-23:
I remember now, this was already implemented in v3 (ESP32). I don't fancy porting it back to v2 (ESP8266) if you don't mind.

suysh, 2021-02-23:
Sorry, Now it is a bit unclear to me.
I'm using ESP8266, in which version it will be fixed?
Is there a major upgrade v3 soon? Or only for ESP32?

proddy, 2021-02-23:
It's only for the ESP32. I did a re-write on how data variables are rendered for the web, UI and MQTT. Turned out that the new implementation used too much memory on the ESP8266 which caused crashes,  so decided to make it exclusive for the ESP32.

I could look at any way of doing the boolean 1 & 0 on v2 if its important for you.

suysh, 2021-02-24:
OK. Then it's better to finally switch to ESP32 hardware.
Thanks for the advice.

proddy, 2021-02-24:
There's a whole discussion on this in there GitHub discussions. We've just hit the max on the ESP8266 hardware so no new features on v2. BBQKees is working on an ESP32 version of the gateway. On the ESP32 we can continue adding some cool stuff like German translations, heat curves, secure APIs 

..but I can still make the boolean work on the esp8266. I think I have a quick fix. Would true/false also work for you, which is standard in JSON. e..g `{"heatingactive":false,....}` ?

suysh, 2021-02-24:
No problem.
True/False will be better (if not in a string format)... didn't try that yet.
I'm reconfiguring my whole system (upgrade to openHAB3) so I can also migrate to ESP32.

Just hoping that the mqtt structure will be fixed, this has changed a lot last times.

MichaelDvP, 2021-02-24:
In v3(esp32) this is native boolean, but in v2 it is always a string for on/off, true/false/ and 1/0 

@proddy: I've tested a bit with v2 and added in Helpers:
```
void Helpers::json_boolean(JsonObject & json, const char * name, uint8_t value) {
    if (value == EMS_VALUE_BOOL_NOTSET) {
        return;
    }
    if (bool_format() == BOOL_FORMAT_ONOFF) {
        json[name] = (value != EMS_VALUE_BOOL_OFF) ? "on" : "off";
    } else if (bool_format() == BOOL_FORMAT_TRUEFALSE) {
        json[name] = (value != EMS_VALUE_BOOL_OFF); // ? true : false;
    } else {
        json[name] = (value != EMS_VALUE_BOOL_OFF) ? 1 : 0;
    }
}
```
Now i have to change all constructs like
```
    if (Helpers::hasValue(ignWork_, EMS_VALUE_BOOL)) {
        json["ignWork"] = Helpers::render_value(s, ignWork_, EMS_VALUE_BOOL);
	}
```
to `Helpers::json_boolean(json, "ignWork", ignWork_);`

Works on the first few values i've changed.

proddy, 2021-02-24:
that was exactly what I was thinking of too for v2 at first. Passing the json object. Then I was thinking since it's JSON and MQTT the values should really be treated as boolean (`true` and `false` without quotes). So my plan was to replace all the

```C
if (Helpers::hasValue(ignWork_, EMS_VALUE_BOOL)) {
        json["ignWork"] = Helpers::render_value(s, ignWork_, EMS_VALUE_BOOL);
}
```

to just
```C
json["ignWork"] = (ignWork_ != EMS_VALUE_BOOL_OFF);
```

but either method works

proddy, 2021-02-25:
and in the Web and ConsoleI think it should be fixed to showing just `on` or `off`, so no other option.

So if OpenHAB and Domoticz are ok with `true` and `false` in the JSON MQTT payloads we can probably remove the option entirely.  With HA there's no problem. Someone would need to check for me on the Domoticz/OpenHAB side

suysh, 2021-02-25:
I can test in openHAB3,
Since I'm not yet upgraded to ESP32, I can only test this on my old ESP8266.
Did you plan to do this modifications for the ESP8266 version? Let me know when there is a modified compiled bin file.

proddy, 2021-02-25:
you can also test without EMS-ESP. Just inject some MQTT payloads (e.g. via MQTTExplorer) replacing the "1" and "0" with true and false and see how OpenHAB detect it.

suysh, 2021-02-25:
I tried with manual mosquitto_pub commands to update an item with Type=Switch in openHAB3
Working: "OFF", "ON", OFF, ON
Not Working: "0", "1", 0, 1, True, False, true, false, "true", "false", "True", "False", on, off, "on", "off"

Do I have to post a message on the openHAB community about this JSON behaviour for booleans?

proddy, 2021-02-25:
I seems people have mentioned this before on the OpenHAB forums. It may be just easier if we add support for "ON" and "OFF" in capitals.

suysh, 2021-02-25:
There is a possibility for translating/mapping the message after JSON, but I'm learning the new version openHAB3 and need to search how to do.
Maybe it's a good solution to add support for "ON" "OFF".

suysh, 2021-02-25:
Actually, there are indeed some problems to log the status for a switch item with influxdb.
Since the influxdb program records switches as 0 and 1 values, I can bypass this problem to set my openHAB item as Type=Number.

The items with Type=Switch needs the "ON" or "OFF" mqtt messages.
It's easy to swap the type of an item later, since the logging of Numbers (only 0 and 1) are the same as for Switches.

proddy, 2021-02-25:
one idea may be to use node-red to convert the MQTT json values to whatever format you need? boolean true and false are the standards for boolean in JSON.

proddy, 2021-02-27:
made the change in v3 (ESP32)

suysh, 2021-03-02:
I upgraded to ESP32 today, tested the "ON" "OFF" values and it works in openHAB3.
Thank you so much for this enhancement.

closed 2021-03-02 by suysh