emsesp/EMS-ESP #715: hc3 data measured in ems-esp, not displayed in Home Assistant

swa72, 2021-02-23:
**Question**
hc3 data measured in ems-esp, not displayed in Home Assistant

**Screenshots**
[image]
[image]

I also get load of errors in the log file of HA. Here are some examples ...

```
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:469
Integration: MQTT (documentation, issues)
First occurred: 21 February 2021, 21:21:54 (15229 occurrences)
Last logged: 15:40:22

Invalid modes mode:
Invalid modes mode:
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}}
```
or
```
Logger: homeassistant.helpers.template
Source: helpers/template.py:514
First occurred: 21 February 2021, 21:58:58 (6 occurrences)
Last logged: 21 February 2021, 21:58:58

Error parsing value: 'dict object' has no attribute 'hc3' (value:
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}},
template: {{value_json.hc3.seltemp}})
Error parsing value: 'dict object' has no attribute 'hc3' (value:
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}},
template: {{value_json.hc3.mode}})
Error parsing value: 'dict object' has no attribute 'hc2' (value:
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}},
template: {{value_json.hc2.hatemp}})
Error parsing value: 'dict object' has no attribute 'hc2' (value:
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}},
template: {{value_json.hc2.seltemp}})
Error parsing value: 'dict object' has no attribute 'hc2' (value:
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}},
template: {{value_json.hc2.mode}})
```
or
```
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:415
Integration: MQTT (documentation, issues)
First occurred: 21 February 2021, 21:58:58 (4 occurrences)
Last logged: 21 February 2021, 21:58:58

Could not parse temperature from
{"hc1":{"seltemp":0,"currtemp":21.8,"hatemp":21.8,"targetflowtemp":0,"mode":"auto","modetype":"eco"}}
```

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information from http://ems-esp.local/api?device=system&cmd=info*

Running that command I get ...

```
This site can’t be reached
The connection was reset.
Try:

Checking the connection
Checking the proxy and the firewall
Running Windows Network Diagnostics
ERR_CONNECTION_RESET
```
I run a device from bbqkees that I received a couple of days ago, hooked up directly to the service jack (not external power source)  ...

[image]

proddy, 2021-02-23:
It looks like the MQTT does not contain all the heating circuits. Can you check which version you are? Also try and go the telnet console and type `show` to see if the hc3 is show. And do a `su` and then `call system info` to get some more data.

For MQTT you'll need to sniff what is being sent. A good tool for that is MQTTExplorer. See https://emsesp.github.io/docs/#/MQTT?id=monitoring-the-queue

swa72, 2021-02-24:
I may have to add that the single RC300 controls all three circuits
* hc1: radiators ground/first floor
* hc2: under floor heating ground/first floor
* hc3: radiators second floor

Regarding MQTT, number for publish fails have been zero.

Here we go ...

```
ems-esp:/$ show
EMS-ESP version 2.2.0
Boiler: Buderus Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductID:208, Version:01.04)
Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:11.09)
Mixer: MM100 (DeviceID:0x20, ProductID:160, Version:24.03)
Mixer: MM100 (DeviceID:0x21, ProductID:160, Version:24.03)
Mixer: MM100 (DeviceID:0x22, ProductID:160, Version:24.03)
Mixer: MM100 (DeviceID:0x28, ProductID:160, Version:24.03)
Solar: SM200/MS200 (DeviceID:0x30, ProductID:164, Version:25.04)
Controller: 9000i (DeviceID:0x09, ProductID:224, Version:35.03)
```
and
```
ems-esp:/# call system info
{
  "System": {
    "version": "2.2.0",
    "uptime": "000+09:57:48.231",
    "freemem": 24,
    "fragmem": 54
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 2,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 69413,
    "#read requests sent": 22734,
    "#write requests sent": 0,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus Logamax Plus/GB192/Condens GC9000 (DeviceID:0x08 ProductI                                                                                                          D:208, Version:01.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x2A 0x33 0x34 0                                                                                                          x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, Pr                                                                                                          oductID:158, Version:11.09)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0                                                                                                           0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0                                                                                                          x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x20, ProductID:160, Version:24.03)",
      "handlers": "0x2D7"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x21, ProductID:160, Version:24.03)",
      "handlers": "0x2D8"
    },
    {
      "type": "Mixer",
      "name": null,
      "handlers": "0x2D9"
    },
    {
      "type": "Mixer",
      "name": null,
      "handlers": "0x331"
    }
  ]
}
```
Lastly, how HA sees the connection ...

[image]

swa72, 2021-02-24:
I have no experience with ESP devices. One thing I noticed: the device is sluggish at best. Accessing the web interface doesn't work 6/10 times. After connecting, I have to refresh the devices list often with varying results. Putty connects instantly but gives me "network error: software caused connection abort" after a short time. I've read about power issues in another post (with the GB192i) - could this be a problem?

MichaelDvP, 2021-02-24:
You have a lot of devices and heating-circuits. I think the esp8266 ran out of memory to handle all the values. The low memory causes the connection issues with web and telnet. You can try to set mqtt sending intervall to different values for thermostat/boiler/mixer/solar (maybe 25/26/27/28 sec) to ensure that a message is send out before another device allocates sending memory. 
But best would be to use a esp32. There are compatible boards if you search for "esp32 mini" ,  from china for <5€, or fast delivery from europe for <10€.

swa72, 2021-02-24:
Hi Michael, thanks for the feedback. Will try out the MQTT sequence. From my perspective (as a layman) the mem usage looks ok ... but the uptime doesn't. I connected the device a couple of days ago. That's not the uptime of the current telnet session, is it?

```
ems-esp:/system$ show
Uptime:        000+00:00:39.165
Chip ID:       0x00d0e7e7
SDK version:   2.2.2-dev(38a443e)
Core version:  2_7_4
Full version:  SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
Boot version:  31
Boot mode:     1
CPU frequency: 160 MHz
Flash chip:    0x00164068 (4194304 bytes)
Reset reason:  Exception
Reset info:    Fatal exception:29 flag:2 (Exception) epc1:0x402337c0 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
Sketch size:   974544 bytes (2170880 bytes free)
Free heap:                6216 bytes
Free mem:                 24  %
Heap fragmentation:       49 %
Maximum free block size:  2928 bytes
Free continuations stack: 0 bytes

WiFi: Connected
SSID: hades
BSSID: 3C:A6:2F:D9:35:58
RSSI: -57 dBm (86 %)
MAC address: E8:DB:84:D0:E7:E7
Hostname: ems-esp
IPv4 address: 192.168.178.71/255.255.255.0
IPv4 gateway: 192.168.178.1
IPv4 nameserver: 192.168.178.1
Syslog: disabled
```

MichaelDvP, 2021-02-24:
The uptime counts from last processor reset. Reset reason is exception 29 (access to invalide address), which is a result from not enough memory. 
If memory is running low the first sign is that reloading the webinterface, takes long, needs some retries, or crashes the esp. 
On esp8266 is it not possible to load/reload web while telnet is running, you can navigate in a loaded web while telnet is open, but not reload it. 
But with closed telnet and mqtt not sending simultaniously the messages should be sended with your configuration. Check with mqtt explorer.

swa72, 2021-02-24:
Ok, changed to 

[image]

will check during the day, how the system performs. Good to know that its either web or telnet but not both :-/

swa72, 2021-02-24:
Well, HA still complains ...

```
Logger: homeassistant.components.mqtt.climate
Source: components/mqtt/climate.py:469
Integration: MQTT (documentation, issues)
First occurred: 23 February 2021, 18:53:33 (6451 occurrences)
Last logged: 14:21:36

Invalid modes mode: {"wwextra1":0}
Invalid modes mode:
Invalid modes mode: {"hc2":{"seltemp":23,"currtemp":22.4,"hatemp":22.4,"targetflowtemp":37,"mode":"auto","modetype":"comfort"},"hc3":{"seltemp":19,"currtemp":22.4,"hatemp":22.4,"targetflowtemp":32,"mode":"auto","modetype":"comfort"}}
Invalid modes mode: {"time":"09:53:13 24/02/2021","dampedtemp":8.6,"minexttemp":-12,"building":"medium","wwmode":"own_prog","wwtemp":60,"wwtemplow":45,"wwcircmode":"on","hc2":{"seltemp":23,"currtemp":22.4,"hatemp":22.4,"targetflowtemp":37,"mode":"auto","modetype":"comfort"},"hc3":{"seltemp":19,"currtemp":22.4,"hatemp":22.4,"targetflowtemp":32,"mode":"auto","modetype":"comfort"}}
Invalid modes mode: {"wwextra1":0,"hc1":{"seltemp":0,"currtemp":23.5,"hatemp":23.5,"targetflowtemp":0,"mode":"auto","modetype":"eco"},"hc2":{"seltemp":23,"currtemp":23.5,"hatemp":23.5,"targetflowtemp":30,"mode":"auto","modetype":"comfort"}}
```

proddy, 2021-02-24:
@swa72 those HA errors are not caused by the MQTT. It looks like the HA config is wrong. Can you use MQTTExplorer and removed all the entries from /ems-esp and /homeassistant, restart EMS-ESP and watch them load with MQTTExplorer active?

swa72, 2021-02-24:
Before going into MQTTExplorer, I checked the following: Removed Mosquitto from HA, removed MQTT integration from HA, rebootet. Added Mosquitto again. Upon adding the integration, one thermostat device is added and it has numerous entities that also refer to hc2 and hc3.  

I assume https://bbqkees-electronics.nl/wiki/gateway/home-assistant-configuration-heating-circuits.html  is only relevant if I have another thermostat, right? If I check the thermostat device and click on "MQTT INFO", I get a list of all entities that receive messages and I think all hc1, hc2 and hc3 are there.

[image]

If I then select e.g. "'Thermostat hc3 Control mode' (sensor.thermostat_hc3_control_mode" and show the last ten received messages, I see something like

```
Received 15:48:40
QoS: 0
Payload: time: '15:48:42 24/02/2021'
errorcode: (0)
dampedtemp: 14.7
minexttemp: -12
building: medium
wwmode: own_prog
wwtemp: 60
wwtemplow: 45
wwextra1: 0
wwcircmode: 'on'
hc1:
  seltemp: 22
  currtemp: 23.4
  hatemp: 23.4
  comforttemp: 22
  ecotemp: 20
  manualtemp: 23
  nofrosttemp: 0
  heatingtype: radiator
  targetflowtemp: 29
  offsettemp: 0
  designtemp: 75
  roominfluence: 3
  minflowtemp: 25
  maxflowtemp: 80
  summertemp: 17
  summermode: auto
  controlmode: outdoor
  program: 1
  mode: auto
  modetype: comfort
hc2:
  seltemp: 23
  currtemp: 23.4
  hatemp: 23.4
  comforttemp: 23
  ecotemp: 19
  manualtemp: 5
  nofrosttemp: 0
  heatingtype: floor
  targetflowtemp: 27
  offsettemp: 0
  designtemp: 45
  roominfluence: 3
  minflowtemp: 25
  maxflowtemp: 50
  summertemp: 17
  summermode: auto
  controlmode: outdoor
  program: 1
  mode: auto
  modetype: comfort
hc3:
  seltemp: 19
  currtemp: 23.4
  hatemp: 23.4
  comforttemp: 19
```
and there is very little hc3 info in there. So I reckon, its the sender ;-) Will try MQTTExplorer tonight ...

MichaelDvP, 2021-02-24:
I think the json size is to small for 3 heating circuits. 
If you have the thermostat_data from mqtt-explorer you can check the values from the api output: `http://ems-esp.local/api?device=thermostat&cmd=info&id=3`  for only hc3
and `http://ems-esp.local/api?device=thermostat&cmd=info` for all. 
(maybe with ip-address instead of ems-esp.local)
Mqtt is now limited to 1024 bytes (stack), the API can display up to 2k data. Iirc we can use dynamic ram for mqtt, but limited to 1500 bytes handeled by AsyncMqttClient.

proddy, 2021-02-24:
I think that is indeed the issue. The payload is getting cut. And as it doesn't have the `mode` tag, HA is showing an error. I'm experimenting with a new MQTT library that hopefully addresses this

swa72, 2021-02-24:
Ok, that info is a bit over my head ;-) Are you saying that each MQTT message cannot handle more than 1k of data and all the stuff that is sent exceeds that 1k? From "errorcode" to "comforttemp: 19" in the above example there are 855 characters (including blanks). So there is a tiny bit more room ;-)

`http://192.168.178.71/api?device=thermostat&cmd=info&id=3` gives `Invalid cmd`
`http://192.168.178.71/api?device=thermostat&cmd=info` gives `Invalid`

another try at `http://192.168.178.71/api?device=thermostat&cmd=info` gives `Invalid cmd`

Did I mention that the device takes friggin ages to respond :-/

I'm surprised that I'm the first user with three circuits.

Why not split the thermostat message into n pieces (for n circuits)?

Anyway, I got MQTTExplorer up and connected to `homeassistant-eth` (the raspi running HA) with the user that I created. What output would be most interesting?

[image]

If I click on `thermostat_data` it pretty much shows the same stuff 

```
{
  "time": "20:01:33 24/02/2021",
  "errorcode": "(0)",
  "dampedtemp": 14.6,
  "minexttemp": -12,
  "building": "medium",
  "wwmode": "own_prog",
  "wwtemp": 60,
  "wwtemplow": 45,
  "wwextra1": 0,
  "wwcircmode": "on",
  "hc1": {
    "seltemp": 22,
    "currtemp": 22.5,
    "hatemp": 22.5,
    "comforttemp": 22,
    "ecotemp": 20,
    "manualtemp": 23,
    "nofrosttemp": 0,
    "heatingtype": "radiator",
    "targetflowtemp": 33,
    "offsettemp": 0,
    "designtemp": 75,
    "roominfluence": 3,
    "minflowtemp": 25,
    "maxflowtemp": 80,
    "summertemp": 17,
    "summermode": "auto",
    "controlmode": "outdoor",
    "program": 1,
    "mode": "auto",
    "modetype": "comfort"
  },
  "hc2": {
    "seltemp": 23,
    "currtemp": 22.5,
    "hatemp": 22.5,
    "comforttemp": 23,
    "ecotemp": 19,
    "manualtemp": 5,
    "nofrosttemp": 0,
    "heatingtype": "floor",
    "targetflowtemp": 32,
    "offsettemp": 0,
    "designtemp": 45,
    "roominfluence": 3,
    "minflowtemp": 25,
    "maxflowtemp": 50,
    "summertemp": 17,
    "summermode": "auto",
    "controlmode": "outdoor",
    "program": 1,
    "mode": "auto",
    "modetype": "comfort"
  },
  "hc3": {
    "seltemp": 0,
    "currtemp": 22.5,
    "hatemp": 22.5,
    "comforttemp": 19
  }
}
```

I guess it wouldn't make much of a difference upgrading to an ESP32 instead of an ESP8266 (which means sending the stuff back to bbqkees), uh? I don't really care if I pay a couple of EUR more, I just want a reliable system. And I haven't been to the point where I can actually optimize my heating (which is the whole purpose of this exercise).

I'm still confused that anytime I telnet into the system, it gives me uptime of less than a day, and now its an 28 and not a 29.

```
Reset info:    Fatal exception:28 flag:2 (Exception) epc1:0x4024c34d epc2:0x00000000 epc3:0x00000000 excvaddr:0x0000000b depc:0x00000000
```

Thanks for any help or hints in advance
-Stefan.

proddy, 2021-02-25:
I'd like to test your setup, will simulate it with the test code to see what is really happening. Could you grab some example telegrams from your thermostat? Go to console and do a `watch raw <n>` for each of the heating circuits, so `n` is 3D, 47 and 51.

Then I can experiment if we can increase the MQTT json buffer sizes or indeed need to split into separate MQTT topics.

The stability issues you're facing is because EMS-ESP has many EMS devices loaded, in your case the Mixers. Each device eats memory and the ESP8266 doesn't have a lot.

swa72, 2021-02-25:
```
ems-esp:/$ watch raw 3D
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x3D
ems-esp:/$ watch on
Watching incoming telegrams, displayed in decoded format
000+21:31:22.659 N 0: [emsesp] Solar(0x30) -> Me(0x0B), SM100SystemConfig(0x358), data: 00 00 00 00 00 00 (offset 25)
000+21:31:23.055 N 1: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 A5 00 A5 80 00 00 80 00 80 00 80 00 00
000+21:31:23.287 N 2: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 8A 5A 0F 09 01 25 40 80 00 80 00 01 43 00 9B 12 2D 48 00 C8 00 02 18 00 00
000+21:31:23.569 N 3: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:31:23.830 N 4: [emsesp] Solar(0x30) -> Me(0x0B), SM100SolarCircuitConfig(0x35A), data: 78 05 00 5A 14 01 08 32 64 00 00 28 5A 0C
000+21:31:24.406 N 5: [emsesp] Solar(0x30) -> Me(0x0B), SM100Monitor(0x362), data: 00 FD 01 A9 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80
000+21:31:24.998 N 6: [emsesp] Solar(0x30) -> Me(0x0B), SM100Monitor(0x362), data: 80 00 (offset 24)
000+21:31:25.580 N 7: [emsesp] Solar(0x30) -> Me(0x0B), SM100Monitor2(0x363), data: 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00
000+21:31:26.136 N 8: [emsesp] Solar(0x30) -> Me(0x0B), SM100Config(0x366), data: 0D 6F 00 13 08
000+21:31:26.666 N 9: [emsesp] Solar(0x30) -> Me(0x0B), SM100CollectorConfig(0x380), data: 5A 64 00 00 3F 01 00 00 01
000+21:31:27.298 N 10: [emsesp] Solar(0x30) -> Me(0x0B), SM100Energy(0x38E), data: 00 00 27 AB 00 00 3E E1 00 01 FA 3D
000+21:31:27.932 N 11: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 06 64 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+21:31:28.507 N 12: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
000+21:31:28.698 N 13: [emsesp] Controller(0x09) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+21:31:29.174 N 14: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
000+21:31:33.036 N 15: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 8F 5A 0F 09 01 25 40 80 00 80 00 01 44 00 92 12 2D 48 00 C8 00 02 18 00 00
000+21:31:33.294 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:31:33.580 N 17: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 A5 00 A4 80 00 00 80 00 80 00 80 00 00
000+21:31:35.011 N 18: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 90 5A 0F 09 01 25 40 80 00 80 00 01 44 00 99 12 2D 48 00 C8 00 02 18 00 00
000+21:31:35.319 N 19: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:31:42.986 N 20: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 92 5A 0F 09 01 25 40 80 00 80 00 01 45 00 9B 12 2D 48 00 C8 00 02 18 00 00
000+21:31:43.295 N 21: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:31:43.555 N 22: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 A5 00 A4 80 00 00 80 00 80 00 80 00 00
000+21:31:45.794 N 23: [emsesp] Mixer(0x22) <- Boiler(0x08), UBAParameters(0x16), data: 01
000+21:31:45.808 N 24: [emsesp] Boiler(0x08) -> Mixer(0x22), UBAParameters(0x16), data: FF
000+21:31:45.832 N 25: [emsesp] Mixer(0x22) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+21:31:45.870 N 26: [emsesp] Mixer(0x22) <- Boiler(0x08), UBAParameterWW(0x33), data: 0B
000+21:31:45.894 N 27: [emsesp] Boiler(0x08) -> Mixer(0x22), UBAParameterWW(0x33), data: 00 FF 3C FB 00 14 FF 03 46 00 00
000+21:31:45.922 N 28: [emsesp] Mixer(0x22) -> All(0x00), ?(0x255), data: 00
000+21:31:46.292 N 29: [emsesp] Solar(0x30) -> All(0x00), SM100Config(0x366), data: 0D 6F 00 13 08
000+21:31:46.874 N 30: [emsesp] Mixer(0x22) <- Thermostat(0x10), ?(0x293), data: 01 (offset 5)
000+21:31:46.900 N 31: [emsesp] Thermostat(0x10) -> Mixer(0x22), ?(0x293), data: 00 (offset 5)
000+21:31:46.943 N 32: [emsesp] Mixer(0x22) -> All(0x00), MMPLUSStatusMessage_HC(0x2D9), data: 00 02 00 00 F1 00 00 00 03
000+21:31:51.587 N 33: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A6), data: 00 9A 03 08 (offset 13)
000+21:31:53.029 N 34: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 A5 00 A5 80 00 00 80 00 80 00 80 00 00
000+21:31:53.295 N 35: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 00 07 01 01 00 00 00 00 00 00 00
000+21:31:53.511 N 36: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 95 5A 0F 09 01 25 40 80 00 80 00 01 48 00 99 12 2D 48 00 C8 00 02 18 00 00
000+21:31:53.744 N 37: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:31:54.060 N 38: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 90 80 00 80 00 00 00 00 0A 00 84 82 13 29 64 00 00 00 11 96 EB 00 7F 41 01 6E
000+21:31:54.383 N 39: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 91 06 0A 13 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+21:31:56.035 N 40: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 9A 00 EC (offset 13)
000+21:31:56.228 N 41: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A7), data: 00 40 02 CC (offset 13)
000+21:32:02.036 N 42: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 91 80 00 80 00 00 00 00 0A 00 84 82 13 29 64 00 00 00 11 96 EB 00 7F 41 01 71
000+21:32:02.560 N 43: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 01 71 (offset 25)
ems-esp:/$ log off

```

swa72, 2021-02-25:
```
ems-esp:/$ watch raw 47
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x47
ems-esp:/$ watch on
Watching incoming telegrams, displayed in decoded format
000+21:33:24.705 N 44: [emsesp] Solar(0x30) -> Me(0x0B), SM100Monitor(0x362), data: 00 F3 01 A9 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80
000+21:33:25.297 N 45: [emsesp] Solar(0x30) -> Me(0x0B), SM100Monitor(0x362), data: 80 00 (offset 24)
000+21:33:25.878 N 46: [emsesp] Solar(0x30) -> Me(0x0B), SM100Monitor2(0x363), data: 80 00 80 00 00 00 80 00 80 00 80 00 00 80 00
000+21:33:26.435 N 47: [emsesp] Solar(0x30) -> Me(0x0B), SM100Config(0x366), data: 0D 6F 00 13 08
000+21:33:26.964 N 48: [emsesp] Solar(0x30) -> Me(0x0B), SM100CollectorConfig(0x380), data: 5A 64 00 00 3F 01 00 00 01
000+21:33:27.597 N 49: [emsesp] Solar(0x30) -> Me(0x0B), SM100Energy(0x38E), data: 00 00 27 AB 00 00 3E E1 00 01 FA 3D
000+21:33:28.230 N 50: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 06 64 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+21:33:28.806 N 51: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
000+21:33:28.878 N 52: [emsesp] Controller(0x09) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+21:33:29.498 N 53: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
000+21:33:33.035 N 54: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 A8 5A 0F 09 01 25 40 80 00 80 00 01 56 00 96 12 2D 48 00 C8 00 02 18 00 00
000+21:33:33.343 N 55: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:33.654 N 56: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 A5 00 A5 80 00 00 80 00 80 00 80 00 00
000+21:33:43.053 N 57: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 00 A5 00 A4 80 00 00 80 00 80 00 80 00 00
000+21:33:43.335 N 58: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 AA 5A 0F 09 01 25 40 80 00 80 00 01 56 00 98 12 2D 48 00 C8 00 02 18 00 00
000+21:33:43.592 N 59: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:45.110 N 60: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 AB 5A 00 09 01 04 40 80 00 80 00 01 56 00 9B 12 30 59 00 CC 00 00 00 00 00
000+21:33:45.418 N 61: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:46.085 N 62: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 AC 5A 00 01 01 04 40 80 00 80 00 01 56 00 02 12 30 59 00 CC 00 00 00 00 00
000+21:33:46.318 N 63: [emsesp] Mixer(0x22) <- Boiler(0x08), UBAParameters(0x16), data: 01
000+21:33:46.332 N 64: [emsesp] Boiler(0x08) -> Mixer(0x22), UBAParameters(0x16), data: FF
000+21:33:46.355 N 65: [emsesp] Mixer(0x22) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+21:33:46.393 N 66: [emsesp] Mixer(0x22) <- Boiler(0x08), UBAParameterWW(0x33), data: 0B
000+21:33:46.417 N 67: [emsesp] Boiler(0x08) -> Mixer(0x22), UBAParameterWW(0x33), data: 00 FF 3C FB 00 14 FF 03 46 00 00
000+21:33:46.446 N 68: [emsesp] Mixer(0x22) -> All(0x00), ?(0x255), data: 00
000+21:33:46.643 N 69: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:47.371 N 70: [emsesp] Mixer(0x22) <- Thermostat(0x10), ?(0x293), data: 01 (offset 5)
000+21:33:47.404 N 71: [emsesp] Thermostat(0x10) -> Mixer(0x22), ?(0x293), data: 00 (offset 5)
000+21:33:47.447 N 72: [emsesp] Mixer(0x22) -> All(0x00), MMPLUSStatusMessage_HC(0x2D9), data: 00 02 00 00 F1 00 00 00 03
000+21:33:47.766 N 73: [emsesp] Solar(0x30) -> All(0x00), SM100Config(0x366), data: 0D 6F 00 13 08
000+21:33:48.084 N 74: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 AE 5A 00 01 01 04 40 80 00 80 00 01 56 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:33:48.392 N 75: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:49.556 N 76: [emsesp] Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
000+21:33:52.833 N 77: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 98 00 EE (offset 13)
000+21:33:53.025 N 78: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A6), data: 00 98 03 0A (offset 13)
000+21:33:53.259 N 79: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 B6 5A 00 01 01 04 40 80 00 80 00 01 56 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:33:53.518 N 80: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:53.784 N 81: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 91 80 00 80 00 00 00 06 00 00 84 82 13 29 66 00 00 00 11 96 ED 00 7F 41 01 91
000+21:33:54.108 N 82: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 91 06 0A 13 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+21:33:54.328 N 83: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 01 2D 80 00 00 80 00 80 00 80 00 00
[... cut ...]
000+21:33:55.084 N 86: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 BA 5A 00 01 01 04 40 80 00 80 00 01 56 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:33:55.342 N 87: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:33:57.060 N 88: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 C2 5A 00 01 01 04 40 80 00 80 00 01 56 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:33:57.318 N 89: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
ems-esp:/$ log off
```

swa72, 2021-02-25:
```
ems-esp:/$ watch raw 51
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a device ID or telegram type of 0x51
ems-esp:/$ watch on
Watching incoming telegrams, displayed in decoded format
000+21:35:43.083 N 90: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 F7 5A 00 01 01 00 40 80 00 80 00 01 53 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:35:43.291 N 91: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:35:43.602 N 92: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 00 00 80 00 00 80 00 80 00 80 00 00
000+21:35:46.041 N 93: [emsesp] Mixer(0x22) <- Boiler(0x08), UBAParameters(0x16), data: 01
000+21:35:46.055 N 94: [emsesp] Boiler(0x08) -> Mixer(0x22), UBAParameters(0x16), data: FF
000+21:35:46.079 N 95: [emsesp] Mixer(0x22) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+21:35:46.116 N 96: [emsesp] Mixer(0x22) <- Boiler(0x08), UBAParameterWW(0x33), data: 0B
000+21:35:46.141 N 97: [emsesp] Boiler(0x08) -> Mixer(0x22), UBAParameterWW(0x33), data: 00 FF 3C FB 00 14 FF 03 46 00 00
000+21:35:46.170 N 98: [emsesp] Mixer(0x22) -> All(0x00), ?(0x255), data: 00
000+21:35:46.846 N 99: [emsesp] Mixer(0x22) <- Thermostat(0x10), ?(0x293), data: 01 (offset 5)
000+21:35:46.876 N 100: [emsesp] Thermostat(0x10) -> Mixer(0x22), ?(0x293), data: 00 (offset 5)
000+21:35:46.919 N 101: [emsesp] Mixer(0x22) -> All(0x00), MMPLUSStatusMessage_HC(0x2D9), data: 00 02 00 00 F1 00 00 00 03
000+21:35:48.789 N 102: [emsesp] Solar(0x30) -> All(0x00), SM100Config(0x366), data: 0D 6F 00 13 08
000+21:35:52.252 N 103: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode2(0x31D), data: 09 (offset 2)
000+21:35:53.002 N 104: [emsesp] Thermostat(0x10) -> Mixer(0x28), ?(0x33B), data: 01 (offset 1)
000+21:35:53.176 N 105: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 00 00 80 00 00 80 00 80 00 80 00 00
000+21:35:53.421 N 106: [emsesp] Mixer(0x28) -> All(0x00), MMPLUSStatusMessage_WWC(0x331), data: 1E (offset 3)
000+21:35:53.605 N 107: [emsesp] Mixer(0x28) -> All(0x00), MMPLUSStatusMessage_WWC(0x331), data: 1E (offset 5)
000+21:35:53.793 N 108: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 01 00 07 01 01 00 00 00 00 00 00 00
000+21:35:54.033 N 109: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 F5 5A 00 01 01 00 40 80 00 80 00 01 53 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:35:54.241 N 110: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:35:54.558 N 111: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 00 91 80 00 80 00 00 00 04 00 00 84 82 13 29 66 00 00 00 11 96 ED 00 7F 41 01 2E
000+21:35:54.838 N 112: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 96 00 F0 (offset 13)
000+21:35:55.030 N 113: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A6), data: 00 96 03 0C (offset 13)
000+21:35:55.256 N 114: [emsesp] Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 91 06 0A 13 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+21:35:55.572 N 115: [emsesp] Mixer(0x28) -> All(0x00), MMPLUSStatusMessage_WWC(0x331), data: 01 (offset 8)
000+21:35:56.084 N 116: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A7), data: 00 3C 02 D0 (offset 13)
000+21:35:56.272 N 117: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 00 96 (offset 8)
000+21:35:56.927 N 118: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A6), data: 00 96 (offset 8)
000+21:35:57.119 N 119: [emsesp] Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A7), data: 00 3C (offset 8)
000+21:36:00.132 N 120: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 F3 5A 00 01 01 00 40 80 00 80 00 01 51 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:36:00.440 N 121: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:36:01.832 N 122: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 91 80 00 80 00 00 00 04 00 00 84 82 13 29 66 00 00 00 11 96 ED 00 7F 41 01 2B
000+21:36:02.357 N 123: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 01 2B (offset 25)
000+21:36:02.893 N 124: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 00 FF 3C FB 00 14 FF 03 46 00 00 FF 00
000+21:36:03.132 N 125: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 F3 5A 00 01 01 00 40 80 00 80 00 01 51 00 00 12 30 59 00 CC 00 00 00 00 00
000+21:36:03.441 N 126: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 00 00 00 00 (offset 27)
000+21:36:03.677 N 127: [emsesp] Boiler(0x08) -> All(0x00), MC10Status(0x2A), data: 00 00 00 00 00 00 00 01 2B 00 00 80 00 00 80 00 80 00 80 00 00
000+21:36:04.017 N 128: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300Monitor(0x2A5), data: 00 E6 21 2C 1C 00 2C 28 00 96 03 03 01 00 96 00 F0 00 00 11 01 03 08 FD 00
000+21:36:04.341 N 129: [emsesp] Mixer(0x28) <- Boiler(0x08), UBAParameters(0x16), data: 01
000+21:36:04.355 N 130: [emsesp] Boiler(0x08) -> Mixer(0x28), UBAParameters(0x16), data: FF
000+21:36:04.378 N 131: [emsesp] Mixer(0x28) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00
000+21:36:04.417 N 132: [emsesp] Mixer(0x28) -> Boiler(0x08), ?(0x1B), data: 00
[... cut ...]
000+21:36:29.184 N 216: [emsesp] Thermostat(0x10) -> Mixer(0x28), ?(0x33B), data: 01 01 00
000+21:36:29.236 N 217: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 15 02 12 19 00 23 03 00 10 FF 00
000+21:36:29.953 N 218: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 24)
000+21:36:30.546 N 219: [emsesp] Solar(0x30) -> Me(0x0B), SM100Time(0x391), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (offset 50)
ems-esp:/$ log off
Log level = off
```

swa72, 2021-02-25:
I hope that helps ...

proddy, 2021-02-25:
oops, you have an RC300. Wrong telegrams. Can you do `watch raw <n>` with 2A5, 2A6 and 2A7. Don't do a `'watch on`, just sit back and wait 10-20 seconds until one or two records come in.

swa72, 2021-02-25:
Ok, here we go ...

```
ems-esp:/$ watch raw 2A5
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a telegram type of 0x2A5
000+00:10:34.713 N 0: [telegram] Rx: 10 00 FF 00 01 A5 00 E5 21 2C 1D 00 2C 28 00 8A 03 03 01 00 8A 00 FC 00 00 11 01 03 08 F3 00 00
000+00:10:35.849 N 1: [telegram] Rx: 10 00 FF 19 01 A5 06 04 FF E5 00 00 FF 64 50 00 3C 01 FF 01 02 8C
000+00:10:51.631 N 2: [telegram] Rx: 10 00 FF 0D 01 A5 00 89 00 FD 92
```
and
```
ems-esp:/$ watch raw 2A6
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a telegram type of 0x2A6
000+00:11:55.645 N 5: [telegram] Rx: 10 00 FF 0D 01 A6 00 88 03 1A 47
000+00:12:18.546 N 6: [telegram] Rx: 10 0B FF 00 01 A6 00 E5 21 2E 1E 00 2E 26 00 88 03 03 01 00 88 03 1A 00 00 11 01 03 08 F2 00 F3
000+00:12:19.172 N 7: [telegram] Rx: 10 0B FF 19 01 A6 06 04 00 03 00 00 FF 64 32 00 3C 01 FF 01 02 21
```
and
```
ems-esp:/$ watch raw 2A7
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a telegram type of 0x2A7
000+00:12:38.971 N 8: [telegram] Rx: 10 00 FF 00 01 A7 00 E5 21 26 00 00 26 22 00 2E 03 03 01 00 2E 02 DE 00 00 11 01 03 08 F2 00 52
000+00:12:39.652 N 9: [telegram] Rx: 10 00 FF 19 01 A7 06 07 FF E2 00 00 FF 00 50 00 3C 01 FF 01 02 33
000+00:12:54.387 N 10: [telegram] Rx: 10 00 FF 0D 01 A7 00 2D 02 DF 36
000+00:12:56.101 N 11: [telegram] Rx: 10 00 FF 08 01 A7 00 2D 2D
```

proddy, 2021-02-25:
@MichaelDvP I see three possible fixes for this:

The first would be to up the value of `EMSESP_MAX_JSON_SIZE_LARGE` from 1024 to 1500 in `EMSESP::publish_device_values()`  and hope the hc3 fits in, and that there is enough stack memory left. If that fails, use `DynamicJsonDocument` which looking at his available heap may be dangerous. His system is close to crashing because of memory.

The second is to advise using MQTT format 'Single' so each hc is published on its own MQTT topic. Then modify the Home Assistant discovery code to handle this variant. It'll be some extra work since the same will need to be done for the Mixer and the Dallas Sensor, and a new option for HA Discovery like in v3. I'd rather not do this!

The third, and easiest, is to spend 5 euros on replacing the ESP8266 with a ESP32 and switching over to v3 !

swa72, 2021-02-25:
Thanks for the analysis. That’s a tough decision;-) I will ask bbqkees to exchange the device and pay the difference/handling fee. Everything else seems like way to much futzing around.

proddy, 2021-02-26:
great, moving to ESP32 greatly simplifies things. You can pick up a ESP32 for a few euros and replace the ESP8266 on your gateway yourself. See https://bbqkees-electronics.nl/wiki/gateway/upgrade-the-gateway-to-an-esp32.html

swa72, 2021-02-26:
Yupp, I've read that. But I'm not into soldering stuff ;-) Kees offered me a return so I'll rather take the easy route :-) So I think, we can close this issue.

closed 2021-02-26 by proddy