emsesp/EMS-ESP #716: ESP32 hardware connections

suysh, 2021-02-24:
In the documentation I found the connections for ESP8266 Wemos D1 mini.
[image]
To upgrade to ESP32 I can't find the needed connections. I have this board:
[image]
[image]
Kind regards, Hans

MichaelDvP, 2021-02-24:
This board is pin compatible to Wemos D1 Mini if you use the white framed pins and parts side up. 
The settings are default for esp32: 
TX: io5, RX: io23, LED: io2 (inverted), DALLAS: io18

proddy, 2021-02-24:
Also good to add to https://github.com/proddy/EMS-ESP/issues/690

bbqkees, 2021-02-24:
I just updated my wiki:
https://bbqkees-electronics.nl/wiki/gateway/upgrade-the-gateway-to-an-esp32.html

glitter-ball, 2021-02-24:
@bbqkees - thanks! Brilliant docs as ever and I’ve just ordered an ESP32 to give it a go.

If, in time, someone finds a case which fits (or could provide 3D print file) that would be perfect but really happy to give it a go like this.

bbqkees, 2021-02-24:
@glitter-ball If you cut the case lid as described it actually fits together pretty nicely afterwards.
The only thing that will be missing is the side piece which is not really that important I think, especially if you use the USB port.
If you really want you can also cut the side piece in half so it will fit again as well. Then only the ESP32 board will stick out a little bit.

suysh, 2021-03-02:
I upgraded today to ESP32 and it works as now described in the excellent documentation.

closed 2021-03-02 by suysh