emsesp/EMS-ESP #718: Strange behavior of warm water after connection of ems-esp

appliko, 2021-02-27:
Hi

I have an issue with warm water. After installation of ems-esp board, it is slower and less hot.

I'm using **esp8266 / 2.2.2-dev(38a443e)**, service key mode
Boiler is a **Buderus Logamax plus GB152-16 with RC35** with external buffer
Boiler is recognized as **BK13/BK15/Smartline/GB1x2**
I'm using it from Domoticz via MQTT set as "on change"

**Before** installation of the board, warm water was immediately available, even with low flow.
Warm water was also quickly available with little opening of nozzle. Hot water was really hot
**After** installation, temperature of warm and hot water is generally low. It needs to fully open to get it warm
Also, it needs time to get it warm. And, more curious, if I close and open the nozzle again I got a shot of cold water

- Warm water was set to 60°C on RC35 and auto on the boiler, before and now too
- I did try to set recirculation pump to always on from rc35 but nothing change
- I did try to change Boiler Comfort Mode from Hot to Intelligent or echo, but every time I change it backs alone to Hot
- I did try to switch off and on the RC35 and the boiler, but nothing changed

Any idea ?

Thanks

MichaelDvP, 2021-02-28:
Please post the telnet output of `show`-command (or <F2>). Then we can see all settings.

appliko, 2021-02-28:
Here it is:
--------------
EMS-ESP version 2.2.0

Boiler: BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:03.06)
  Heating active: on
  Warm water/DHW active: off
  Service code: -H
  Service code number: 200
  Selected flow temperature: 57 °C
  Burner selected max power: 100 %
  Burner current power: 37 %
  Pump modulation: 100 %
  Outside temperature: 5.1 °C
  Current flow temperature: 55.2 °C
  Return temperature: 54.5 °C
  Mixer switch temperature: 0.0 °C
  System pressure: 1.6
  Max temperature: 59.0 °C
  Gas: on
  Flame current: 6.6 uA
  Pump: on
  Fan: on
  Ignition: off
  Heating activated: on
  Heating temperature setting: 90 °C
  Circuit pump modulation max power: 100 %
  Circuit pump modulation min power: 55 %
  Circuit pump delay time: 5 min
  Burner min period: 10 min
  Burner min power: 0 %
  Burner max power: 100 %
  Temperature hysteresis on: -6 °C
  Temperature hysteresis off: 6 °C
  Burner starts: 93839
  Burner active time: 234 days 17 hours 42 minutes
  Heating active time: 176 days 21 hours 16 minutes
  Boiler total uptime: 1437 days 16 hours 3 minutes
  Maintenance message:  
  Scheduled maintenance: off
  Warm water selected temperature: 60 °C
  Warm water set temperature: 60 °C
  Warm water disinfection temperature: 70 °C
  Warm water type: buffer
  Warm water charging type: 3-way valve
  Warm water circulation pump available: on
  Warm water circulation pump freq: 2x3min
  Warm water circulation active: off
  Warm water current temperature (intern): 58.0 °C
  Warm water current temperature (extern): 58.0 °C
  Warm water current tap water flow: 0.0 l/min
  Warm water storage temperature (extern): 58.0 °C
  Warm water activated: on
  Warm water one time charging: off
  Warm water disinfecting: off
  Warm water charging: off
  Warm water recharging: off
  Warm water temperature ok: on
  Warm water active: off
  Warm water heating: off
  Warm water starts: 5456
  Warm water active time: 57 days 20 hours 26 minutes

Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.06)
  Time: 10:30:47 28/02/2021
  Error code: (0)
  Damped outdoor temperature: 1 °C
  Temperature sensor 1: 21.0 °C
  Temperature sensor 2: 21.6 °C
  Offset int. temperature: 0.0
  Min ext. temperature: -10.0 °C
  Building: medium
  Warm water mode: auto
  Warm water circulation mode: auto
  (hc 1) Setpoint room temperature: 21.0 °C
  (hc 1) Current room temperature: 21.0 °C
  (hc 1) Day temperature: 21.0 °C
  (hc 1) Night temperature: 19.0 °C
  (hc 1) Holiday temperature: 17.0 °C
  (hc 1) Nofrost temperature: 5 °C
  (hc 1) Heating type: radiator
  (hc 1) Target flow temperature: 57 °C
  (hc 1) Offset temperature: 0 °C
  (hc 1) Design temperature: 70 °C
  (hc 1) Room influence: 3 °C
  (hc 1) Flow temperature offset: 5 °C
  (hc 1) Min. flow temperature: 5 °C
  (hc 1) Max. flow temperature: 75 °C
  (hc 1) Summer temperature: 20 °C
  (hc 1) Reduce mode: reduce
  (hc 1) Timer program: 1
  (hc 1) Control mode: outdoor
  (hc 1) Mode: auto
  (hc 1) Mode type: day

Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)

appliko, 2021-02-28:
Waiting on telnet, I also got some crc errors. Random interval, 15s/1min
Dashboard/Ems status  is reporting 100% on RX but 99% on TX line quality
----------------
e003+02:52:01.142 E 14: [telegram] Rx: FF 08 00 18 00 39 02 4D 64 00 01 01 20 60 80 00 02 43 02 4D 00 00 0F 30 59 00 CC 00 00 00 63 (CRC 63 != 44)
e003+02:52:36.084 E 15: [telegram] Rx: FF 08 00 18 00 39 02 3B 64 00 01 01 20 60 80 00 02 43 02 39 00 00 10 30 59 00 CC 00 00 00 BD (CRC BD != 9A)
e003+02:53:15.105 E 16: [telegram] Rx: FF 08 00 18 00 39 02 27 64 00 01 01 20 60 80 00 02 43 02 23 00 00 10 30 59 00 CC 00 00 00 D4 (CRC D4 != F3)
e003+02:53:30.784 E 17: [telegram] Rx: FF 08 00 18 00 39 02 21 64 00 01 01 A0 60 80 00 02 43 02 1B 00 00 0F 30 59 00 CC 00 00 00 45 (CRC 45 != 62)

MichaelDvP, 2021-02-28:
Warm Water settings seems ok, circulation is set to 2x3min per hour, maybe it's cooling down to fast. Try to increase the value and check if it works. : Telnet login als `su` (password if not changed `ems-esp-neo`) and set `call boiler wwcircmode 7`.

appliko, 2021-02-28:
I did try to set circulation "always on" from RC35 but no effect
Anyway I'm going to try

proddy, 2021-02-28:
I was working with @appliko on this issue on Gitter earlier. What's strange is that the boiler's hot water was working fine before EMS-ESP was plugged in. There are no Tx/Write commands sent when EMS-ESP boots up so it can't be changing any values and the Shower Alert code is disabled. I recommended setting TxMode to 0 (off) and trying again to be 100% sure. Also a reset of the Boiler back to factory settings might b a good thing to do

appliko, 2021-02-28:
I did try to set TX mode to 0 and restart everything, but nothing changed
I'm not aware of how to reset boiler to factory settings. I have a concern about specific installation settings that would be required, but here maybe this community can help

MichaelDvP, 2021-02-28:
@appliko : boiler and thermostat are different settings, RC35 circulation "always on" means no day/night switching, but does not influence the boiler circulationmode.

@proddy I don't think this is caused by attaching the ems-esp, maybe accidental changes by trying out. Also i do not recomment resetting boiler to factory defaults, you have to set all settings specific for the system too.

appliko, 2021-02-28:
The only thing I did touch is "Boiler Comfort Mode" as exposed on Domoticz through MQTT
It is showing "Hot" as value.
I did try to change it, and after showing new value, it comes back alone to Hot
I never try commands via telnet other than "show" until today

2021-02-27 22:00:36 | Hot | Admin
2021-02-27 22:00:36 | Intelligent | Admin
2021-02-18 19:56:24 | Hot | Admin
2021-02-18 19:56:24 | Eco | Admin
2021-02-18 19:14:30 | Hot | Admin
2021-02-18 19:14:30 | Intelligent | Admin
2021-02-16 16:34:09 | Hot

appliko, 2021-02-28:
How can I check if "call boiler wwcircmode 7" is effective ?
I got no error on the command but if I show values I still see 2x3 min on circ pump freq

appliko, 2021-02-28:
Supposing RC35 is overwriting, I found RC35 service manual and I set recirculation mode to continuous
Now it is reported correctly. 
"Warm water circulation pump freq: continuous"
I'll post if something change

appliko, 2021-02-28:
I confirm is on now. Lets see if warm water is better.
[image]

MichaelDvP, 2021-02-28:
Are you still on tx-mode 0? Then setting commands do not work. 
RC35 service menu is used to change all settings including boiler. The "circ pump frequ" in this menu only changes the boiler mode, you can check this with `show` command.

appliko, 2021-02-28:
> Are you still on tx-mode 0? Then setting commands do not work.
> RC35 service menu is used to change all settings including boiler. The "circ pump frequ" in this menu only changes the boiler mode, you can check this with `show` command.

I did set back to mode 1 after testing mode 0 for one day, after restarting everything. No change.
RC35 service menu did change circulation pump frequency. I can see it on data.
it was 2/3 min before the change, and it is continuous after.
Also, I can see from data that warm water circulation pump is always on now. (see timeline on previous message)
But now, after one hour it is a bit warmer but not anyway as before

I was wondering instead if before for some reasons max temp of hot water was 80 degrees instead of 60
I left RC35 disconnected for some time few days before ems-esp interface installation. Maybe he back to default settings ?
I remember hot water full open was really hot, that cannot keep the hand under more than 1s

MichaelDvP, 2021-02-28:
The RC35 has a limiter to 60 degree, you can overwrite this limit in the service menu. Maybe the limiter is resetted on power loss after buffer cap is empty. 
This value is not read by ems-esp, it is in telegram 0x37 at pos. 8. You can check in telnet by `read 10 37`, 9'th byte is 3C (60°C). 
There is also no dedicated command to change it in ems-esp, to set the limit to 80°C: `call send "0B 10 37 08 50"`
Then it should be possible to set the wwtemp to 80 degree: `call boiler wwtemp 80`

appliko, 2021-02-28:
Changed by service menu
Now max allowable is 80°C and set is 70°C. Also reported 70 on show command
The "feeling" at full hot is like before, but still kind of slow.
I set also recirculation to continuous and will check later
Because on service menu all settings on RC35 was at default, maybe really that he got reset. Unfortunately I don't have a previous snapshot of situation.
Anyway, I will keep trying on recirculation parameters.
Thanks until now for the support, especially because probably it is not ems-esp directly related

MichaelDvP, 2021-02-28:
Not all values are reset to default, the system parameters are stored permanently, e.g. default is no warm water, no circulation, reducemode outdoor, designtemperature 75°C, etc. Also the defaults fits for most systems and are not changed on installation.

But the DHW-limit is a safety function, at 60°C skin burning is at >1second contact, enough time to react, higher temperatures can be dangerous for some people. I understand that Bosch resets this on unclear situations (longer powercut or Rc35 moved to another system?)

appliko, 2021-03-01:
I saw on service on RC35 menu that you can choose between "3-way valve" and "cilynder"
Is this regarding external buffer ? Because I have an external buffer.
All overall feeling is that heating is coming from a direct heating boiler, while water here is coming from buffer
Also, if I close hot water for some seconds, when I reopen is very cold. This make me thinking that recirculation too is cold 
This maybe a good hint to think about

glitter-ball, 2021-03-12:
Is this instantaneous/flow/combi hot water or from a storage cylinder? If it's instant, many boilers have an option to keep the hot water plate heat exchanger warm (to give faster hot water response) or not (to save energy). Bosch boilers in the UK call this 'Preheat'  or 'ECO'. Maybe this setting has been disturbed somehow?

On my boiler with storage cylinder, the Preheat and ECO settings are used to define when the cylinder is reheated. In 'Preheat' mode, cylinder heats with 5ºC temperature drop. In 'ECO', cylinder heats when a 10ºC drop is reached.

appliko, 2021-03-12:
There is an external cylinder for hot water. 
The _boiler comfort mode_ setting hot/eco/intelligent, is fix to **hot**
If I change it, it back to hot by itself, so I think the thermostat is keeping to write it. 
Is it something I can change on the RC35 ?

on RC35 service menu the heating circuit is set to _3-way valve_ but I don't know if it is the original setting. 
Should be changed to _Cylinder charge pump_ ? Do you think there is any risk on trying ?

What is strange to me is that if I close hot water for even few seconds, I get one very cold water shot when I reopen. 
This means to me that is the boiler that do it, not pipes temperature, and that water is not coming directly from cylinder

MichaelDvP, 2021-03-12:
>  on RC35 service menu the heating circuit is set to 3-way valve but I don't know if it is the original setting.

Look at your installation. Charging pump (left) has a seperate pump for charging the DHW-buffer, With 3-way-valve (right) you have only one pump and a 3-way-valve that switches between DHW-loading and heating circuit. For the logic it means: in left case only the pump has to be activated to heat up DHW, in right case the pump and the valve has to activated.

[image]