emsesp/EMS-ESP #721: Auto-discovery for Home Assistant for the shower_data

rkuijer, 2021-03-03:
**Is your feature request related to a problem? Please describe.**
Currently, the shower data isn't available with auto-discovery for Home Assistant like the Shower Active binary sensor is. For consistency, I would say that the shower data should also be promoted with an auto-discovery message.

**Describe the solution you'd like**
Make an auto-discovery message for the shower data. 
`homeassistant/sensor/ems-esp/shower_data/config`

**Describe alternatives you've considered**
Brain fart, add the shower active and data as a device within the ems-esp and not part of the boiler.

**Additional context**

rkuijer, 2021-03-03:
I've played around with the following auto-discovery message
```
{
  "name": "Shower Data",
  "uniq_id": "shower_data",
  "~": "ems-esp",
  "json_attr_t": "~/shower_data",
  "stat_t": "~/shower_data",
  "val_tpl": "{{value_json['shower_timer']}}",
  "ic": "mdi:shower",
  "dev": {
    "ids": [
      "ems-esp-boiler"
    ]
  }
}
```
using the Mqtt::publish_ha method but I'm not sure if this is the way to go. 
I've tested it locally and it works.

proddy, 2021-03-03:
that's the correct way.  I can add it to the code if you like, or you can via a PR?

rkuijer, 2021-03-03:
Let me try to create a PR.
Normally I don't work with GitHub so it's a little bit new for me, learning on the job.........

PR #724 (2021-03-05): Send HA shower_data discovery information.
Proposal to send the HA auto discovery for the shower data published by EMS-ESP, Fixes  #721 

As seen by HA
[image]
closed 2021-03-05 by proddy