emsesp/EMS-ESP #726: Reduce TX-Modes to 0-4

MichaelDvP, 2021-03-05:
The Tx-modes 5-255 are mainly a relict from esp32 development with it's different break logic (see #380, etc.). I think no one uses the modes with timer control. On v2 8266 memory is very low and the extra modes consumes some 100 bytes.  Also setting the right mode is not clear to all users. I suggest to use only modes 0-4 and replace the input by a dropdown list:
[image]

@proddy What do you think?

proddy, 2021-03-05:
I think it's a wonderful idea! We would need to update the documentation to, especially to explain #4

MichaelDvP, 2021-03-06:
v2, v3 and wiki updated.

closed 2021-03-06 by MichaelDvP