emsesp/EMS-ESP #727: Extension of boiler data with current power and temperatures

Archidus, 2021-03-06:
Attached is an Excel file showing the information menu structure off an HPC400 and the values at that time and mapping to the values from telegrams 0x48d and 0x48f. The data is for a boiler identified as "Enviline/Compress 6000AW/Hybrid 7000iAW".

In 0x48d is the current compressor power
0x48f shows the temperatures of the outdoor unit.

[image]
[image]
[image]
[image]
[EMS data.xlsx](https://github.com/proddy/EMS-ESP/files/6095094/EMS.data.xlsx)

proddy, 2021-03-06:
nice, thanks for sharing. some comments
- is the boiler/heatpump a Bosch Compress? Which model
- does EMS-ESP detect a Heat Pump device? or just the Boiler?
- does the HPC 400 module get detected as a seperate device too?
- Since it's a lot of additional data (lots of temperatures) we need a seperate MQTT topic for this
- Could you help with the English translations? Multi-locale is on the feature list for v3 (supportin DE, NL and UK)
- I suggest adding to v3 (ESP32) and not v2, since it's easier to extend with new variables

Archidus, 2021-03-06:
- is the boiler/heatpump a Bosch Compress? Which model
it`s a Bosch SupraEco SAS-6-2 air water heatpump

- does EMS-ESP detect a Heat Pump device? or just the Boiler?
just the boiler

- does the HPC 400 module get detected as a seperate device too?
it is deceted as a thermostat. Seems to be very similar to the cw400.

[image]

proddy, 2021-03-11:
Hi @Archidus, thanks for putting the data into the XLS. I'd like to go ahead and implement this if you're up to it using some of your real live data.

to summerize:

### 0x48D
Leistung Kompressor - div10 - in kW - at position 11 of the data block

### 0x38F
  
TC0 | Wärmeträgerflüssigkeit Eingang - div10 - in C - at position ? of the data block
TC1 | Wärmeträgerflüssigkeit Ausgang - div10 - in C - at position ? of the data block
TC3 | Verflüssigertemperatur - div10 - in C - at position ? of the data block
TR3 | Tremperaturfühler Kältemittel (Flüssigkeit) - div10 - in C - at position ? of the data block
TR4 | Verdampfer Eintritt - div10 - in C - at position ? of the data block
TR5 | Temperaturfühler Kompessoransaugleitung - div10 - in C - at position ? of the data block
TR6 | Temperaturfühler Kompressorausgangsleitung - div10 - in C - at position ? of the data block
TR7 | Temperaturfühler Kältemittel (Gas) - div10 - in C - at position ? of the data block
TL2 | Außenlufttemperaturfühler - div10 - in C - at position ? of the data block
PL1 | Niedrigdruckfühler - div10 - in C - at position ? of the data block
PH1 | Hochdruckfühler - div10 - in C - at position ? of the data block

can you help fill in the data block positions? 0 is the first byte

Archidus, 2021-03-11:
I´m completely fine with using the data.

0x**3**8D

Leistung Kompressor - div10 - in kW - at position 11 of the data block

0x38F

- TC0 | Wärmeträgerflüssigkeit Eingang - div10 - in C 
  highbyte at position 6 of the data block, lowbyte at position 7 of the data block

- TC1 | Wärmeträgerflüssigkeit Ausgang - div10 - in C 
  highbyte at position 4 of the data block, lowbyte at position 5 of the data block

- TC3 | Verflüssigertemperatur - div10 - in C 
 highbyte at position 2 of the data block, lowbyte at position 3 of the data block

- TR3 | Tremperaturfühler Kältemittel (Flüssigkeit) - div10 - in C 
 highbyte at position 16 of the data block, lowbyte at position 17 of the data block

- TR4 | Verdampfer Eintritt - div10 - in C 
 highbyte at position 18 of the data block, lowbyte at position 19 of the data block

- TR5 | Temperaturfühler Kompessoransaugleitung - div10 - in C 
 highbyte at position 20 of the data block, lowbyte at position 21 of the data block

- TR6 | Temperaturfühler Kompressorausgangsleitung - div10 - in C 
 highbyte at position 0 of the data block, lowbyte at position 1 of the data block

- TR7 | Temperaturfühler Kältemittel (Gas) - div10 - in C 
 offset 24, highbyte at position 7 of the data block, lowbyte 8 at position of the data block

- TL2 | Außenlufttemperaturfühler - div10 - in C 
 highbyte at position 12 of the data block, lowbyte at position 13 of the data block

- PL1 | Niedrigdruckfühler - div10 - in C 
 offset 24, highbyte at position 3 of the data block, lowbyte 4 at position of the data block

- PH1 | Hochdruckfühler - div10 - in C 
 offset 24, highbyte at position 5 of the data block, lowbyte 6 at position of the data block

The values are calculated as follows:
temp=(highbyte*256+lowbyte)/divisor

proddy, 2021-03-11:
got it. It should be very easy to add. Now I just need the english names as this wil be used in the web, shortened for MQTT and the C++ variables in the code. can you translate them?

Archidus, 2021-03-11:
The folowing translation should fit

TC0 | Water temperature condenser inlet

TC1 | Water temperture condenser output 

TC3 | Condenser temperature

TR3 | Refrigerant temperature liquid side (condenser output)

TR4 | Evaporator inlet temperature 

TR5 | Compressor Inlet temperature

TR6 | Compressor outlet temperature

TR7 | Refrigerant temperature gas side (condenser input) 

TL2 | Air inlet temperature

PL1 | Low pressure side temperature

PH1 | High pressure side temperature