emsesp/EMS-ESP #728: Set HC3 offset via MQTT

swa72, 2021-03-06:
I'm absolutely new to MQTT and messaging stuff around. I have defined a script in HA to change the room offset. But for some reason or the other nothing changes. I can do `call thermostat offsettemp 1 3` on the command line nicely. All data visualizes in HA fine (so HA receives loads of stuff). Here is my script which does exactly ... nothing.

```
alias: Heating HC3 Offset Plus1
sequence:
  - service: mqtt.publish
    data:
      topic: 'ems-esp/thermostat_cmd'
      payload: '{"cmd":"offsettemp","data":1,"id":3}'
mode: single
```
By now I have messed around with any permutation of `thermostat_cmd` or `thermostat` , `"id"` or ` "hc3"` or god what not - to no avail.

proddy, 2021-03-07:
The topic should be `thermostat` or `thermostat_hc1`. See https://emsesp.github.io/docs/#/MQTT?id=sending-commands

EMS-ESP subscribes to MQTT topics, basically it's telling the MQTT broker/server to send any messages that match a topic to it. You can see which these are by going to EMS-ESP's console and typing `show mqtt` e.g.

[image]

Also best to use a `data_template` in HA instead of `data`. There's an example here: https://emsesp.github.io/docs/#/Home-Assistant?id=example-activating-one-time-hot-water-charging-dhw-once

let us know how you get on. We can add your script to the library

swa72, 2021-03-07:
Ok, got, it. 
```
MQTT topic subscriptions:
 ems-esp/system
 ems-esp/boiler
 ems-esp/solar
 ems-esp/thermostat
 ems-esp/thermostat_hc1
 ems-esp/thermostat_hc2
 ems-esp/thermostat_hc3
MQTT queue is empty
```
I'll did a 
```
alias: Heating HC3 Offset Minus3
sequence:
  - service: mqtt.publish
    data:
      topic: ems-esp/thermostat_hc3
      payload: '{"cmd":"offsettemp","data":-3}'
mode: single
```
and observe ...
```
000+06:01:25.498 I 1: [thermostat] Setting thermostat temperature to -3.0 for heating circuit 1, mode offset
```
And a blink of an eye later, HC1 (!) in HA turns to offset -3. I find this confusing. Luckily ...
```
alias: Heating HC3 Offset Minus3
sequence:
  - service: mqtt.publish
    data:
      topic: ems-esp/thermostat
      payload: '{"cmd":"offsettemp","data":-3, "id":3}'
mode: single
```
behaves as expected :-)

proddy, 2021-03-07:
wow, you found a bug. I'm fixing it now!

swa72, 2021-03-07:
Thanks ;-) Forgot to mention that `data_template` is deprecated. You can just use `data` these days.

proddy, 2021-03-07:
Ok, I added an extra error line. I was mistaken at first. With HA the MQTT topic `thermostat_hc<n>` has a special use case. It takes either a mode string (auto, heat, off...) or a temperature value. Is just how the silly HA climate component works. So you can't send commands in the format `{"cmd":"<cmd>", "data":<data>, "id":<n>}` to these topics. Like you did, you have to use the plain old `thermostat` one. Thanks for spotting the mistake.

closed 2021-03-07 by proddy