emsesp/EMS-ESP #732: Burner selected max power can not be changed

tp1de, 2021-03-07:
The Burner selected max power (mqtt: selBurnPow) can not be changed.
Telnet console commands: _call boiler burnmaxpower 60_  do not change the value. (Standard 70% for me).

EMS+ RC310/KB192i

proddy, 2021-03-07:
can you do the command again in the Console with `log debug` so we can see what telegrams come back after the call

tp1de, 2021-03-08:
ems-esp:/#
ems-esp:/# log debug
Log level = debug
ems-esp:/# call boiler burnmaxpower 60
000+17:27:53.484 I 181: [boiler] Setting boiler max power to 60 %
000+17:27:53.702 D 182: [telegram] Sending write Tx [#67], telegram: 0A 08 E6 06 3C 63
000+17:27:53.727 D 183: [emsesp] Last Tx write successful
000+17:27:53.727 D 184: [telegram] Sending post validate read, type ID 0xE6 to dest 0x08
000+17:27:53.927 D 185: [telegram] Sending read Tx [#68], telegram: 0A 88 E6 00 20 17
000+17:27:53.986 D 186: [emsesp] Last Tx read successful
000+17:27:54.302 D 187: [telegram] Sending read Tx [#69], telegram: 0A 88 E6 19 20 25
000+17:27:54.337 D 188: [emsesp] Last Tx read successful
000+17:27:55.632 D 189: [emsesp] No telegram type handler found for ID 0xBF (src 0x08)
000+17:27:57.440 D 190: [emsesp] No telegram type handler found for ID 0x2D6 (src 0x08)
000+17:27:57.719 D 191: [emsesp] No telegram type handler found for ID 0x8E4 (src 0x08)
000+17:28:01.104 D 192: [emsesp] Fetching values for device ID 0x08
000+17:28:01.104 D 193: [telegram] Tx read request to device 0x08 for type ID 0x14
000+17:28:01.104 D 194: [telegram] Tx read request to device 0x08 for type ID 0xE6
000+17:28:01.104 D 195: [telegram] Tx read request to device 0x08 for type ID 0xEA
000+17:28:01.104 D 196: [emsesp] Fetching values for device ID 0x09
000+17:28:01.104 D 197: [emsesp] Fetching values for device ID 0x10
000+17:28:01.104 D 198: [telegram] Tx read request to device 0x10 for type ID 0x2A5
000+17:28:01.104 D 199: [telegram] Tx read request to device 0x10 for type ID 0x2B9
000+17:28:01.104 D 200: [telegram] Tx read request to device 0x10 for type ID 0x2AF
000+17:28:01.104 D 201: [telegram] Tx read request to device 0x10 for type ID 0x29B
000+17:28:01.104 D 202: [telegram] Tx read request to device 0x10 for type ID 0x2A6
000+17:28:01.104 D 203: [telegram] Tx read request to device 0x10 for type ID 0x2BA
000+17:28:01.104 D 204: [telegram] Tx read request to device 0x10 for type ID 0x2B0
000+17:28:01.104 D 205: [telegram] Tx read request to device 0x10 for type ID 0x29C
000+17:28:01.104 D 206: [telegram] Tx read request to device 0x10 for type ID 0x2F5
000+17:28:01.104 D 207: [telegram] Tx read request to device 0x10 for type ID 0x23A
000+17:28:01.104 D 208: [telegram] Tx read request to device 0x10 for type ID 0x240
000+17:28:01.104 D 209: [emsesp] Fetching values for device ID 0x21
000+17:28:01.104 D 210: [telegram] Tx read request to device 0x21 for type ID 0x2D8
000+17:28:01.104 D 211: [emsesp] Fetching values for device ID 0x48
000+17:28:01.329 D 212: [telegram] Sending read Tx [#70], telegram: 0A 88 14 00 20 F4
000+17:28:01.360 D 213: [emsesp] Last Tx read successful
000+17:28:02.077 D 214: [telegram] Sending read Tx [#71], telegram: 0A 88 E6 00 20 17
000+17:28:02.135 D 215: [emsesp] Last Tx read successful
000+17:28:02.377 D 216: [telegram] Sending read Tx [#85], telegram: 0A 88 E6 19 20 25
000+17:28:02.412 D 217: [emsesp] Last Tx read successful
000+17:28:02.602 D 218: [telegram] Sending read Tx [#72], telegram: 0A 88 EA 00 20 27
000+17:28:02.659 D 219: [emsesp] Last Tx read successful
000+17:28:02.952 D 220: [telegram] Sending read Tx [#73], telegram: 0A 90 FF 00 20 01 A5 59
000+17:28:03.402 D 221: [telegram] Sending read Tx [#86], telegram: 0A 90 FF 00 20 01 A5 59
000+17:28:03.504 D 222: [emsesp] Last Tx read successful
000+17:28:03.927 D 223: [telegram] Sending read Tx [#87], telegram: 0A 90 FF 19 20 01 A5 91
000+17:28:04.000 D 224: [emsesp] Last Tx read successful
000+17:28:04.252 D 225: [telegram] Sending read Tx [#74], telegram: 0A 90 FF 00 20 01 B9 45
000+17:28:06.072 D 226: [emsesp] No telegram type handler found for ID 0x8E4 (src 0x08)
000+17:28:06.252 D 227: [telegram] Sending read Tx [#88], telegram: 0A 90 FF 00 20 01 B9 45
000+17:28:06.339 D 228: [emsesp] Last Tx read successful
000+17:28:06.802 D 229: [telegram] Sending read Tx [#75], telegram: 0A 90 FF 00 20 01 AF 53
000+17:28:06.872 D 230: [emsesp] Last Tx read successful
000+17:28:07.179 D 231: [telegram] Sending read Tx [#76], telegram: 0A 90 FF 00 20 01 9B 67
000+17:28:07.242 D 232: [emsesp] Last Tx read successful
000+17:28:07.703 D 233: [telegram] Sending read Tx [#77], telegram: 0A 90 FF 00 20 01 A6 5A
000+17:28:07.802 D 234: [emsesp] Last Tx read successful
000+17:28:08.003 D 235: [telegram] Sending read Tx [#89], telegram: 0A 90 FF 19 20 01 A6 92
000+17:28:08.076 D 236: [emsesp] Last Tx read successful
000+17:28:08.304 D 237: [telegram] Sending read Tx [#78], telegram: 0A 90 FF 00 20 01 BA 46
000+17:28:08.394 D 238: [emsesp] Last Tx read successful
000+17:28:08.579 D 239: [telegram] Sending read Tx [#79], telegram: 0A 90 FF 00 20 01 B0 4C
000+17:28:08.642 D 240: [emsesp] Last Tx read successful
000+17:28:08.954 D 241: [telegram] Sending read Tx [#80], telegram: 0A 90 FF 00 20 01 9C 60
ems-esp:/# log off
Log level = off
ems-esp:/#

MichaelDvP, 2021-03-08:
> ems-esp:/# call boiler burnmaxpower 60
000+17:27:53.484 I 181: [boiler] Setting boiler max power to 60 %
000+17:27:53.702 D 182: [telegram] Sending write Tx [#67], telegram: 0A 08 E6 06 3C 63
000+17:27:53.727 D 183: [emsesp] Last Tx write successful
000+17:27:53.727 D 184: [telegram] Sending post validate read, type ID 0xE6 to dest 0x08
000+17:27:53.927 D 185: [telegram] Sending read Tx [#68], telegram: 0A 88 E6 00 20 17
000+17:27:53.986 D 186: [emsesp] Last Tx read successful

Sending and readback are successfull. Seems the boier does not accept the value or it is set back by another device.
Please try again with `log all` to see the telegrams, and also try other values, like 80%, maybe there is a internal limit.

tp1de, 2021-03-08:
@MichaelDvP 
Other values are not working as well  ...
In the km200 interface I see nominal power (wich is allways 100%) and a powerSetpoint wich in my case is only valid for heating.
I set it to 70% on RC310. For Warmwater the boiler is going up to 100% if needed. WW is loaded into WW-Buffer.

Anyhow my boiler always starts with 52% modulation for 2 minutes. Buderus told me that this ins needed to secure stable gas ignition and can not be changed.  Reducing modulation is afterwards quite slow and boiler is "takting" if heating requirement is not high enough.

I could imagine that in my case (KB192i) there might be another telegram to be used for the powerSetpoint. Maybe burnmaxpower tries to change the nominal power ... just a guess.

Would it make sense to change on RC310 and log on console?
If yes everything or for a specific device?

tp1de, 2021-03-08:
ems-esp:/# call boiler burnmaxpower 60
000+19:01:22.268 I 280: [boiler] Setting boiler max power to 60 %
000+19:01:22.420 T 281: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4A2), data: 01 (offset 4)
000+19:01:22.453 T 282: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4A2), data: <empty> (offset 4)
000+19:01:22.497 D 283: [telegram] Sending write Tx [#175], telegram: 0A 08 E6 06 3C 63
000+19:01:22.522 D 284: [emsesp] Last Tx write successful
000+19:01:22.523 D 285: [telegram] Sending post validate read, type ID 0xE6 to dest 0x08
000+19:01:22.745 T 286: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48A), data: 01
000+19:01:22.778 T 287: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48A), data: <empty>
000+19:01:22.801 T 288: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48A), data: 01 (offset 1)
000+19:01:22.830 T 289: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48A), data: <empty> (offset 1)
000+19:01:22.905 T 290: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 48 00 C8 37 02 21 46 24 03 00 00 00 00 00 01 17 01 62 10 00 02 21 80 00
000+19:01:23.059 T 291: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 25 80 00 7F FF 80 00 (offset 27)
000+19:01:23.255 T 292: [emsesp] Boiler(0x08) -> All(0x00), UBAParametersPlus(0xE6), data: 01 50 1E 5A 46 12 64 00 06 FA 3C 03 05 64 00 00 00 28 00 41 03 00 00 00 00 00 00
000+19:01:23.429 T 293: [emsesp] Boiler(0x08) -> All(0x00), UBAParametersPlus(0xE6), data: 00 00 00 (offset 27)
000+19:01:23.622 D 294: [telegram] Sending read Tx [#176], telegram: 0A 88 E6 00 20 17
000+19:01:23.680 D 295: [emsesp] Last Tx read successful
000+19:01:23.680 T 296: [emsesp] Boiler(0x08) -> Me(0x0A), UBAParametersPlus(0xE6), data: 01 50 1E 5A 46 12 64 00 06 FA 3C 03 05 64 00 00 00 28 00 41 03 00 00 00 00 00 00
000+19:01:23.897 T 297: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4A2), data: 02 (offset 1)
000+19:01:23.928 T 298: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4A2), data: <empty> (offset 1)
000+19:01:23.953 T 299: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4A4), data: 01 (offset 1)
000+19:01:23.978 T 300: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 1)
000+19:01:24.001 T 301: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4A4), data: 01 (offset 4)
000+19:01:24.028 T 302: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 4)
000+19:01:24.049 T 303: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48D), data: 02 (offset 10)
000+19:01:24.078 T 304: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48D), data: <empty> (offset 10)
000+19:01:24.132 T 305: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 48 00 C8 37 02 22 46 23 03 00 00 00 00 00 01 17 01 5F 10 00 02 22 80 00
000+19:01:24.284 T 306: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 25 80 00 7F FF 80 00 (offset 27)
000+19:01:24.572 D 307: [telegram] Sending read Tx [#177], telegram: 0A 88 E6 19 20 25
000+19:01:24.608 D 308: [emsesp] Last Tx read successful
000+19:01:24.608 T 309: [emsesp] Boiler(0x08) -> Me(0x0A), UBAParametersPlus(0xE6), data: 00 00 00 00 00 (offset 25)
000+19:01:24.721 T 310: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4A4), data: 01 (offset 5)
000+19:01:24.753 T 311: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 5)
000+19:01:24.777 T 312: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48D), data: 01
000+19:01:24.803 T 313: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48D), data: <empty>
000+19:01:24.829 T 314: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48D), data: 01 (offset 8)
000+19:01:24.853 T 315: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48D), data: <empty> (offset 8)
000+19:01:25.169 T 316: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48F), data: 02 (offset 8)
000+19:01:25.178 T 317: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 8)
000+19:01:25.419 T 318: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48F), data: 02 (offset 10)
000+19:01:25.428 T 319: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 10)
000+19:01:25.649 T 320: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48F), data: 02 (offset 12)
000+19:01:25.678 T 321: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 12)
000+19:01:25.931 T 322: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 48 00 C8 37 02 24 46 23 03 00 00 00 00 00 01 17 01 58 10 00 02 24 80 00
000+19:01:26.085 T 323: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 25 80 00 7F FF 80 00 (offset 27)
000+19:01:26.273 T 324: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlowPlus(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 24 23 46 50 00 27 46 00
000+19:01:26.449 T 325: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48F), data: 02 (offset 38)
000+19:01:26.478 T 326: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 38)
000+19:01:26.506 T 327: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x48F), data: 02 (offset 14)
000+19:01:26.528 T 328: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 14)
000+19:01:26.553 T 329: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4A4), data: 01 (offset 3)
000+19:01:26.578 T 330: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 3)
000+19:01:26.601 T 331: [emsesp] Gateway(0x48) <- Boiler(0x08), ?(0x4AC), data: 01
000+19:01:26.628 T 332: [emsesp] Boiler(0x08) -> Gateway(0x48), ?(0x4AC), data: <empty>
[... cut ...]
000+19:01:27.700 T 339: [emsesp] Thermostat(0x10) -> Boiler(0x08), ?(0xE7), data: 01 02 00
000+19:01:27.747 T 340: [emsesp] Thermostat(0x10) -> All(0x00), RC300Floordry(0x267), data: 00 00
000+19:01:27.980 T 341: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 48 00 C8 37 02 25 46 22 03 00 00 00 00 00 01 17 01 53 10 00 02 25 80 00
000+19:01:28.135 T 342: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFastPlus(0xE4), data: 02 25 80 00 7F FF 80 00 (offset 27)
ems-esp:/# log off

MichaelDvP, 2021-03-08:
In first post you wrote: 
>  (Standard 70% for me).

But in 0xE6 the setting is always 100% and it seems that it can not be changed.
If you can change a value in KM200, than you should try to change it there and catch the telegrams while changing.

tp1de, 2021-03-08:
in km200 the value is read-only. I will change on RC310. 

[image]

> But in 0xE6 the setting is always 100% and it seems that it can not be changed.

I want to change the burner selected max power --- it is 70% in ems-esp too.

MichaelDvP, 2021-03-08:
This is the `selBurnPow` from monitor and is not writable. The setting is `burnMaxPower` or in web ""Burner max power". And i think this is 100% in your settings.
I think `selBurnPow` should better be described by "Burner selected power"  (without the max). I dont have a modulated burner and can not check how this value varies.

tp1de, 2021-03-08:
The **actual** modulation is in **Burner current power**
The **maximum** heating power is in **selBurnPow**

These values are same as in km200 so far so good.
We just need to find out how to change the maximum heating power.

tp1de, 2021-03-08:
I tested with changing the maximum heating power on the rc310. 
Hera are my findings out of syslog:

Changing to 80%:  [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAParametersPlus(0xE6), data: 50 (offset 4)
Changing to 60%:  [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAParametersPlus(0xE6), data: 3C (offset 4)
Changing to 70%:  [emsesp] Thermostat(0x10) -> Boiler(0x08), UBAParametersPlus(0xE6), data: 46 (offset 4)

tp1de, 2021-03-08:
@MichaelDvP what would the respective data commands byte-strings for this examples to test by mqtt?

proddy, 2021-03-08:
make Syslog report the raw data bytes by changing the setting in the Web UI so you get the complete command the RC310 is issuing, then you can use the `send` command like
- either via the console after a `su` and then  `call system send "XX XX XX"`
- or via a URL like `http://ems-esp/api?device=system&cmd=send&data="XX XX XX"`
- or via MQTT with topic=`ems-esp/system` and payload=`{"cmd":"send", "data":"XX XX XX"}` 

it should work! haven't tested it in a long time....

MichaelDvP, 2021-03-08:
On the console you get a feedback after sending, also if you use send from mqtt you get the readback in topic `response`. 
How is this parameter called in the rc310? E6 is the paramter telegram, it should be writeable.

tp1de, 2021-03-08:
In RC310 it is called _Maximale Heizleistung_ (German)
Any changes there are represented by selBurnPow

Do you have an idea which raw telegram I can test?

MichaelDvP, 2021-03-08:
I've checked the E6 from older logs and think it is simply wrong fields for burnMaxPower (offset4 instead of 6) and burnMinPower (offset5 instead of 7).

tp1de, 2021-03-08:
OK. If you tell me what to change in sorce code, I will test

MichaelDvP, 2021-03-08:
Boiler.cpp: offset for maxpower from 6 to 4, minpower from 7 to 5
line 468, 469
`    has_update(telegram->read_value(burnMaxPower_, 4));`
`    has_update(telegram->read_value(burnMinPower_, 5));`
line 733:
`        write_command(EMS_TYPE_UBAParametersPlus, 5, v, EMS_TYPE_UBAParametersPlus);`
line 751
`        write_command(EMS_TYPE_UBAParametersPlus, 4, v, EMS_TYPE_UBAParametersPlus);`

proddy, 2021-03-08:
@tp1de if you get on Discord I'll help you set up GitHub/VSC

tp1de, 2021-03-08:
> @tp1de if you get on Discord I'll help you set up GitHub/VSC

@proddy  I will have a look at Discord tomorrow (I am not using it now) - Thanks for your help offer
Do I need a headset or cam?

tp1de, 2021-03-08:
The changes from @MichaelDvP are working fine - Thanks a lot
burnMaxPower can now be set and burnSelPow will then be the same.
burnMinPower contains now minimum modulation - in my case 18%

commit 18a1406d (2021-03-09): fix #732
closed 2021-03-09 by MichaelDvP