emsesp/EMS-ESP #733: Burner min period

swa72, 2021-03-08:
I noticed that there is a setting `boiler burnperiod` which reads as `sensor.boiler_burner_min_period` in HA and as `Burner min period` in EMS-ESP. 

I don't want to sound too anal, but users may misunderstand this as a minimal duration that the burner fires.  The manual of the RC300 describes this setting as the duration the burner pauses **after** the end of a burn period ("Taktsperre" in German), regardless of the burn period.

MichaelDvP, 2021-03-08:
Do you have a suggestion for a correct name?

proddy, 2021-03-08:
was going to ask the same :-)

tp1de, 2021-03-08:
A direct translation of German _Taktsperre_ would be _clock inhibit_. But I think this is not much better. 
maybe: Burner inhibit time - or leave it as it is ....

Just to be clear: The definition is not : 
> RC300 describes this setting as the duration the burner pauses after the end of a burn period

In my case it is the minimum time period between 2 boiler starts. I have set it to 15 minutes. 
If boiler is on for 10 minutes, then the waiting time is 5 minutes if additionally the boiler temp is below selected flow temp - hysteresis value.

swa72, 2021-03-08:
@tp1de thanks for clarification, I was looking for something in KM200's data structure (https://github.com/proddy/EMS-ESP/issues/687) but didn't find anything. For the time being, I would leave it as is and suggest updating the docs...

MichaelDvP, 2021-03-08:
>  it is the minimum time period between 2 boiler starts.

That's what i also have in mind. But then we have the right name already, a period is defined from start-to start, as synonym for `period` we can use `cycleTime`.

tp1de, 2021-03-08:
km200 does not include any heating control parameters for the boiler. 
These can only be changed on the RC310 and actually partial by ems-esp.

By the way: changing burnperiod seems to work, hysteresis as well. 
I just struggle with the selected max power ... see #732 

RC310 / KB192i / v3.0.0b5

nagyrobi, 2021-03-08:
Mine is set 3 minutes (by the factory?)

swa72, 2021-03-09:
@tp1de  running `call boiler burnmaxpower 50` worked like a charm on my GB192, RC300 - I'm still on version 3.0.0b1.

MichaelDvP, 2021-03-09:
@swa72 Can you please check which telegram your boiler uses. 
I hope you get an answer to `read 8 16` but empty telegram on `read 8 e6`.  
In #732 we are on the way to change E6 interpretation, but dont want to break settings that works on you system.

swa72, 2021-03-09:
Sure. Please give me a hint how to determine the telegram.

MichaelDvP, 2021-03-09:
Go to telnet, su, and try the mentioned read commands `read 08 16` and `read 08 E6`. One of the telegram 16 or E6 will respond with data, the other will repond with telegram empty. 
But there are also a lot of other ways to determine the telegrams.

swa72, 2021-03-09:
`
ems-esp:/# read 08 E6
001+23:01:51.774 N 3: [emsesp] Boiler(0x08) -> Me(0x0B),
UBAParametersPlus(0xE6), data: <empty>
ems-esp:/# read 08 16
001+23:02:09.902 N 4: [emsesp] Boiler(0x08) -> Me(0x0B),
UBAParameters(0x16), data: FF 52 32 00 06 FA 1E 01 01 64 46 00 00 00 00 00
00 00 00 00 00 00 00 00 28 00 3C
001+23:02:10.427 N 5: [emsesp] Boiler(0x08) -> Me(0x0B),
UBAParameters(0x16), data: 00 3C 00 (offset 25)
`

tp1de, 2021-03-09:
@MichaelDvP  Should be ok with the changes done yesterday.
My boiler responds to UBAParametersPlus(0xE6) and the GB192 from @swa72 to UBAParameters(0x16)

MichaelDvP, 2021-03-09:
Yes, no conflicts, we can change the E6 and close #732. @tp1de have you checked the swapping of #723, can we close this also? If it works i port it to v2.

Back to topic: Rename `burner min period` to `burner min cycle time`? Or another name?

tp1de, 2021-03-09:
@MichaelDvP  Yes I confirm:  both #732 and #723 are working fine.

> Back to topic: Rename `burner min period` to `burner min cycle time`? Or another name?

I would recommend to stay for the meantime to the existing name: `burner min period`
Midterm I believe that km200 and ems-esp namestructure should be harmonized as much as possible. (see #687)
I will prepare an overview of names ... but it is work in progress. It might take 2-3 weeks.

glitter-ball, 2021-03-12:
Burner anti cycle time is another option. This setting is there to stop the burner repeatedly firing and shutting down when the system load is low. It's usually paired with a temperature setting. The flow temperature must fall below (setpoint - anti cycle temperature) and the anti cycle time expire before the burner will light again.

closed 2021-10-04 by proddy