emsesp/EMS-ESP #735: Junkers Ceraclass with built-in FW200 thermostat?

nagyrobi, 2021-03-09:
Just installed a BbqKees Gateway Premuim II (P2) with my Junkers Ceraclass Excellence ZSC 28-3 MFA E 23 boiler.
It detects the following devices:
- Boiler, Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3
- Controller, HT3
- Thermostat, Junkers FW200

[image]

Looks like this (sorry for the bad quality photo):

[image]

Please help me understand the role of the FW200 in this setup. I especially don't understand how the `(hc 1) Setpoint room temperature` and the `(hc 1) Current room temperature` should be interpreted.
Although this is labelled as Thermostat, as Kees pointed out to me by e-mail, this is more probably just a controller handling the operation of the system. The boiler has 2 temperature sensors, one outdoor, and one measuring the temperature of the warm water buffer tank. It has no indoor/room temperature sensor.

The boiler also has a dry relay contact input which was originally connected to a bimetal thermostat, now it's connected to a relay commanded by Tasmota and Home Assistant. I'd like to eliminate this and instead manipulate the system directly from Home Assistant. 

Also I'd like to manually control the warm water preparation as there's also a solar system present in the system - by making smarter decisions based on the expected weater/sunshine etc.

Question: is this built-in FW200 enough for me to achieve my goal? I tried to activate one time charging without success. Also tried to disable the warm water activated status without success.

pipacsba, 2021-03-09:
I also have a FW200 thermostat with a Bosch boiler.
The FW200 contains a room temperature sensor, and based on the settings it can control the system to achieve the setpoint room temperature (although the indoor temperature can be weighted in the FW200, so it can also neglect the internal temperature).

I have no experience with it, but you can probably use the FW200 for your goal, as it can handle solar systems. For this you probably need to find out what messages you need to send to the FW200 from the gateway.

Installation manual that describes the solar settings of the FW200:
http://www.bosch-climate.us/files/201107251719310.6720643256_BoschGreenstar_FW200-ISM.pdf
To find the messages you can probably use:
https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT_EMS_Bus_messages.pdf

pipacsba, 2021-03-09:
I am pretty sure it is an FW200, but according to the manual it offers possibility to Installation in boiler or Wall mounting (page 14 and 15). Some options in the manual are hidden if you have boiler or wall mount (e.g. room influence setting is only available if it is wall mounted).

nagyrobi, 2021-03-09:
Hm thanks... I see now. It looks like FW200 auto-detects if it's installed inside the boiler.
And I assume that the temperature shown in EMS-ESP is indeed the one sensed by the device, however not taken into consideration (thus hidden from the user).

nagyrobi, 2021-03-09:
The dry relay contact input is on the Heatronic 3 (HT3) board of the boiler. It's intended to be used with any third party thermostat which just closes contacts when desired temperature is reached. It's independent of FW200.

**But!** It looks like Bosch / Junkers remote controllers FB10 or FB100 are supported by FW200 on the bus, and it can be set to which heating circuit should it apply to. These have temperature sensors built-on.

Sooo, my question would be "simple": to simulate an FB10 or an FB100 on the bus for FW200. Any ideas?