emsesp/EMS-ESP #737: Is there as temporaryRoomSetpoint per hc available similar to km200?

tp1de, 2021-03-09:
I am looking for a temporaryRoomSetpoint  per hc and can't find it in ems-esp.
Is this datafield available like in km200 and writeable?

In km200 it can have the following values:

-1: Automatic mode
0: hc is switched off
5-30: temperature setpoint in °C for hc

The setpoint is valid until next switchpoint from active switchprogram when in automatic mode.
This is very convenient to use, since it sets back automatic to time program mode.

tp1de, 2021-03-09:
I did some debugging by changing values by km200 and watching on console:

Gateway(0x48) -> Thermostat(0x10)
hc1:  RC300Set(0x2B9) offset 8
hc2:  RC300Set(0x2BA) offset 8

value 0  :  data: 00
value -1:   data: FF
value 20:  data: 28

tp1de, 2021-03-09:
@proddy @MichaelDvP 
I can't find processing of messages 2B9 and 2BA offset 8 in thermostat.cpp.
So I believe that this datavariable(s) doesn't exist yet.

proddy, 2021-03-09:
very easy to add. Do you want to try if I show you what and where to add?

tp1de, 2021-03-09:
@proddy 
> very easy to add. Do you want to try if I show you what and where to add?

Of course I would like and support as much as possible. Just tell me the logic how to add datafields and assign values for reading and writing.

proddy, 2021-03-09:
its handy for working real-time, via chatting & screen-sharing than posting github issues. But since your managed to get EMS-ESP v3 to build using VSC I think you're all set

tp1de, 2021-03-09:
@MichaelDvP 
I just tested. 
For hc1 I can set the temporaryRoomSetpoint to values between 5-30°C and to 0=Off. 
-1= Back to auto mode does'nt work - it then takes 5°C as a minimum. How to set back to auto mode?

For hc2 I haven't found a way by console command or mqtt at all. What would be the syntax?

Should'nt the telegram RC300Set beeing processed per hc for offset 8 to set the respective temporary setpoints correctly?
Technically it is not just per room but per hc. E.g. I am switching off the floor heating hc2 temporary when sun warming is present / expected by forecast.

MichaelDvP, 2021-03-09:
The syntax for commands is in the [wiki](https://emsesp.github.io/docs/#/API)

tp1de, 2021-03-11:
> The syntax for commands is in the [wiki](https://emsesp.github.io/docs/#/API)

I am to stupid for this wiki. I do not find any solution that works on mqtt to write a temporary room setpoints to hc1 and hc2.
Is anyboddy willing to help?

closed 2021-03-11 by tp1de