emsesp/EMS-ESP #738: (ESP32) Bring back MQTT single topics for Thermostat (and possible others)

proddy, 2021-03-09:
In v3 only the nested MQTT was used, but there is a request from Christian to have separate topics per thermostat heating circuits which makes it easier when working with FHEM.

bbqkees, 2021-03-09:
What was the reason to remove the 'single' structure from V3?

proddy, 2021-03-09:
> What was the reason to remove the 'single' structure from V3?

can't remember!

MichaelDvP, 2021-03-09:
I think this was it in  [#632: Reason I'm asking is that is greatly simplifies the new code I'm writing for all the EMS devices (not dallas)](https://github.com/proddy/EMS-ESP/issues/632#issuecomment-738856619)

proddy, 2021-03-09:
Amazing memory you have Michael! I’m working on a solution now and testing with thermostat first.

proddy, 2021-03-09:
implemented for mixer and thermostat, and made some other improvements along the way. Still need to fully test with web and HA, but first implementation is checked in.

proddy, 2021-03-10:
also @MichaelDvP I took the liberty of lower casing all the text strings. Also the `show` in Console reflects the tags used.

MichaelDvP, 2021-03-10:
Yes, i've seen that, i switched back to esp32 this morning. But the change of the boiler-tags crashes the path in ha `stat-t`.
Also the `val_tpl` have to be adapted to non-nested.
I've checked with mqtt-explorer and set the boiler -tags back to `data`, `data_ww`, ..
in mqtt for stat_t:
```
    if (device_type == EMSdevice::DeviceType::BOILER) {
        snprintf_P(stat_t, sizeof(stat_t), PSTR("~/boiler_%s"), EMSdevice::tag_to_string(tag).c_str());
    } else if (device_type == EMSdevice::DeviceType::SYSTEM) {
        snprintf_P(stat_t, sizeof(stat_t), PSTR("~/heartbeat"));
    } else if (nested_format_ || !have_prefix) {
        snprintf_P(stat_t, sizeof(stat_t), PSTR("~/%s_data"), device_name);
    } else {
        snprintf_P(stat_t, sizeof(stat_t), PSTR("~/%s_data_%s"), device_name, EMSdevice::tag_to_string(tag).c_str());
    }
    doc["stat_t"] = stat_t;

```
and for val_tpl:
```
   if (nested_format_) {
        snprintf_P(val_tpl, sizeof(val_tpl), PSTR("{{value_json.%s}}"), new_entity);
    } else {
        snprintf_P(val_tpl, sizeof(val_tpl), PSTR("{{value_json.%s}}"), uuid::read_flash_string(entity).c_str());
    }
    doc["val_tpl"] = val_tpl;
```
and skip the prefixes in emsdevice.cpp:
```
                if (have_tag && !DeviceType::BOILER) {
```
This looks good in mqtt-explorer, but removes some of your intended changes.

proddy, 2021-03-10:
yes, I haven't got around to changing the HA Discovery yet. I was feeling lazy and thinking if someone is using HA then they wouldn't care what format the MQTT is in so I'll force it to the default 'nested'.

The error I see when not in nested mode is in the thermostat data
```
 [22] (Pub) topic=thermostat_data payload={"id":192,"seltemp":20.7,"currtemp":23,"mode":"holiday","modetype":"eco"}
 [23] (Pub) topic=thermostat_data_hc1 payload={"seltemp":20.5,"currtemp":22.8,"mode":"holiday"}
 [24] (Pub) topic=thermostat_data_hc2 payload={"seltemp":20.6,"currtemp":22.9,"mode":"auto","modetype":"nofrost"}
 [25] (Pub) topic=thermostat_data_hc3 payload={"seltemp":20.7,"currtemp":23,"mode":"holiday","modetype":"eco"}
```

the topic `thermostat_data` should only contain the device values with the TAG_NONE. I'm fixing that now.

BTW I'm doing most of the testing standalone, with an ESP which is faster to debug. I'm on Win10 with the Linux subsystem (WSL) installed (just an option in windows). Ubuntu comes with gcc+ and gnumake so in the pio terminal window doing a `make run` will build the EMS-ESP project in a few seconds and fire up the console. Then `test` will run any test I want. Saves me hours.

proddy, 2021-03-10:
fix the bug from last night. next is the HA implementation to work with non-nested MQTT

proddy, 2021-03-10:
@MichaelDvP @bbqkees the latest changes with tagging also has an influence on how the data is displayed in the web, see:

[image]

the tag is prefixed, also to match the MQTT topics now. Question: 
- how does this look to you?
- should we display the tag at all for the boiler ? For a thermostat, it is neede (e.g. (hc1) .... )
- is this the right logical grouping anyway?

One advantage is that it does means we can shorten the name, so now "Warm Water Set Temperature" is now just "set temperature" for the grouping "warm water" which saves a few bytes here and there.

MichaelDvP, 2021-03-11:
We have headline boiler, i think the (boiler)-tag is redundant and does not fit to the other devices: thermostat have for normal values also no tag, only for (hc). (warm water) is ok to me. Info was used mainly to keep the boiler_data in the mqtt package limit. Do we need it with the new mqtt-lib?

proddy, 2021-03-11:
ok, agree. I'll leave only (warm water) and remove (boiler) and also remove the `boiler_info` topic. It's a breaking change and people will moan, but hey we're at the bleeding edge of technology! With the new MQTT lib we can have payloads up to 4KB easily. I was thinking of seeing what would happen of all the boiler data was on a single `bolier_data` topic. it's only changing one line in the code.

bbqkees, 2021-03-11:
The boiler tag in the web may be redundant, but it's still a clear indication such that the overview is easy to understand in a single glance. If you show a tag everywhere except for one type, why not add it as well so it all looks the same?

For MQTT we should keep in mind a lot of people are actually using it. So with every breaking change every single one has to update their (custom) interpreters.

gurkc006, 2021-03-11:
"single" type MQTT is working for me again! Thanks! Perfect!

proddy, 2021-03-11:
thinks its all done now. @MichaelDvP I've merged in your changes. Hope everything still works

closed 2021-03-11 by proddy