emsesp/EMS-ESP #739: Unknown devices on Buderus system

csubakk, 2021-03-09:
The system is: Buderus Logamax Plus GB172i-42i (2 pieces), Buderus MM100 modules (2 pc), Buderus MC 400 module (1 pc) and Buderus RC310 Thermostat (1 pc).

I can read or write informations from the Thermostat and read info from MM100 modules.

My problem: I cannot see all informations, EMS-ESP does not recognize all devices 

The system info result is:

{
  "System": {
    "version": "2.2.0",
    "uptime": "000+00:07:48.999",
    "freemem": 45,
    "fragmem": 22
  },
  "Settings": {
    "enabled": "on",
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "mqtt_format": 3,
    "mqtt_qos": 0,
    "mqtt_retain": "off",
    "tx_mode": 1,
    "ems_bus_id": 11,
    "master_thermostat": 0,
    "rx_gpio": 13,
    "tx_gpio": 15,
    "dallas_gpio": 14,
    "dallas_parasite": "off",
    "led_gpio": 2,
    "hide_led": "off",
    "api_enabled": "off",
    "bool_format": 1,
    "analog_enabled": "off"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 878,
    "#read requests sent": 103,
    "#write requests sent": 3,
    "#incomplete telegrams": 51,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0
  },
  "Devices": [
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x20, ProductID:160, Version:24.05)",
      "handlers": "0x2D7"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x28, ProductID:160, Version:24.05)",
      "handlers": "0x331"
    },
    {
      "type": "Unknown",
      "name": "unknown (DeviceID:0x08, ProductID:210, Version:02.05)",
      "handlers": ""
    },
    {
      "type": "Unknown",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x70, ProductID:123, Version:07.00)",
      "handlers": ""
    },
    {}
  ]
}

proddy, 2021-03-09:
It looks we first need to
- add your GB172i-42i boiler to the database so it gets recognized as deviceID 0x08
- figure out why the MC 400 is not showing up - if you can provide details from the controller that would help
- figure out what that deviceID 0x70 really is - @MichaelDvP any ideas?

MichaelDvP, 2021-03-09:
@csubakk Please try with [new dev](https://github.com/proddy/EMS-ESP/releases/tag/esp8266_dev). 
I added the boiler at 0x08, the second is type unknown. 
Please check and use telnet and command `watch on 70`  for some minutes and post the log here.

csubakk, 2021-03-10:
Wao...

1 boiler and 2 unknown devices:

[image]

MC400 is an ESP device, a manual in english is here: https://www.libble.eu/buderus-mc-400/online-manual-877928/

But the data: Is possibile from the MC 400:

[image]

The log:

> [Kems-esp:/$ [Kems-esp:/$ [K000+00:37:44.064 E 1: [telegram] Rx: 10 0B FF 00 01 F5 02 FF 02 03 00 00 08 01 00 08 04 00 00 00 (CRC 00 != 2F)
> ems-esp:/$ [K000+00:37:45.471 E 2: [telegram] Rx: F8 08 00 E4 1B 00 00 00 00 80 00 00 00 00 00 00 00 00 64 64 E2 (CRC E2 != 26)
> ems-esp:/$ [K000+00:37:57.096 E 3: [telegram] Rx: 10 08 00 FF 00 06 07 00 00 00 00 4E 4E 64 64 0E 0E 4E 00 00 AA (CRC AA != 0D)
> ems-esp:/$ [K000+00:37:58.338 E 4: [telegram] Rx: 10 0B FF 00 01 40 32 00 06 00 00 00 00 (CRC 00 != DA)
> ems-esp:/$ [K000+00:38:02.893 E 5: [telegram] Rx: 28 00 BF 00 28 A0 00 00 00 00 00 (CRC 00 != 80)
> ems-esp:/$ [K000+00:38:04.516 E 6: [telegram] Rx: FF 08 00 E3 00 00 00 01 00 00 00 00 00 00 00 00 01 D9 00 00 5A 00 2F 64 45 (CRC 45 != F7)
> ems-esp:/$ [K000+00:38:07.672 N 7: [emsesp] Thermostat(0x10) <- Unknown(0x70), UBAMonitorFastPlus(0xE4), data: 04 (offset 31)
> ems-esp:/$ [K000+00:38:07.684 N 8: [emsesp] Unknown(0x70) -> Thermostat(0x10), UBAMonitorFastPlus(0xE4), data: <empty> (offset 31)
> ems-esp:/$ [K000+00:38:07.704 N 9: [emsesp] Thermostat(0x10) <- Unknown(0x70), UBAMonitorFastPlus(0xE4), data: 19 (offset 6)
> ems-esp:/$ [K000+00:38:07.721 N 10: [emsesp] Unknown(0x70) -> Thermostat(0x10), UBAMonitorFastPlus(0xE4), data: 05 02 48 00 00 00 (offset 6)
> ems-esp:/$ [K000+00:38:07.745 N 11: [emsesp] Thermostat(0x10) <- Unknown(0x70), ?(0xD2), data: 01 (offset 2)
> ems-esp:/$ [K000+00:38:09.780 E 12: [telegram] Rx: 10 0B FF 00 01 A5 00 ED 23 2E 2F 00 2E 2E 05 A0 03 03 03 05 A0 01 E6 00 00 11 01 03 09 43 00 D2 (CRC D2 != 46)
> ems-esp:/$ [K000+00:38:11.540 N 13: [emsesp] Thermostat(0x10) <- Unknown(0x70), ?(0xD2), data: 01 (offset 2)
> ems-esp:/$ [K000+00:38:11.556 N 14: [emsesp] Unknown(0x70) -> Thermostat(0x10), ?(0xD2), data: 00 (offset 2)
> ems-esp:/$ [K000+00:38:11.666 N 15: [emsesp] Thermostat(0x10) <- Unknown(0x70), ?(0x6DC), data: 06
> ems-esp:/$ [K000+00:38:11.687 N 16: [emsesp] Unknown(0x70) -> Thermostat(0x10), ?(0x6DC), data: 01 00 01 00 0C 82 DE
> ems-esp:/$ [K000+00:38:11.709 E 17: [telegram] Rx: FF 10 F4 D2 02 01 90 (CRC 90 != 7E)
> ems-esp:/$ [K000+00:38:14.742 E 18: [telegram] Rx: FE 08 00 E3 00 00 00 01 00 00 00 00 00 00 00 00 01 D9 00 00 5A 00 2F 64 45 (CRC 45 != 50)
> ems-esp:/$ [K000+00:38:17.256 E 19: [telegram] Rx: FE 08 00 E4 1B 00 00 00 00 80 00 00 00 00 00 00 00 00 50 64 8A (CRC 8A != 50)
> ems-esp:/$ [K000+00:38:20.569 E 20: [telegram] Rx: 10 0B FF 00 01 F5 02 FF 02 03 00 00 08 01 00 08 04 00 00 00 (CRC 00 != 2F)
> ems-esp:/$ [K000+00:38:22.539 E 21: [telegram] Rx: 10 0B FF 00 01 40 32 00 06 00 00 00 00 00 FF 02 F6 06 FF 00 00 00 00 00 (CRC 00 != 5C)
> ems-esp:/$ [K000+00:38:33.252 E 22: [telegram] Rx: 10 0B FF 00 01 A5 00 ED 23 2E 2F 00 2E 2E 05 A0 03 03 03 05 A0 01 E6 00 00 10 97 (CRC 97 != 63)
> ems-esp:/$ [K000+00:38:33.992 E 23: [telegram] Rx: 10 08 00 E3 00 00 00 01 00 00 00 00 00 00 00 00 01 D9 00 00 5A 00 2F 64 45 (CRC 45 != CF)
> ems-esp:/$ [K000+00:38:57.486 E 24: [telegram] Rx: F8 08 00 FF 00 06 07 00 00 00 00 4E 4E 64 64 0E 0E 4E 00 00 AA (CRC AA != 6E)
> ems-esp:/$ [K000+00:39:04.162 E 25: [telegram] Rx: 10 00 BF 00 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC 00 != 53)
> ems-esp:/$ [K000+00:39:04.483 E 26: [telegram] Rx: FE 08 00 E4 17 01 D5 01 D5 00 00 00 00 80 00 00 00 00 00 00 00 00 50 64 0D (CRC 0D != 18)
> ems-esp:/$ [K000+00:39:05.922 E 27: [telegram] Rx: FE 08 00 FF 00 05 E0 01 10 02 00 21 07 60 4E 02 0C 10 10 00 C8 00 78 (CRC 78 != 3B)
> ems-esp:/$ [K000+00:39:07.397 E 28: [telegram] Rx: 10 0B FF 00 01 F5 02 FF 02 03 00 00 08 01 00 08 04 00 00 (CRC 00 != 9B)
> ems-esp:/$ [K000+00:39:08.481 N 29: [emsesp] Thermostat(0x10) <- Unknown(0x70), UBAMonitorFastPlus(0xE4), data: 04 (offset 31)
> ems-esp:/$ [K000+00:39:23.853 N 30: [emsesp] Thermostat(0x10) <- Unknown(0x70), UBAMonitorFastPlus(0xE4), data: 04 (offset 31)
> ems-esp:/$ [K000+00:39:23.870 N 31: [emsesp] Unknown(0x70) -> Thermostat(0x10), UBAMonitorFastPlus(0xE4), data: <empty> (offset 31)
> ems-esp:/$ [K000+00:39:23.894 N 32: [emsesp] Thermostat(0x10) <- Unknown(0x70), UBAMonitorFastPlus(0xE4), data: 19 (offset 6)
> ems-esp:/$ [K000+00:39:23.917 N 33: [emsesp] Unknown(0x70) -> Thermostat(0x10), UBAMonitorFastPlus(0xE4), data: 05 02 47 00 00 00 (offset 6)
> ems-esp:/$ [K000+00:39:23.943 N 34: [emsesp] Thermostat(0x10) <- Unknown(0x70), ?(0xD2), data: 01 (offset 2)
> ems-esp:/$ [K000+00:39:23.952 N 35: [emsesp] Unknown(0x70) -> Thermostat(0x10), ?(0xD2), data: 00 (offset 2)
> ems-esp:/$ [K000+00:39:24.059 N 36: [emsesp] Thermostat(0x10) <- Unknown(0x70), ?(0x6DC), data: 06
> ems-esp:/$ [K000+00:39:24.081 N 37: [emsesp] Unknown(0x70) -> Thermostat(0x10), ?(0x6DC), data: 01 00 01 00 0C 82 DE
> ems-esp:/$ [K000+00:39:24.102 E 38: [telegram] Rx: FF 10 F4 D2 02 01 90 (CRC 90 != 7E)
> ems-esp:/$ [K000+00:39:28.318 E 39: [telegram] Rx: 10 0B FF 00 01 40 32 00 06 00 00 00 00 00 FE 93 (CRC 93 != 78)
> ems-esp:/$ [K000+00:39:38.761 E 40: [telegram] Rx: FC 08 00 E3 00 00 00 01 00 00 00 00 00 00 00 00 01 CF 00 00 5A 00 2F 64 B8 (CRC B8 != FA)
> ems-esp:/$ [K000+00:39:40.996 E 41: [telegram] Rx: BC 08 00 E4 1B 00 00 00 00 80 00 00 00 00 00 00 00 00 50 64 8A (CRC 8A != 03)
[... cut ...]
> ems-esp:/$ [K000+00:40:39.204 N 56: [emsesp] Unknown(0x70) -> Thermostat(0x10), UBAMonitorFastPlus(0xE4), data: <empty> (offset 31)
> ems-esp:/$ [K000+00:40:39.227 N 57: [emsesp] Thermostat(0x10) <- Unknown(0x70), UBAMonitorFastPlus(0xE4), data: 19 (offset 6)
> ems-esp:/$ [K000+00:40:39.243 N 58: [emsesp] Unknown(0x70) -> Thermostat(0x10), UBAMonitorFastPlus(0xE4), data: 05 02 46 00 00 00 (offset 6)
> ems-esp:/$ [K000+00:40:39.304 N 59: [emsesp] Thermostat(0x10) <- Unknown(0x70), ?(0x6DC), data: 06

PS: firmware update did not work on MAC (firefox, safari), work on Windows... (crome)

MichaelDvP, 2021-03-10:
Can you give some more information about your system. How it is connected to the ems-bus?
Boiler 1 on EMS1, Boiler 2 on EMS2, Thermostat on EMS i think? Where are the mixers and the ems-esp? 
What position is the switch of the MC400?
You have a lot of CRC errors, i think ems-esp is buspowered. Try with usb-power (pull jumper in ems-esp) and also try if there is a difference if you connect to boiler 1, boiler2 or thermostat.  (boiler connection can also be done with service-jack and jack-power).
If you do on each connection a `log all` for more than a minute we hopefully learn more what happend on the bus(es) and find a solution to integrate to ems.-esp.

@proddy I think the MC400 is product-id 210, but recognizes itself as boiler id 0x08, the two boilers are now 0x70 and 0x74 (because of possible MC400 cascading).

bbqkees, 2021-03-10:
I asked someone else with a MC400 and 3 boilers to provide some details about his system. Hope he joins this discussion.

DLX2550, 2021-03-10:
> I asked someone else with a MC400 and 3 boilers to provide some details about his system. Hope he joins this discussion.

Here is my setup :

3 boilers in cascade via cascade controller Bosch MC 400 with EMS, EMS+ and EMS2

Bosch GC7000IW 42KW -> Via BUS1 on MC400
Bosch GC7000IW 42KW -> Via BUS2 on MC400
Bosch GC7000IW 35KW + WW -> Via BUS3 on MC400

Bosch MC 400 cascade controller via EMS bus
Bosch MB LAN II via EMS bus
Bosch CW400 Thermostaat vis EMS Bus
EMS-ESP via EMS bus

3 x Bosch MM100, (2 not mixing) and (1 mixed group)

hc1 -> MM100 -> NOT mixed -> controlled via MD contact by LOXONE mini server output
hc2 -> MM100 -> NOT mixed -> controlled via MD contact by LOXONE mini server output
hc3 -> MM100 -> MIXED -> currently controlled via CW400

hc1= Jacuzzi heating circuit
hc2= Pool heating circuit
hc3= Radiators heating circuit

As to my understanding, the boilers can not be visible from the EMS bus side or the EMS-ESP since the boilers are connected to BUS1, BUS2, BUS3 and are controlled by the MC 400?

I want to control the MM100 hc3 flow temperature and pump status via the LOXONE mini-server virtual outputs using REST API. Since there is no way to control the MM100 directly, this should be achieved by controlling the CW 400, I suppose? 

At this moment reading the parameters in LOXONE via virtual inputs works fine, however writing does not work, hence the "Allow WEB API to write" is activated. Also cmd=info does not work when passing the id as hc1, hc2 or hc3?

Does anyone have some examples or ideas?
 
[image]
[EMS-ESP.pdf](https://github.com/proddy/EMS-ESP/files/6115950/EMS-ESP.pdf)

DLX2550, 2021-03-10:
0x70 , 0x74, 0x78 show the details of the MM100 hc1, hc2 and hc3

@MichaelDvP this is a premium II gateway and I could not find a way to update higher than 2.2 , I have ordered the new S32 gateway today.

[image]

proddy, 2021-03-10:
@DLX2550 latest dev builds for the ESP8266 (v2) are here: https://github.com/proddy/EMS-ESP/releases/tag/esp8266_dev

csubakk, 2021-03-10:
@MichaelDvP 

MC 400:

[image]

BUS1 --> Boiler 1
BUS2 --> Boiler 2
BUS -->MM100 -->MM100 (first cable)
BUS -->Thermostat and ems-esp (second cable)

The switch position on MC400: 2

The ems-esp was bus-powered, now is USB powered.

[vssh-osx-192.168.68.99.log](https://github.com/proddy/EMS-ESP/files/6116583/vssh-osx-192.168.68.99.log)

Log from BUS1 and/or BUS2 later

MichaelDvP, 2021-03-10:
Nice, a bit more power and the crc-errors are all gone. Setup is mailnly like @DLX2550 and helps to understand the system.

It will take some time to read the log, but on first look the thermostat reads mainy some E4 values direct from boilers, but the cascade publishes them to all. Than reading the cascade will give all information. 

Maybe i've missed something, but i've added the cascaded boilers without values to device-list. 
Can you try with [https://github.com/MichaelDvP/EMS-ESP/releases/tag/esp8266_dev](https://github.com/MichaelDvP/EMS-ESP/releases/tag/esp8266_dev). If it works as expected i'll check in. 
@proddy See changes [here](https://github.com/MichaelDvP/EMS-ESP/commit/0db1677ea2cc8048c15d357f6ff3b274ae404a54)

Adding new values will blow esp8266 memory, if we find more than a few values in the cacaded moodules, this will only got to esp32-v3-version, sorry.

DLX2550, 2021-03-10:
My EMS-ESP is updated to the latest dev now.

Now the cascade shows up as boiler.

Writing to the thermostat via REST API now also works.

Still, need to figure out how to write the target flow temperature value of hc3 and start the pump of hc3?

Any suggestions?

Thanks.

[image]

csubakk, 2021-03-10:
Wired to BUS1:

[image]

:)

And the log:

[vssh-osx-192.168.68.99.log](https://github.com/proddy/EMS-ESP/files/6116992/vssh-osx-192.168.68.99.log)

DLX2550, 2021-03-10:
@csubakk 
Is BUS1, BUS2,BUS3,BUS4 not exclusively reserved for connecting the boilers? As otherwise the cascade controller gets confused when controlling the boilers directly from the EMS-ESP?

csubakk, 2021-03-10:
With new firmware:

[image]

[image]

Thank You @MichaelDvP! I needed the 2 selected sensor data, my problem is resolved.

Boilers are without data, but that isn't a problem for me.

@DLX2550 yes, BUS 1-4 are for boilers, but they are ESP buses...

MichaelDvP, 2021-03-10:
@DLX2550 Yes, but EMS1,.. are sperarated ems-buses and we can read on them. On this  bus the MC400 has device_id 0x18, paying the role of a single thermostat, and the boiler is device_id 0x08. 
It's like NAT in your internet-router. 
The right position for ems-esp in on EMS connector, but this test (thanks to @csubakk) was good to understand the module.

@csubakk Yes, thats what i expected, i'll push here.

DLX2550, 2021-03-10:
@MichaelDvP 

I will notify you once I receive the ESP32. 
Meanwhile, I will experiment with some settings.

[image]

csubakk, 2021-03-11:
[image]

MichaelDvP, 2021-03-11:
@DLX2550 When fetching the free heap there was telnet opened, i think. 
@csubakk This are heap values i expected, it's nearly the minimum to allow the webpage to be loaded.

proddy, 2021-03-11:
> Maybe i've missed something, but i've added the cascaded boilers without values to device-list.
> Can you try with https://github.com/MichaelDvP/EMS-ESP/releases/tag/esp8266_dev. If it works as [...]

@MichaelDvP nice addition.

csubakk, 2021-03-11:
[image]

DLX2550, 2021-03-11:
> @DLX2550 When fetching the free heap there was telnet opened, i think.
> @csubakk This are heap values i expected, it's nearly the minimum to allow the webpage to be loaded.

@MichaelDvP 
Nothing was open except the webpage and just rebooted. As said, the webpage is mostly unresponsive or not available.